medusalix/xow #74: xow fails with "Failed to load firmware" on latest HEAD

NoXPhasma, 2020-05-24:
When I clone the latest HEAD, build and install xow, it won't start due to not being able to load the firmware:
```
Mai 24 15:46:03 NoXP-PC systemd[1]: Started Xbox One Wireless Dongle Driver.
Mai 24 15:46:04 NoXP-PC xow[3448867]: 2020-05-24 15:46:03 INFO  - xow v0.4-47-g81b882c ©Severin v. W.
Mai 24 15:46:04 NoXP-PC xow[3448867]: 2020-05-24 15:46:04 INFO  - Waiting for device...
Mai 24 15:46:05 NoXP-PC xow[3448867]: 2020-05-24 15:46:05 ERROR - Firmware reset timed out
Mai 24 15:46:05 NoXP-PC xow[3448867]: terminate called after throwing an instance of 'Mt76Exception'
Mai 24 15:46:05 NoXP-PC xow[3448867]:   what():  Failed to load firmware
Mai 24 15:46:05 NoXP-PC systemd[1]: xow.service: Main process exited, code=dumped, status=6/ABRT
Mai 24 15:46:05 NoXP-PC systemd[1]: xow.service: Failed with result 'core-dump'.
```

The firmware blob has been successfully downloaded at the make process.

medusalix, 2020-05-24:
Does your `dmesg` show anything related to the `mt76` driver? I've recently updated the blacklist for that thing, maybe that's causing this issue?

NoXPhasma, 2020-05-24:
There is nothing about mt76 in the dmesg. When I go back to 2fc09761f3b2f5739f4568fd868f6f86cae5e34b then there is no error but it also can't initialize the device, as it hangs on:
```
sudo ./xow
2020-05-24 18:09:42 INFO  - xow v0.4-41-g2fc0976 ©Severin v. W.
2020-05-24 18:09:42 INFO  - Waiting for device...
```

medusalix, 2020-05-24:
Very strange. I don't think I've made any changes to the firmware loading process since version 0.4. You aren't dual-booting between Windows and Linux or running multiple instances of xow, are you?

NoXPhasma, 2020-05-24:
I don't dual boot and there is no other instance running. If there is any other xow process running, i get the to expect error about busy USB device:
```
sudo ./xow
2020-05-24 18:09:42 INFO  - xow v0.4-41-g2fc0976 ©Severin v. W.
2020-05-24 18:09:42 INFO  - Waiting for device...
```

medusalix, 2020-05-24:
Can you pinpoint the exact commit where the problem was introduced? I don't have any other clues right now...

NoXPhasma, 2020-05-24:
I will try to bisect it tomorrow.

GloriousEggroll, 2020-05-25:
I was just hitting this. Unplugging and replugging the device then restarting the service allowed it to work correctly

NoXPhasma, 2020-05-25:
You are right, I just needed to unplug and replug the dongle and now it works. And it seems that now it works reliable. Any idea what could cause this?

NoXPhasma, 2020-05-25:
The hint of @GloriousEggroll about repluggin the device helped me to find out that 2fc09761f3b2f5739f4568fd868f6f86cae5e34b actually works. And the issue I had in https://github.com/medusalix/xow/issues/74#issuecomment-633253708 was solved by replugging the dongle.

So the issue definitely starts after that commit.

NoXPhasma, 2020-05-25:
It seems that only the last commit has introduced this issue, as even a1fa00a6 works fine for me.

NoXPhasma, 2020-05-25:
Okay, this is hard to get pinned down to the exact commit, as it introduces hard to reproduce behaviour. Like it works with 35075586, but when restarting it several times it will fail eventually.

medusalix, 2020-05-25:
> Like it works [...], but when restarting it several times it will fail eventually.

This sentence is a perfect summary of what developing this driver feels like :smile:.
I'm wondering whether completely removing the firmware reset might work. Try [this patch](https://github.com/medusalix/xow/files/4676386/fw.txt) on HEAD.

NoXPhasma, 2020-05-25:
Okay, the patch works for the firmware issue. But introduced 2 different issues:

First issue is that the dongle won't accept any new connections after 2 service restarts. Until I replugin the device.

The second is much more serious, as it kills my USB devices until a reboot: 
DMESG:
```
Mai 25 11:54:28 NoXP-PC kernel: xhci_hcd 0000:25:00.0: ERROR unknown event type 15
Mai 25 11:54:33 NoXP-PC kernel: xhci_hcd 0000:25:00.0: xHCI host not responding to stop endpoint command.
Mai 25 11:54:33 NoXP-PC kernel: xhci_hcd 0000:25:00.0: xHCI host controller not responding, assume dead
Mai 25 11:54:33 NoXP-PC kernel: xhci_hcd 0000:25:00.0: HC died; cleaning up
Mai 25 11:54:33 NoXP-PC kernel: usb 3-1: USB disconnect, device number 2
Mai 25 11:54:33 NoXP-PC kernel: usb 3-1.3: USB disconnect, device number 20
Mai 25 11:54:33 NoXP-PC kernel: usb 4-1: USB disconnect, device number 2
Mai 25 11:54:33 NoXP-PC kernel: usb 3-2: USB disconnect, device number 3
```
xow:
```
Mai 25 11:52:37 NoXP-PC systemd[1]: Started Xbox One Wireless Dongle Driver.
Mai 25 11:52:37 NoXP-PC xow[200815]: 2020-05-25 11:52:37 INFO  - xow v0.4-47-g81b882c ©Severin v. W.
Mai 25 11:52:37 NoXP-PC xow[200815]: 2020-05-25 11:52:37 INFO  - Waiting for device...
Mai 25 11:52:37 NoXP-PC xow[200815]: 2020-05-25 11:52:37 INFO  - Wireless address: 62:45:b4:fe:81:16
Mai 25 11:52:37 NoXP-PC xow[200815]: 2020-05-25 11:52:37 INFO  - Dongle initialized
Mai 25 11:54:33 NoXP-PC xow[200815]: 2020-05-25 11:54:33 ERROR - Error in bulk read: LIBUSB_ERROR_NO_DEVICE
Mai 25 11:54:33 NoXP-PC xow[200815]: 2020-05-25 11:54:33 ERROR - Error in bulk read: LIBUSB_ERROR_NO_DEVICE
Mai 25 11:54:33 NoXP-PC xow[200815]: 2020-05-25 11:54:33 INFO  - Shutting down...
Mai 25 11:54:33 NoXP-PC xow[200815]: 2020-05-25 11:54:33 ERROR - Error in bulk write: LIBUSB_ERROR_NO_DEVICE
Mai 25 11:54:33 NoXP-PC xow[200815]: 2020-05-25 11:54:33 ERROR - Failed to write command
Mai 25 11:54:33 NoXP-PC xow[200815]: 2020-05-25 11:54:33 ERROR - Failed to set LED mode
Mai 25 11:54:33 NoXP-PC xow[200815]: 2020-05-25 11:54:33 ERROR - Failed to turn off LED
Mai 25 11:54:33 NoXP-PC xow[200815]: 2020-05-25 11:54:33 ERROR - Error in bulk write: LIBUSB_ERROR_NO_DEVICE
Mai 25 11:54:33 NoXP-PC xow[200815]: 2020-05-25 11:54:33 ERROR - Failed to write command
Mai 25 11:54:33 NoXP-PC xow[200815]: 2020-05-25 11:54:33 ERROR - Failed to set power mode
Mai 25 11:54:33 NoXP-PC xow[200815]: 2020-05-25 11:54:33 ERROR - Failed to turn off radio
Mai 25 11:54:33 NoXP-PC xow[200815]: 2020-05-25 11:54:33 ERROR - Error releasing interface: LIBUSB_ERROR_NO_DEVICE
Mai 25 11:54:33 NoXP-PC systemd[1]: xow.service: Succeeded.
```

medusalix, 2020-05-25:
Oh wow, I didn't imagine this would break your whole USB stack. I guess `git revert`-ing commit f9d03b1ff98878ce655557529711acdf1e1151c0 is probably the best choice. Don't ask me why it fails after several restarts though (was that also introduced by a recent commit?).

NoXPhasma, 2020-05-25:
It's hard to say, I think I've hit that USB issue before with xow, when using git a few weeks ago. I then went back to 0.4 and never happened again. I wasn't entirely sure back then it was xow but now I'm 100% certain.

I completely forgot about that but I think it's not really part of this patch right now but has been introduced since 0.4 at some point.

NoXPhasma, 2020-05-25:
For the restart issue, that doesn't happen on 0.4 and 2fc09761 at all. I can restart the service as often as I want and it still accepts connections. If it has been introduced on master by this patch is impossible to me to say, as it doesn't work reliable without it.

NoXPhasma, 2020-05-25:
About the USB stack issue, it happened when I was playing a game with a friend, which is why I just reverted to 0.4 and didn't have done a major thought about it, as we just wanted to play. However, point why I mention that is, that this must have happened before the 10th of May as that was the last time I played it. I hope that might help to find the causing commit.

medusalix, 2020-05-25:
So commit f9d03b1ff98878ce655557529711acdf1e1151c0 is the only culprit for the firmware issue? I can't reproduce any of these issues, so I'll need your confirmation for that one.

NoXPhasma, 2020-05-25:
Yes indeed, that commit introduced that issue, going back just one more commit (2fc09761) doesn't have that issue.

medusalix, 2020-05-25:
OK, I've reverted that one.
Can you reliably trigger that USB stack issue? I'm asking because the first commit that came to my mind would be 7f755b5b4a8d70b9564326b81ce5d25f3e4cefbf, where I rewrote the USB implementation.

NoXPhasma, 2020-05-25:
Sadly I can not. When it happened a few wees ago, it happened several times after playing a while. Suddenly the Gamepad lost connection and my mouse didn't worked anymore. Was probably just "luck" that it happened this time that early. 

And that commit came into my mind as well. It fits in the time frame and also is about USB implementation.

NoXPhasma, 2020-05-25:
I pulled the latest master and firmware issue is gone.

medusalix, 2020-05-25:
Awesome. I'm wondering whether it could be a bug in the driver of your USB controller that got triggered by xow. I'm just speculating here, it's usually xow's fault :smile:.

NoXPhasma, 2020-05-25:
It's not impossible but very unlikely as it never happened before in several years as I have this board. It only started once I used the git a few weeks ago and after I reverted to 0.4 it never happened again. And it only came back today with xow git again.

medusalix, 2020-05-25:
Well, I'm closing this one now. Make sure to open an issue if you hit that bug again.

closed 2020-05-25 by medusalix