medusalix/xow #76: xow crashes and kills my USB stack after a while

NoXPhasma, 2020-05-25:
As already reported in https://github.com/medusalix/xow/issues/74#issuecomment-633491834, xow crashes and kills my USB stack and only can be recovered by a reboot. I was able to pin down commit https://github.com/medusalix/xow/commit/7f755b5b4a8d70b9564326b81ce5d25f3e4cefbf as the offender. Any commit before that doesn't crash even after several hours of running.

xow log:
```
Mai 25 20:33:40 NoXP-PC systemd[1]: Started Xbox One Wireless Dongle Driver.
Mai 25 20:33:40 NoXP-PC xow[321318]: 2020-05-25 20:33:40 INFO  - xow v0.4-15-g7f755b5 ©Severin v. W.
Mai 25 20:33:40 NoXP-PC xow[321318]: 2020-05-25 20:33:40 INFO  - Waiting for device...
Mai 25 20:33:40 NoXP-PC xow[321318]: 2020-05-25 20:33:40 INFO  - Wireless address: 62:45:b4:fe:81:16
Mai 25 20:33:42 NoXP-PC xow[321318]: 2020-05-25 20:33:42 INFO  - Controller '1' connected
Mai 25 20:33:43 NoXP-PC xow[321318]: 2020-05-25 20:33:43 INFO  - Device announced, product id: 02ea
Mai 25 20:33:43 NoXP-PC xow[321318]: 2020-05-25 20:33:43 INFO  - Serial number: 02600108383810
Mai 25 21:03:26 NoXP-PC xow[321318]: 2020-05-25 21:03:26 ERROR - Error in bulk read: LIBUSB_ERROR_NO_DEVICE
Mai 25 21:03:26 NoXP-PC xow[321318]: 2020-05-25 21:03:26 ERROR - Error in bulk read: LIBUSB_ERROR_NO_DEVICE
Mai 25 21:03:26 NoXP-PC xow[321318]: 2020-05-25 21:03:26 ERROR - Shutting down due to error...
Mai 25 21:03:26 NoXP-PC xow[321318]: 2020-05-25 21:03:26 ERROR - Error in bulk write: LIBUSB_ERROR_NO_DEVICE
Mai 25 21:03:26 NoXP-PC xow[321318]: 2020-05-25 21:03:26 ERROR - Failed to write command
Mai 25 21:03:26 NoXP-PC xow[321318]: 2020-05-25 21:03:26 ERROR - Failed to send controller packet
Mai 25 21:03:26 NoXP-PC xow[321318]: 2020-05-25 21:03:26 ERROR - Failed to turn off controller
Mai 25 21:03:26 NoXP-PC xow[321318]: 2020-05-25 21:03:26 ERROR - Error in bulk write: LIBUSB_ERROR_NO_DEVICE
Mai 25 21:03:26 NoXP-PC xow[321318]: 2020-05-25 21:03:26 ERROR - Failed to write command
Mai 25 21:03:26 NoXP-PC xow[321318]: 2020-05-25 21:03:26 ERROR - Failed to set LED mode
Mai 25 21:03:26 NoXP-PC xow[321318]: 2020-05-25 21:03:26 ERROR - Failed to turn off LED
Mai 25 21:03:26 NoXP-PC xow[321318]: 2020-05-25 21:03:26 ERROR - Error in bulk write: LIBUSB_ERROR_NO_DEVICE
Mai 25 21:03:26 NoXP-PC xow[321318]: 2020-05-25 21:03:26 ERROR - Failed to write command
Mai 25 21:03:26 NoXP-PC xow[321318]: 2020-05-25 21:03:26 ERROR - Failed to set power mode
Mai 25 21:03:26 NoXP-PC xow[321318]: 2020-05-25 21:03:26 ERROR - Failed to turn off radio
Mai 25 21:03:26 NoXP-PC systemd[1]: xow.service: Succeeded.
Mai 25 21:03:27 NoXP-PC systemd[1]: xow.service: Scheduled restart job, restart counter is at 1.
Mai 25 21:03:27 NoXP-PC systemd[1]: Stopped Xbox One Wireless Dongle Driver.
Mai 25 21:03:27 NoXP-PC systemd[1]: Started Xbox One Wireless Dongle Driver.
Mai 25 21:03:27 NoXP-PC xow[351409]: 2020-05-25 21:03:27 INFO  - xow v0.4-15-g7f755b5 ©Severin v. W.
Mai 25 21:03:27 NoXP-PC xow[351409]: 2020-05-25 21:03:27 INFO  - Waiting for device...
```
dmesg:
```
Mai 25 21:03:21 NoXP-PC kernel: xhci_hcd 0000:25:00.0: ERROR unknown event type 15
Mai 25 21:03:26 NoXP-PC kernel: xhci_hcd 0000:25:00.0: xHCI host not responding to stop endpoint command.
Mai 25 21:03:26 NoXP-PC kernel: xhci_hcd 0000:25:00.0: xHCI host controller not responding, assume dead
Mai 25 21:03:26 NoXP-PC kernel: xhci_hcd 0000:25:00.0: HC died; cleaning up
Mai 25 21:03:26 NoXP-PC kernel: usb 3-1: USB disconnect, device number 2
Mai 25 21:03:26 NoXP-PC kernel: usb 3-1.3: USB disconnect, device number 4
Mai 25 21:03:26 NoXP-PC kernel: usb 4-1: USB disconnect, device number 2
Mai 25 21:03:26 NoXP-PC kernel: usb 3-2: USB disconnect, device number 3
```

medusalix, 2020-05-25:
I think it's very interesting that the `ERROR unknown event type 15` from `xhci_hcd` seems to happen before xow gets the `LIBUSB_ERROR_NO_DEVICE` from `libusb`. I'll have to investigate...

medusalix, 2020-05-26:
OK, there are multiple things worth checking out (individually):

- Plugging the dongle into a USB 2.0 port or disabling xHCI in the BIOS/UEFI to make all USB 3.0 ports fall back to 2.0
- Completely disabling USB autosuspend (see [this](https://unix.stackexchange.com/a/175035), should also work on Manjaro)
- Enabling libusb debug logging for xow (set environment variable `LIBUSB_DEBUG=4`). Might give some clues.
- Setting `USB_TIMEOUT_READ` in `usb.cpp` to zero (use `kill -9` to terminate xow, stop won't work)

Possibly related issues:

- Ubuntu bug report from 2014 that's still ongoing ([link](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1313279))
- Valve Index bug report for a very similar issue ([link](https://github.com/ValveSoftware/SteamVR-for-Linux/issues/222))
- libfreenect2 bug report ([link](https://github.com/OpenKinect/libfreenect2/issues/216)). *Very* similar to this issue.

Are you using an ASMedia USB controller?

NoXPhasma, 2020-05-26:
> Are you using an ASMedia USB controller?

On [my board](https://www.asus.com/us/Motherboards/PRIME-X370-PRO/specifications/) is an ASMedia USB 3.1 controller and the USB hub on which the dongle is connected, is on that 3.1 port. I will try to use a different USB 3 port (non ASMedia). I have no USB 2 ports on my motherboard, only on the front panel, which I would like not to use. Disabling USB 3.0 is no option as I need that for other devices.

I can also try to disable the autosuspend on Manjaro.

NoXPhasma, 2020-05-26:
Using a different USB 3 port _seems_ to fix it. In 1.5 hours of constant connection there was no crash yet. I will try more and report back tomorrow.

NoXPhasma, 2020-05-27:
Since I use a different USB 3 port I couldn't reproduce that issue anymore. I would say it's fixed on my end. But still wonder why it suddenly started to happen with that mentioned commit.

medusalix, 2020-05-27:
Yeah, me too. Strangely enough, the `libfreenect2` project had the exact same issue with ASMedia controllers, although the Kinect actually requires USB 3.0. It may be related to some unusual thing both projects do with regards to USB devices which triggers some edge-case in the ASMedia driver? Like the timeout I mentioned earlier?

NoXPhasma, 2020-05-27:
So it's most probably a driver issue and not xow. I'll close the issue then and if it ever happened again, on a non ASMedia controller, I'll answer to it.

closed 2020-05-27 by NoXPhasma

idallen, 2022-01-19:
I have a similar problem with an ASMedia device.

    Ubuntu 20.04.3 LTS
    Linux idallen-oak 5.13.0-27-generic #29~20.04.1-Ubuntu SMP Fri Jan 14 00:32:30 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

I have a StarTech SDOCK4U33 4-slot USB 3 hard drive dock using ASMedia Technology:

    Bus 004 Device 007: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
    Bus 004 Device 008: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
    Bus 004 Device 009: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
    Bus 004 Device 010: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge

If I reboot my Linux machine, the dock causes these USB errors during the boot:

    Jan 18 21:56:52 idallen-oak kernel: [   13.254357] usb 4-3: device descriptor read/8, error -110
    Jan 18 21:57:03 idallen-oak kernel: [   24.398290] xhci_hcd 0000:00:14.0: Abort failed to stop command ring: -110
    Jan 18 21:57:03 idallen-oak kernel: [   24.400554] xhci_hcd 0000:00:14.0: xHCI host controller not responding, assume dead
    Jan 18 21:57:03 idallen-oak kernel: [   24.402820] xhci_hcd 0000:00:14.0: HC died; cleaning up

Losing that controller at boot means many of my other USB devices are
not detected.

If I unplug the hard drive dock USB cable, the machine boots fine.  If I
then plug in the dock USB cable after the machine is up and running,
the above error happens (and I again lose many other USB devices).
Surely plugging in a USB device should not cause the whole controller
to fail and lose all the other working USB devices?

If I power off all of the drives in the hard drive dock and then
power them on, that seems to reset the dock and then everything works
fine again when I reboot.  The dock is correctly detected at boot and
everything works.  If I then reboot with the dock running, the problem
returns until I power cycle the dock and reboot again.

There is something about this dock that xhci_hcd doesn't like when the
dock is left powered on across a reboot.

More details (syslog, etc.) upon request.