medusalix/xow #77: controller not in `libinput list-devices` and no libinput udev "device added" events.

ekpyron, 2020-05-28:
While ``libinput record`` shows the controller correctly as:
```
/dev/input/event19:     Xbox One Wireless Controller
```
``libinput list-devices`` does not list the device.

Generally, apparently a libinput udev context will not receive ``LIBINPUT_EVENT_DEVICE_ADDED`` events for the controller.

I'm not sure if this is an udev or libinput issue - any ideas why this is the case?

medusalix, 2020-05-28:
According to [this](https://gitlab.freedesktop.org/libinput/libinput/-/issues/255) issue, it doesn't seem like gamepads are supported by `libinput`.

ekpyron, 2020-05-28:
Ok, thank you - I just read through the source code of ``libinput record`` and apparently it doesn't actually only use ``libinput``, but falls back to ``libevdev``, which apparently does support gamepads (since ``libinput record`` does show proper controller events and they have to be coming from somewhere).
I actually intended to use a libinput udev context to watch for a gamepad to be hotplugged, but it seems like I'll have to figure out another way to do that.
In any case I'm closing this.

closed 2020-05-28 by ekpyron