medusalix/xow #79: Raspbian Buster: When running as a systemd service, a permission denied error occurs

laurinneff, 2020-05-29:
Hardware:
Raspberry Pi 3 Model B v1.2
Raspbian Buster
Controller model 1708

When I'm running xow manually as root, xow works perfectly fine, however if I'm running it with the systemd unit, I get this error whenever I connect a controller:
```
May 29 23:58:09 raspberrypi systemd[1]: Started Xbox One Wireless Dongle Driver.
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 INFO  - xow v0.4-54-g4993ddf ©Severin v. W.
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 INFO  - Waiting for device...
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - Opening device...
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - Firmware already loaded, resetting...
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - Firmware loaded
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - ASIC version: 7632
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - MAC version: 7662
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - Chip id: 7613
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 INFO  - Wireless address: 62:45:b4:fe:04:08
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - Channel 1, power: 31
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - Channel 6, power: 32
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - Channel 11, power: 33
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - Channel 36, power: 43
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - Channel 40, power: 43
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - Channel 44, power: 45
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - Channel 48, power: 45
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - Channel 149, power: 39
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - Channel 153, power: 39
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - Channel 157, power: 40
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - Channel 161, power: 40
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 DEBUG - Channel 165, power: 40
May 29 23:58:09 raspberrypi xow[1637]: 2020-05-29 23:58:09 INFO  - Dongle initialized
May 29 23:58:15 raspberrypi xow[1637]: terminate called after throwing an instance of 'InputException'
May 29 23:58:15 raspberrypi xow[1637]:   what():  Error opening device: Permission denied
May 29 23:58:15 raspberrypi systemd[1]: xow.service: Main process exited, code=killed, status=6/ABRT
May 29 23:58:15 raspberrypi systemd[1]: xow.service: Failed with result 'signal'.
```

medusalix, 2020-05-30:
Sounds similar to issue #62. Either way, you'll have to figure out why the `udev` rules aren't applied as noted in the README.

laurinneff, 2020-05-30:
Thanks, that was the same issue as I had. If anyone finds this, in /lib/udev/rules.d/56-steamlink.rules, I changed both instances of 0660 to 0666

closed 2020-05-30 by laurinneff