medusalix/xow #82: Controllers Disconnecting When xow Run as Service

HailstoneRyan, 2020-06-05:
Just installed xow onto my new Raspberry pi and couldn't get my controllers to stay connected for more than 5 seconds. I rebuilt using debug and ran it from the command line as opposed to as a service and voila, my controllers remained connected. Logs of when it worked below. I'm novice when it comes to Linux so not sure where I can get the logs from xow when its run as service on a pi.

`2020-06-04 23:49:23 INFO  - xow v0.5 ©Severin v. W.
2020-06-04 23:49:23 INFO  - Waiting for device...
2020-06-04 23:49:23 DEBUG - Opening device...
2020-06-04 23:49:24 DEBUG - Firmware loaded
2020-06-04 23:49:24 DEBUG - ASIC version: 7632
2020-06-04 23:49:24 DEBUG - MAC version: 7662
2020-06-04 23:49:24 DEBUG - Chip id: 7612
2020-06-04 23:49:24 INFO  - Wireless address: 62:45:b4:e6:fc:60
2020-06-04 23:49:24 DEBUG - Channel 1, power: 31
2020-06-04 23:49:24 DEBUG - Channel 6, power: 33
2020-06-04 23:49:24 DEBUG - Channel 11, power: 34
2020-06-04 23:49:24 DEBUG - Channel 36, power: 43
2020-06-04 23:49:24 DEBUG - Channel 40, power: 43
2020-06-04 23:49:24 DEBUG - Channel 44, power: 43
2020-06-04 23:49:24 DEBUG - Channel 48, power: 43
2020-06-04 23:49:24 DEBUG - Channel 149, power: 38
2020-06-04 23:49:24 DEBUG - Channel 153, power: 38
2020-06-04 23:49:24 DEBUG - Channel 157, power: 38
2020-06-04 23:49:24 DEBUG - Channel 161, power: 38
2020-06-04 23:49:24 DEBUG - Channel 165, power: 38
2020-06-04 23:49:24 INFO  - Dongle initialized
2020-06-04 23:49:35 INFO  - Controller '1' connected
2020-06-04 23:49:36 INFO  - Device announced, product id: 02ea
2020-06-04 23:49:36 DEBUG - Firmware version: 3.1.1215.0
2020-06-04 23:49:36 DEBUG - Hardware version: 1284.1.1.1
2020-06-04 23:49:36 INFO  - Serial number: 02600031272638
2020-06-04 23:49:36 INFO  - Battery level: 2
2020-06-04 23:50:17 INFO  - Controller '1' disconnected
2020-06-04 23:50:24 INFO  - Controller '1' connected
2020-06-04 23:50:24 INFO  - Device announced, product id: 02e3
2020-06-04 23:50:24 DEBUG - Firmware version: 2.3.2388.0
2020-06-04 23:50:24 DEBUG - Hardware version: 3.1.1.1
2020-06-04 23:50:24 INFO  - Serial number: 02680073733641
2020-06-04 23:50:26 INFO  - Battery level: 2
2020-06-04 23:51:08 INFO  - Controller '1' disconnected
^C2020-06-04 23:51:12 INFO  - Shutting down...
2020-06-04 23:51:13 DEBUG - Closing device...`

medusalix, 2020-06-05:
Did you reboot your system after installing xow? You can get the service's logs by running `sudo journalctl -b -u xow`.

HailstoneRyan, 2020-06-05:
Yep, rebooted after installing.

Here's the logs from it running as a service.

> Jun 05 12:33:42 raspberrypi systemd[1]: Started Xbox One Wireless Dongle Driver.
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 INFO  - xow v0.5 ©Severin v. W.
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 INFO  - Waiting for device...
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - Opening device...
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - Firmware loaded
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - Invalid MAC address, correcting...
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - ASIC version: 7632
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - MAC version: 7662
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - Chip id: 7613
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 INFO  - Wireless address: 62:45:bd:01:cb:86
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - Channel 1, power: 28
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - Channel 6, power: 29
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - Channel 11, power: 30
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - Channel 36, power: 43
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - Channel 40, power: 43
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - Channel 44, power: 43
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - Channel 48, power: 43
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - Channel 149, power: 40
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - Channel 153, power: 40
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - Channel 157, power: 40
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - Channel 161, power: 40
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 DEBUG - Channel 165, power: 40
Jun 05 12:33:47 raspberrypi xow[346]: 2020-06-05 12:33:47 INFO  - Dongle initialized
Jun 05 12:34:41 raspberrypi xow[346]: terminate called after throwing an instance of 'InputException'
Jun 05 12:34:41 raspberrypi xow[346]:   what():  Error opening device: Permission denied

medusalix, 2020-06-05:
Ah, you're probably running into a known issue with the built-in `udev` rules. [This](https://github.com/medusalix/xow/issues/62#issuecomment-622460953) comment will help.

HailstoneRyan, 2020-06-05:
Yes, that did it!

closed 2020-06-05 by HailstoneRyan