medusalix/xow #85: Controller disconnects after a few seconds

elmehalawi, 2020-06-08:
I installed xow on my Fedora 32 computer, and the controller connects and I can actually use it for a few seconds before it disconnects. Unfortunately, I don't really see much useful output. Here are the logs when I run xow:

```
$ xow
2020-06-08 16:30:32 INFO  - xow v0.5-2-g73f5273 ©Severin v. W.
2020-06-08 16:30:32 INFO  - Waiting for device...
2020-06-08 16:30:33 INFO  - Wireless address: 62:45:b5:11:38:c1
2020-06-08 16:30:33 INFO  - Dongle initialized
2020-06-08 16:30:41 INFO  - Controller '1' connected
2020-06-08 16:30:41 INFO  - Device announced, product id: 02ea
2020-06-08 16:30:41 INFO  - Serial number: 02600426762639
2020-06-08 16:30:42 INFO  - Battery level: 2
# The following is output when I Ctrl+C in the terminal:
^C2020-06-08 16:32:32 INFO  - Shutting down...
2020-06-08 16:32:33 ERROR - Error in bulk write: LIBUSB_ERROR_TIMEOUT
2020-06-08 16:32:33 ERROR - Failed to write command
2020-06-08 16:32:33 ERROR - Failed to send controller packet
2020-06-08 16:32:33 ERROR - Failed to turn off controller
2020-06-08 16:32:34 ERROR - Error in bulk write: LIBUSB_ERROR_TIMEOUT
2020-06-08 16:32:34 ERROR - Failed to write command
2020-06-08 16:32:34 ERROR - Failed to set LED mode
2020-06-08 16:32:34 ERROR - Failed to turn off LED
2020-06-08 16:32:35 ERROR - Error in bulk write: LIBUSB_ERROR_TIMEOUT
2020-06-08 16:32:35 ERROR - Failed to write command
2020-06-08 16:32:35 ERROR - Failed to set power mode
2020-06-08 16:32:35 ERROR - Failed to turn off radio

```

I built xow with `make BUILD=DEBUG` to try to get more logs. I had also tried `RELEASE` earlier with the same result.

BTW, amazing project! Last time I tried to use my xbox dongle on Linux, it was completely impossible, so this is impressive progress.

elmehalawi, 2020-06-08:
Okay this is very strange. When I tried a different USB controller (the USB ports in the back of the computer), it worked perfectly. I was able to play a few games of RL and had no issues with the controller. Thanks again for this great project!

closed 2020-06-08 by elmehalawi

medusalix, 2020-06-09:
Was that an `ASMedia` USB controller you originally tried the dongle with? I've heard that those can cause problems with xow.

elmehalawi, 2020-06-09:
Yes it was. When I run `lspci | grep USB`:

```
00:14.0 USB controller: Intel Corporation 100 Series/C230 Series Chipset Family USB 3.0 xHCI Controller (rev 31)
04:00.0 USB controller: ASMedia Technology Inc. ASM1142 USB 3.1 Host Controller
```

So I believe that the first one listed there is the one on my motherboard. I believe that is the one that it's connected to now, and it works fine.

medusalix, 2020-06-10:
Alright, I've added a `USB incompatibilities` section to the README that points out these issues.