medusalix/xow #86: Controller loss connection after some seconds and xow freeze

dsiggi, 2020-06-11:
Hi,
at first some informations about my system and my hardware.

System: Gentoo Linux
Kernel: 5.4.38-gentoo

Dongle:
2020-06-11 22:28:04 DEBUG - ASIC version: 7632
2020-06-11 22:28:04 DEBUG - MAC version: 7662
2020-06-11 22:28:04 DEBUG - Chip id: 7612

Controller:
2020-06-11 22:28:12 INFO  - Device announced, product id: 02d1
2020-06-11 22:28:12 DEBUG - Firmware version: 2.3.2381.0
2020-06-11 22:28:12 DEBUG - Hardware version: 1.1.1.1

What did I do?
I startet xow on an terminal and i have this output:

```
2020-06-11 22:27:57 INFO  - xow  ©Severin v. W.
2020-06-11 22:27:57 INFO  - Waiting for device...
2020-06-11 22:28:04 DEBUG - Opening device...
2020-06-11 22:28:04 DEBUG - Firmware loaded
2020-06-11 22:28:04 DEBUG - ASIC version: 7632
2020-06-11 22:28:04 DEBUG - MAC version: 7662
2020-06-11 22:28:04 DEBUG - Chip id: 7612
2020-06-11 22:28:04 INFO  - Wireless address: 62:45:b4:ec:48:30
2020-06-11 22:28:04 DEBUG - Channel 1, power: 35
2020-06-11 22:28:04 DEBUG - Channel 6, power: 37
2020-06-11 22:28:04 DEBUG - Channel 11, power: 37
2020-06-11 22:28:04 DEBUG - Channel 36, power: 44
2020-06-11 22:28:04 DEBUG - Channel 40, power: 44
2020-06-11 22:28:04 DEBUG - Channel 44, power: 43
2020-06-11 22:28:04 DEBUG - Channel 48, power: 43
2020-06-11 22:28:04 DEBUG - Channel 149, power: 39
2020-06-11 22:28:04 DEBUG - Channel 153, power: 39
2020-06-11 22:28:04 DEBUG - Channel 157, power: 39
2020-06-11 22:28:04 DEBUG - Channel 161, power: 39
2020-06-11 22:28:04 DEBUG - Channel 165, power: 39
2020-06-11 22:28:04 INFO  - Dongle initialized
```

Than I switch on the controller

```
2020-06-11 22:28:11 INFO  - Controller '1' connected
2020-06-11 22:28:12 INFO  - Device announced, product id: 02d1
2020-06-11 22:28:12 DEBUG - Firmware version: 2.3.2381.0
2020-06-11 22:28:12 DEBUG - Hardware version: 1.1.1.1
2020-06-11 22:28:13 INFO  - Battery level: 2
```

After some seconds the the controller loss the connection. The xbox logo begins to blink.
No output on the terminal. Now I can disconnect the dongle from the usb port and no output on the terminal. The last line from xow is always the Battery level.

First when I reconnect the dongle to my computer xow is crashing with the following output.

```
2020-06-11 22:36:01 ERROR - Error in bulk read: LIBUSB_ERROR_NO_DEVICE
2020-06-11 22:36:01 INFO  - Shutting down...
2020-06-11 22:36:01 ERROR - Error in bulk read: LIBUSB_ERROR_NO_DEVICE
2020-06-11 22:36:01 ERROR - Error in bulk write: LIBUSB_ERROR_NO_DEVICE
2020-06-11 22:36:01 ERROR - Failed to write command
2020-06-11 22:36:01 ERROR - Failed to send controller packet
2020-06-11 22:36:01 ERROR - Failed to turn off controller
2020-06-11 22:36:01 ERROR - Error in bulk write: LIBUSB_ERROR_NO_DEVICE
2020-06-11 22:36:01 ERROR - Failed to write command
2020-06-11 22:36:01 ERROR - Failed to set LED mode
2020-06-11 22:36:01 ERROR - Failed to turn off LED
2020-06-11 22:36:01 ERROR - Error in bulk write: LIBUSB_ERROR_NO_DEVICE
2020-06-11 22:36:01 ERROR - Failed to write command
2020-06-11 22:36:01 ERROR - Failed to set power mode
2020-06-11 22:36:01 ERROR - Failed to turn off radio
2020-06-11 22:36:01 DEBUG - Closing device...
2020-06-11 22:36:01 ERROR - Error releasing interface: LIBUSB_ERROR_NO_DEVICE
```

I hope you can work with this informations.

medusalix, 2020-06-12:
Sounds very similar to issue #85. The README includes a `USB incompatibilities` section that might help.

dsiggi, 2020-06-12:
Hi,
I have an intel chip.

```
00:14.0 USB controller: Intel Corporation 100 Series/C230 Series Chipset Family USB 3.0 xHCI Controller (rev 31)
```
I'm using here an Lenovo Y700 laptop. This laptop has 3 usb ports.

1x black on the left side
2x red on the right side

I think the black one is usb 2 and the red ones are usb 3.
When I connect the dongle to the black on it's working. When I'm using the red ones I have the error described below.

medusalix, 2020-06-12:
Can you post the output of `sudo lspci -vvv` and `sudo dmesg` after running xow?

dsiggi, 2020-06-12:
Here is the output of lspci -vvv
```
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Host Bridge/DRAM Registers (rev 07)
	Subsystem: Lenovo Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Host Bridge/DRAM Registers
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B+ ParErr- DEVSEL=fast >TAbort- <TAbort+ <MAbort+ >SERR- <PERR- INTx-
	Latency: 0
	Capabilities: <access denied>
	Kernel driver in use: skl_uncore

00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) (rev 07) (prog-if 00 [Normal decode])
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 122
	Bus: primary=00, secondary=01, subordinate=06, sec-latency=0
	I/O behind bridge: 00004000-00004fff [size=4K]
	Memory behind bridge: 93000000-93ffffff [size=16M]
	Prefetchable memory behind bridge: 0000000080000000-0000000091ffffff [size=288M]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ <SERR- <PERR-
	BridgeCtl: Parity- SERR+ NoISA- VGA- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: <access denied>
	Kernel driver in use: pcieport

00:02.0 VGA compatible controller: Intel Corporation HD Graphics 530 (rev 06) (prog-if 00 [VGA controller])
	Subsystem: Lenovo HD Graphics 530
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 126
	Region 0: Memory at 92000000 (64-bit, non-prefetchable) [size=16M]
	Region 2: Memory at a0000000 (64-bit, prefetchable) [size=256M]
	Region 4: I/O ports at 5000 [size=64]
	[virtual] Expansion ROM at 000c0000 [disabled] [size=128K]
	Capabilities: <access denied>
	Kernel driver in use: i915

00:14.0 USB controller: Intel Corporation 100 Series/C230 Series Chipset Family USB 3.0 xHCI Controller (rev 31) (prog-if 30 [XHCI])
	Subsystem: Lenovo 100 Series/C230 Series Chipset Family USB 3.0 xHCI Controller
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B+ ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 129
	Region 0: Memory at 94300000 (64-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: xhci_hcd

00:16.0 Communication controller: Intel Corporation 100 Series/C230 Series Chipset Family MEI Controller #1 (rev 31)
	Subsystem: Lenovo 100 Series/C230 Series Chipset Family MEI Controller
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 255
	Region 0: Memory at 9432a000 (64-bit, non-prefetchable) [disabled] [size=4K]
	Capabilities: <access denied>

00:17.0 SATA controller: Intel Corporation HM170/QM170 Chipset SATA Controller [AHCI Mode] (rev 31) (prog-if 01 [AHCI 1.0])
	Subsystem: Lenovo HM170/QM170 Chipset SATA Controller [AHCI Mode]
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz+ UDF- FastB2B+ ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 127
	Region 0: Memory at 94328000 (32-bit, non-prefetchable) [size=8K]
	Region 1: Memory at 9432d000 (32-bit, non-prefetchable) [size=256]
	Region 2: I/O ports at 5080 [size=8]
	Region 3: I/O ports at 5088 [size=4]
	Region 4: I/O ports at 5060 [size=32]
	Region 5: Memory at 9432b000 (32-bit, non-prefetchable) [size=2K]
	Capabilities: <access denied>
	Kernel driver in use: ahci
[... 113 lines cut ...]

09:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
	Subsystem: Lenovo RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 19
	Region 0: I/O ports at 3000 [size=256]
	Region 2: Memory at 94004000 (64-bit, non-prefetchable) [size=4K]
	Region 4: Memory at 94000000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: <access denied>
	Kernel driver in use: r8169
```

And here from dmesg when using the red (not working) usb port
```
[ 9019.931851] usb 1-2: new high-speed USB device number 13 using xhci_hcd
[ 9020.073427] usb 1-2: New USB device found, idVendor=045e, idProduct=02e6, bcdDevice= 1.00
[ 9020.073433] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9020.073442] usb 1-2: Product: XBOX ACC
[ 9020.073446] usb 1-2: Manufacturer: Microsoft Inc.
[ 9020.073449] usb 1-2: SerialNumber: 527058
[ 9021.632262] usb 1-2: reset high-speed USB device number 13 using xhci_hcd
[ 9025.297757] input: Xbox One Wireless Controller as /devices/virtual/input/input27
```

medusalix, 2020-06-13:
Hmm, very strange. It doesn't look like an issue with the kernel.
Running xow with the environment variable `LIBUSB_DEBUG=4` might give some insights (it should log a bunch of `libusb` related things).

dsiggi, 2020-06-14:
Hi,
the log is extreme long so I uploaded it as file.

At ~26 seconds the controller loss the connection.
At ~30 seconds I disconnect the dongle.
At ~70 seconds I kill xow bis CTRL+C.

[Logfile](https://github.com/medusalix/xow/files/4775969/xow.txt)

medusalix, 2020-06-14:
Thanks for the logs.
It appears like the USB connection to the dongle times out starting at around 24.83 seconds. Oddly enough, the removal of the dongle doesn't seem to get detected by `libusb`.
I've noticed that you're using `libusb` v1.0.21 from 2016. Could you try updating to the latest version (should be v1.0.23)?

dsiggi, 2020-06-14:
v1.0.21 is the last stable version in gentoo. I installed now the unstable version v1.0.23-r1 but with no difference. 

I used the dongle now over 4 hours without an error on the black usb port. On this port xow works like a charm.

medusalix, 2020-06-14:
Hmm, I guess we should either report this to `libusb` or leave it as it is. I don't believe it's caused by a bug in xow.

dsiggi, 2020-06-15:
I also think it's not a problem with xow. I didn't now as much about github. Can you "link" this issue to libusb?

medusalix, 2020-06-15:
You'll have to open an issue on their [GitHub page](https://github.com/libusb/libusb). You can link to this issue from there.

closed 2020-06-17 by medusalix