medusalix/xow #88: Error setting configuration: LIBUSB_ERROR_BUSY

bterrier, 2020-06-16:
I am on ArchLinux, when I connect the wireless adapter the xow crashes with:
```
Started Xbox One Wireless Dongle Driver.
INFO  - xow v0.5-5-gf693ee3 ©Severin v. W.
INFO  - Waiting for device...
terminate called after throwing an instance of 'UsbException'
what():  Error setting configuration: LIBUSB_ERROR_BUSY
systemd[1]: xow.service: Main process exited, code=dumped, status=6/ABRT
systemd[1]: xow.service: Failed with result 'core-dump'.
```

medusalix, 2020-06-17:
Make sure that no other program is currently communicating with the dongle. If you're running a second instance of xow or the `make install` target didn't copy the necessary files to their respective locations, you'll run into issues.

bterrier, 2020-06-18:
I jut retried to get dmesg logs and this time it worked.
I have not changed anything, but maybe I did have another process communicating with the dongle.

tristianc, 2020-06-19:
# xow issues

If I use both an XBox One dongle and another MT76x2u device is present, xow will fail to run because it will fail to reset the dongle. The reset failure doesn't occur for me if I happen to use another wireless dongle, or none at all.

I believe commit 40e1e0d2b684cac86cff6ec2ab02dbdfacdb8cbf is causing this for me.

Just to confirm, there is only one instance of `xow` running, managed via the `xow.service` unit.

# To Reproduce

1. Insert another MT76x2u USB WiFi dongle, but do not insert the XBox dongle yet; it is easier to log if the dongle is inserted afterwards.

2. Run `journalctl -fu xow` in one terminal and `dmesg --follow` in another. The following journalctl output should appear:

```
Jun 19 00:51:43 sanji systemd[1]: Started Xbox One Wireless Dongle Driver.
Jun 19 00:51:43 sanji xow[6928]: 2020-06-19 00:51:43 INFO  - xow v0.5 ©Severin v. W.
Jun 19 00:51:43 sanji xow[6928]: 2020-06-19 00:51:43 INFO  - Waiting for device...
```

3. Insert the XBox One wireless dongle.  `journalctl -fu xow` reports:

```
Jun 19 00:52:42 sanji xow[6928]: terminate called after throwing an instance of 'UsbException'
Jun 19 00:52:42 sanji xow[6928]:   what():  Error setting configuration: LIBUSB_ERROR_BUSY
Jun 19 00:52:42 sanji systemd[1]: xow.service: Main process exited, code=dumped, status=6/ABRT
Jun 19 00:52:42 sanji systemd[1]: xow.service: Failed with result 'core-dump'.
```

At the same time, while I have `dmesg -w` running, dmesg reports:

```
 1272.990915] usb 1-1.4: new high-speed USB device number 6 using ehci-pci
[ 1273.070826] usb 1-1.4: New USB device found, idVendor=045e, idProduct=02fe, bcdDevice= 1.00
[ 1273.070829] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1273.070831] usb 1-1.4: Product: XBOX ACC
[ 1273.070833] usb 1-1.4: Manufacturer: Microsoft Inc.
[ 1273.070834] usb 1-1.4: SerialNumber: 480672
[ 1273.137898] usb 1-1.4: reset high-speed USB device number 6 using ehci-pci
[ 1273.217748] mt76x2u 1-1.4:1.0: ASIC revision: 76320044
[ 1273.255794] mt76x2u 1-1.4:1.0: ROM patch build: 20141115060606a
[ 1273.382572] mt76x2u 1-1.4:1.0: Firmware Version: 0.0.00
[ 1273.382575] mt76x2u 1-1.4:1.0: Build: 1
[ 1273.382576] mt76x2u 1-1.4:1.0: Build Time: 201507311614____
[ 1274.043944] mt76x2u 1-1.4:1.0: error: MCU resp evt:9 seq:1-0
[ 1274.044066] mt76x2u 1-1.4:1.0: error: MCU resp evt:9 seq:1-0
[ 1274.044190] mt76x2u 1-1.4:1.0: error: MCU resp evt:9 seq:1-0
[ 1274.572475] ieee80211 phy7: Selected rate control algorithm 'minstrel_ht'
[ 1275.106894] usb 1-1.4: reset high-speed USB device number 6 using ehci-pci
[ 1275.251822] usb 1-1.4: reset high-speed USB device number 6 using ehci-pci
[ 1275.331515] mt76x2u 1-1.4:1.0: ASIC revision: 76320044
[ 1275.356937] mt76x2u 1-1.4:1.0: ROM patch already applied
[ 1275.357016] mt76x2u 1-1.4:1.0: Firmware Version: 0.0.00
[ 1275.357017] mt76x2u 1-1.4:1.0: Build: 1
[ 1275.357019] mt76x2u 1-1.4:1.0: Build Time: 201507311614____
[ 1276.008435] mt76x2u 1-1.4:1.0: error: MCU resp evt:9 seq:1-0
[ 1276.008556] mt76x2u 1-1.4:1.0: error: MCU resp evt:9 seq:1-0
[ 1276.008680] mt76x2u 1-1.4:1.0: error: MCU resp evt:9 seq:1-0
[ 1276.538263] ieee80211 phy8: Selected rate control algorithm 'minstrel_ht'
[ 1276.539067] usb 1-1.4: usbfs: interface 0 claimed by mt76x2u while 'xow' sets config #1
[ 1276.596328] mt76x2u 1-1.4:1.0 wlp0s26u1u4: renamed from wlan0
```

4. Remove the dongle. The following is reported by `dmesg -w`:

```
[ 1454.967180] mt76u_complete_rx: 12 callbacks suppressed
[ 1454.967199] mt76x2u 1-1.4:1.0: rx urb failed: -71
[ 1454.971426] mt76x2u 1-1.4:1.0: rx urb failed: -71
[ 1454.975675] mt76x2u 1-1.4:1.0: rx urb failed: -71
[ 1454.979925] mt76x2u 1-1.4:1.0: rx urb failed: -71
[ 1454.984174] mt76x2u 1-1.4:1.0: rx urb failed: -71
[ 1454.988425] mt76x2u 1-1.4:1.0: rx urb failed: -71
[ 1454.992674] mt76x2u 1-1.4:1.0: rx urb failed: -71
[ 1454.996923] mt76x2u 1-1.4:1.0: rx urb failed: -71
[ 1455.001172] mt76x2u 1-1.4:1.0: rx urb failed: -71
[ 1455.005425] mt76x2u 1-1.4:1.0: rx urb failed: -71
[ 1455.090299] usb 1-1.4: USB disconnect, device number 6
[ 1455.100246] mt76x2u 1-1.4:1.0: mac specific condition occurred
```

which doesn't make sense to me because I thought mt76x2u shouldn't be bound to the XBox One dongle, per `/etc/modprobe.d/xow-blacklist.conf`.

medusalix, 2020-06-19:
I believe I know what's causing this to happen:

- xow's blacklist prevents the `mt76x2u` driver from loading when you insert an Xbox wireless dongle
- When you plug in a standard wireless dongle, the `mt76x2u` module gets loaded
- The Xbox wireless dongle gets associated to the `mt76x2u` driver, bypassing the blacklist because the module is already loaded

I don't think I can fix this. There are literally only these two options:

- Blacklist the whole `mt76x2u` driver
- Prevent the `mt76x2u` driver from loading *only* for Xbox wireless dongles

Both have their advantages/disadvantages but solving this issue completely is not possible (as far as I know).

closed 2020-06-21 by medusalix

tristianc, 2020-07-05:
## Recap

In xow v0.5-13-g9b99fe3, I can not use the XBox One dongle and an MT76 device at the same time. Attempts to start xow while both devices are connected to my machine will result in the following error:

```
Started Xbox One Wireless Dongle Driver.
INFO  - xow v0.5-5-gf693ee3 ©Severin v. W.
INFO  - Waiting for device...
terminate called after throwing an instance of 'UsbException'
what():  Error setting configuration: LIBUSB_ERROR_BUSY
systemd[1]: xow.service: Main process exited, code=dumped, status=6/ABRT
systemd[1]: xow.service: Failed with result 'core-dump'.
```

## Change

I'm trying to implement the "Prevent the mt76x2u driver from loading only for Xbox wireless dongles" option mentioned in https://github.com/medusalix/xow/issues/88#issuecomment-646561382

I added calls to `libusb_set_auto_detach_kernel_driver` and `libusb_detach_kernel_driver` to `UsbDevice::UsbDevice()`. While `xow` is running, if I run `usb-devices`, this is what is reported for the dongle:

```
T:  Bus=01 Lev=01 Prnt=01 Port=02 Cnt=02 Dev#=  6 Spd=480 MxCh= 0
D:  Ver= 2.01 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=045e ProdID=02fe Rev=01.00
S:  Manufacturer=Microsoft Inc.
S:  Product=XBOX ACC
S:  SerialNumber=480672
C:  #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=500mA
I:  If#=0x0 Alt= 0 #EPs= 8 Cls=ff(vend.) Sub=ff Prot=ff Driver=usbfs
```

## Consequence

The change ensures that the USB dongle is always claimed by xow, even when my MT76 dongle is connected. As opposed to core dumping after `libusb_set_configuration`, xow seems to always make it to `Mt76::loadFirmware()`.

My current problem is that I can only get `Mt76::loadFirmware()` to succeed in one circumstance:

1. The machine has booted without the dongle connected
2. xow has started
3. I've connected the dongle after xow has started

In this circumstance, xow will run successfully and the dongle will pair with controllers.

```
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Opening device...
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Enabled driver detachment on claim
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Detached kernel driver
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Firmware loaded
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Invalid MAC address, correcting...
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - ASIC version: 7632
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - MAC version: 7662
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Chip id: 7613
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 INFO  - Wireless address: 62:45:bd:00:49:b5
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Channel 1, power: 27
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Channel 6, power: 28
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Channel 11, power: 29
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Channel 36, power: 43
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Channel 40, power: 43
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Channel 44, power: 43
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Channel 48, power: 43
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Channel 149, power: 38
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Channel 153, power: 38
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Channel 157, power: 38
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Channel 161, power: 38
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 DEBUG - Channel 165, power: 38
Jul 04 10:42:38 oden xow[1194]: 2020-07-04 10:42:38 INFO  - Dongle initialized
Jul 04 10:43:50 oden xow[1194]: 2020-07-04 10:43:50 INFO  - Controller '1' connected
Jul 04 10:43:51 oden xow[1194]: 2020-07-04 10:43:51 INFO  - Device announced, product id: 02ea
Jul 04 10:43:51 oden xow[1194]: 2020-07-04 10:43:51 DEBUG - Firmware version: 3.1.1221.0
Jul 04 10:43:51 oden xow[1194]: 2020-07-04 10:43:51 DEBUG - Hardware version: 1284.1.1.1
[... 32 lines cut ...]
...
Jul 04 11:27:58 oden xow[8028]: 2020-07-04 11:27:58 DEBUG - Read value: 801140fa
Jul 04 11:27:58 oden xow[8028]: 2020-07-04 11:27:58 ERROR - Firmware reset timed out
Jul 04 11:27:58 oden xow[8028]: terminate called after throwing an instance of 'Mt76Exception'
Jul 04 11:27:58 oden xow[8028]:   what():  Failed to load firmware
Jul 04 11:27:58 oden systemd[1]: xow.service: Main process exited, code=dumped, status=6/ABRT
Jul 04 11:27:58 oden systemd[1]: xow.service: Failed with result 'core-dump'.
```

I don't know what these return values mean. My thought is that something else has to be added to `loadFirmware()` in order to perform a firmware reset in the circumstances I'm entering and get a `0x80000000` return value.

At one point, I decided to comment out the firmware reset success confirmation that follows:

```
        if (!successful)
        {
            Log::error("Firmware reset timed out");

            return false;
        }
```

and xow appeared as if it was functioning correctly, but it would not pair any controllers when the dongle and controller were in pairing mode.

If this issue isn't the right place for discussion, I can move this discussion to a different one.

medusalix, 2020-07-06:
I've read the `debugging.md` from your fork and I'd like to give some comments:

The `mt76x02u_mcu_fw_reset` function you've mentioned corresponds to this function call in xow:

```c++
 controlWrite(MT_FW_RESET_IVB, 0, MT_VEND_DEV_MODE);
```

The actual firmware loading/reset process for `mt76` is very similar to the one xow uses. Once the wrong firmware is loaded (by xow, `mt76` or any other driver, even Microsoft's), the firmware reset fails. This is why detaching the kernel driver from the dongle after the firmware has been loaded won't work. I've tried out all the options, none of them lead to success.
Your best bet is to recompile the `mt76` driver and remove the line for the dongle's USB ids. It's very simple to do and works reliably.