medusalix/xow #89: ld.lld fails to convert firmware.bin to firmware.o

bterrier, 2020-06-16:
I've tried to install xow on ArchLinux using https://aur.archlinux.org/packages/xow-git/

Using ld.lld as the linker the build fails:
```
ld -r -b binary -o firmware.o firmware.bin
ld: error: target emulation unknown: -m or at least one .o file required
```

Using ld.bfd works.

I am not sure if it is an issue with ld.lld or xow.

medusalix, 2020-06-17:
LLVM's `ld.lld` doesn't have a default value for the target emulation when linking binary files. So you have to pass your system architecture to `ld` using the `-m` option.
I can't provide workarounds for tools claiming to be a drop-in replacement for GNU's toolchain when they actually don't even behave in the same way GNU's software does.

bterrier, 2020-06-18:
That's fair enough.

closed 2020-06-18 by bterrier