medusalix/xow #90: Frequent disconnects on v0.5 and v0.4 | Not on v0.3

skrrrm, 2020-06-17:
I believe something has happened since v0.3 that makes my controllers disconnect frequently. I've tried v0.5 in several USB ports (USB 3.1 and USB 2.0) and I've had error messages like in this [issue](https://github.com/medusalix/xow/issues/85) and this [issue](https://github.com/medusalix/xow/issues/86). I've even had my whole USB stack die like in this [issue](https://github.com/medusalix/xow/issues/76). I assume the whole USB stack dying is because of the ASMedia controller.

I just downgraded to v0.3 and no issues for 30 minutes, where on v0.4 and v0.5 I'd had disconnects within 5 minutes.

Output of: `lspci | grep USB`

```
02:00.0 USB controller: Advanced Micro Devices, Inc. [AMD] X370 Series Chipset USB 3.1 xHCI Controller (rev 02)
07:00.0 USB controller: ASMedia Technology Inc. ASM1143 USB 3.1 Host Controller
0d:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) USB 3.0 Host Controller
```

medusalix, 2020-06-17:
Can you identify the exact commit that caused these issues? You might want to start with commit 78cdb31a735bcc0d1b5ce870585c44ce4b55c657.

skrrrm, 2020-06-17:
Already experiencing problems after the commit you mentioned. I'll leave a comment for each commit I go through. Since the logs are a bit spacious. I've removed some of the batter status updates.

Starting with the commit you specified:

- [x] **commit 78cdb31:** `VERIFIED` - No disconnects. `Error in bulk read: LIBUSB_ERROR_OTHER` when ctrl+C out of xow.`

skrrrm, 2020-06-17:
❌ **commit 3a1e5cb:** `FAILED` - did recieve disconnects, but not frequent ones. One time I had to restart the controller.

Output:
```
2020-06-17 21:03:22 INFO  - xow  ©Severin v. W.
2020-06-17 21:03:22 DEBUG - Opening device...
2020-06-17 21:03:23 INFO  - Dongle plugged in
2020-06-17 21:03:23 DEBUG - Firmware loaded
2020-06-17 21:03:23 DEBUG - ASIC version: 7632
2020-06-17 21:03:23 DEBUG - Chip id: 7612
2020-06-17 21:03:23 INFO  - Wireless address: 62:45:b4:f2:02:6f
2020-06-17 21:03:23 INFO  - Dongle initialized
2020-06-17 21:03:41 INFO  - Pairing initiated
2020-06-17 21:03:41 INFO  - Pairing initiated
2020-06-17 21:03:41 INFO  - Pairing initiated
2020-06-17 21:03:41 INFO  - Pairing initiated
2020-06-17 21:03:52 DEBUG - Client pairing: 7e:ed:80:a9:79:79
2020-06-17 21:03:52 DEBUG - Client associating: 7e:ed:80:a9:79:79
2020-06-17 21:03:52 INFO  - Controller '1' connected
2020-06-17 21:03:53 INFO  - Serial number: 02600046126702
2020-06-17 21:03:54 DEBUG - Battery type: 1, level: 2
2020-06-17 21:05:26 DEBUG - Client lost: 1
2020-06-17 21:05:26 INFO  - Controller '1' disconnected
2020-06-17 21:05:45 DEBUG - Client associating: 7e:ed:80:a9:79:79
2020-06-17 21:05:45 INFO  - Controller '1' connected
2020-06-17 21:05:45 INFO  - Serial number: 02600046126702
2020-06-17 21:10:32 DEBUG - Client lost: 1
2020-06-17 21:10:32 INFO  - Controller '1' disconnected
2020-06-17 21:10:32 ERROR - Packet for unconnected controller '1'
2020-06-17 21:10:32 ERROR - Packet for unconnected controller '1'
2020-06-17 21:10:32 ERROR - Packet for unconnected controller '1'
2020-06-17 21:10:32 ERROR - Packet for unconnected controller '1'
2020-06-17 21:10:32 ERROR - Packet for unconnected controller '1'
2020-06-17 21:10:32 ERROR - Packet for unconnected controller '1'
2020-06-17 21:11:11 ERROR - Packet for unconnected controller '1'
2020-06-17 21:11:11 ERROR - Packet for unconnected controller '1'
2020-06-17 21:11:11 ERROR - Packet for unconnected controller '1'
2020-06-17 21:11:11 ERROR - Packet for unconnected controller '1'
2020-06-17 21:11:11 ERROR - Packet for unconnected controller '1'
2020-06-17 21:11:11 ERROR - Packet for unconnected controller '1'
2020-06-17 21:11:11 ERROR - Packet for unconnected controller '1'
2020-06-17 21:11:11 ERROR - Packet for unconnected controller '1'
2020-06-17 21:11:17 ERROR - Packet for unconnected controller '1'
2020-06-17 21:11:17 ERROR - Packet for unconnected controller '1'
2020-06-17 21:11:17 ERROR - Packet for unconnected controller '1'
2020-06-17 21:11:17 ERROR - Packet for unconnected controller '1'
2020-06-17 21:11:17 ERROR - Packet for unconnected controller '1'
2020-06-17 21:11:17 ERROR - Packet for unconnected controller '1'
2020-06-17 21:11:22 DEBUG - Client associating: 7e:ed:80:a9:79:79
2020-06-17 21:11:22 INFO  - Controller '1' connected
2020-06-17 21:11:22 INFO  - Serial number: 02600046126702   # Played a bit after this reconnect.
```

skrrrm, 2020-06-17:
❌**commit dce7bcf:** `FAILED` - Same as previous commit but after restart of controller xow fails.
```
2020-06-17 21:23:12 INFO  - xow  ©Severin v. W.
2020-06-17 21:23:12 DEBUG - Opening device...
2020-06-17 21:23:12 INFO  - Dongle plugged in
2020-06-17 21:23:12 DEBUG - Firmware already loaded, resetting...
2020-06-17 21:23:12 DEBUG - Firmware loaded
2020-06-17 21:23:13 DEBUG - ASIC version: 7632
2020-06-17 21:23:13 DEBUG - Chip id: 7612
2020-06-17 21:23:13 INFO  - Wireless address: 62:45:b4:f2:02:6f
2020-06-17 21:23:13 INFO  - Dongle initialized
2020-06-17 21:23:27 INFO  - Pairing initiated
2020-06-17 21:23:27 INFO  - Pairing initiated
2020-06-17 21:23:28 INFO  - Pairing initiated
2020-06-17 21:23:28 INFO  - Pairing initiated
2020-06-17 21:23:32 DEBUG - Client pairing: 7e:ed:80:a9:79:79
2020-06-17 21:23:32 DEBUG - Client associating: 7e:ed:80:a9:79:79
2020-06-17 21:23:32 INFO  - Controller '1' connected
2020-06-17 21:23:33 INFO  - Serial number: 02600046126702
2020-06-17 21:23:33 DEBUG - Battery type: 1, level: 2
2020-06-17 21:32:04 DEBUG - Client lost: 1
2020-06-17 21:32:04 INFO  - Controller '1' disconnected
2020-06-17 21:32:08 ERROR - Packet for unconnected controller '1'
2020-06-17 21:32:08 ERROR - Packet for unconnected controller '1'
2020-06-17 21:32:08 ERROR - Packet for unconnected controller '1'
2020-06-17 21:32:08 ERROR - Packet for unconnected controller '1'
2020-06-17 21:32:08 ERROR - Packet for unconnected controller '1'
2020-06-17 21:32:08 ERROR - Packet for unconnected controller '1'
2020-06-17 21:32:36 ERROR - Packet for unconnected controller '1'
2020-06-17 21:32:36 ERROR - Packet for unconnected controller '1'
2020-06-17 21:32:36 ERROR - Packet for unconnected controller '1'
2020-06-17 21:32:36 ERROR - Packet for unconnected controller '1'
2020-06-17 21:32:36 ERROR - Packet for unconnected controller '1'
2020-06-17 21:32:36 ERROR - Packet for unconnected controller '1'
2020-06-17 21:32:36 ERROR - Packet for unconnected controller '1'
2020-06-17 21:32:36 ERROR - Packet for unconnected controller '1'
2020-06-17 21:32:36 ERROR - Packet for unconnected controller '1'
2020-06-17 21:32:58 ERROR - Error in bulk write: LIBUSB_ERROR_TIMEOUT
2020-06-17 21:32:58 DEBUG - Device error, terminating...
2020-06-17 21:32:58 ERROR - Failed to write command
2020-06-17 21:32:58 ERROR - Failed to calibrate
2020-06-17 21:32:58 ERROR - Failed to calibrate beacon
2020-06-17 21:32:58 ERROR - Failed to write pairing beacon
xow: io.c:2146: handle_events: Assertion `ctx->pollfds_cnt >= internal_nfds' failed.
[1]    20692 abort (core dumped)  xow
```

skrrrm, 2020-06-17:
❌**commit dce7bcf:** `FAILED` - did forget to try reconnect.
```
2020-06-17 21:57:27 INFO  - xow  ©Severin v. W.
2020-06-17 21:57:27 DEBUG - Opening device...
2020-06-17 21:57:28 INFO  - Dongle plugged in
2020-06-17 21:57:28 DEBUG - Firmware already loaded, resetting...
2020-06-17 21:57:28 DEBUG - Firmware loaded
2020-06-17 21:57:28 DEBUG - ASIC version: 7632
2020-06-17 21:57:28 DEBUG - Chip id: 7612
2020-06-17 21:57:28 INFO  - Wireless address: 62:45:b4:f2:02:6f
2020-06-17 21:57:28 INFO  - Dongle initialized
2020-06-17 21:57:52 INFO  - Pairing initiated
2020-06-17 21:57:52 INFO  - Pairing initiated
2020-06-17 21:57:52 INFO  - Pairing initiated
2020-06-17 21:57:52 INFO  - Pairing initiated
2020-06-17 21:57:58 DEBUG - Client pairing: 7e:ed:80:a9:79:79
2020-06-17 21:57:58 DEBUG - Client associating: 7e:ed:80:a9:79:79
2020-06-17 21:57:58 INFO  - Controller '1' connected
2020-06-17 21:57:58 INFO  - Serial number: 02600046126702
2020-06-17 21:58:00 DEBUG - Battery type: 1, level: 2
2020-06-17 22:03:45 DEBUG - Client lost: 1
2020-06-17 22:03:45 INFO  - Controller '1' disconnected
2020-06-17 22:03:45 ERROR - Packet for unconnected controller '1'
2020-06-17 22:03:45 ERROR - Packet for unconnected controller '1'
2020-06-17 22:03:45 ERROR - Packet for unconnected controller '1'
2020-06-17 22:03:45 ERROR - Packet for unconnected controller '1'
2020-06-17 22:03:45 ERROR - Packet for unconnected controller '1'
2020-06-17 22:03:45 ERROR - Packet for unconnected controller '1'
^C2020-06-17 22:04:16 DEBUG - Stop signal received
2020-06-17 22:04:16 DEBUG - Closing device...
2020-06-17 22:04:16 INFO  - Dongle power-off
2020-06-17 22:04:16 ERROR - Error in bulk read: LIBUSB_ERROR_OTHER
2020-06-17 22:04:16 DEBUG - Device error, terminating...
2020-06-17 22:04:16 ERROR - Error in bulk read: LIBUSB_ERROR_OTHER
2020-06-17 22:04:16 DEBUG - Device error, terminating...
```

medusalix, 2020-06-17:
Wait, so you're getting disconnects on 3a1e5cb5fea7f099ff936f1ab23e6bf7d209ea8d but not on 78cdb31a735bcc0d1b5ce870585c44ce4b55c657?

skrrrm, 2020-06-17:
❌commit 6c5bda7: `FAILED` - took a while until first disconnect though they got more frequent, and reconnects were automatic.
```
2020-06-17 22:09:39 INFO  - xow  ©Severin v. W.
2020-06-17 22:09:39 DEBUG - Opening device...
2020-06-17 22:09:39 INFO  - Dongle plugged in
2020-06-17 22:09:39 DEBUG - Firmware already loaded, resetting...
2020-06-17 22:09:39 DEBUG - Firmware loaded
2020-06-17 22:09:40 DEBUG - ASIC version: 7632
2020-06-17 22:09:40 DEBUG - Chip id: 7612
2020-06-17 22:09:40 INFO  - Wireless address: 62:45:b4:f2:02:6f
2020-06-17 22:09:40 INFO  - Dongle initialized
2020-06-17 22:09:46 INFO  - Pairing initiated
2020-06-17 22:09:46 INFO  - Pairing initiated
2020-06-17 22:09:47 INFO  - Pairing initiated
2020-06-17 22:09:47 INFO  - Pairing initiated
2020-06-17 22:09:50 DEBUG - Client associating: 7e:ed:80:a9:79:79
2020-06-17 22:09:50 INFO  - Controller '1' connected
2020-06-17 22:09:51 INFO  - Serial number: 02600046126702
2020-06-17 22:09:54 DEBUG - Battery type: 1, level: 2
2020-06-17 22:28:35 DEBUG - Battery type: 1, level: 2
2020-06-17 22:29:05 DEBUG - Client lost: 1
2020-06-17 22:29:05 INFO  - Controller '1' disconnected
2020-06-17 22:29:10 DEBUG - Client associating: 7e:ed:80:a9:79:79
2020-06-17 22:29:10 INFO  - Controller '1' connected
2020-06-17 22:29:10 INFO  - Serial number: 02600046126702
2020-06-17 22:30:11 DEBUG - Battery type: 1, level: 2
2020-06-17 22:31:51 DEBUG - Battery type: 1, level: 2
2020-06-17 22:32:45 DEBUG - Client lost: 1
2020-06-17 22:32:45 INFO  - Controller '1' disconnected
2020-06-17 22:32:50 DEBUG - Client associating: 7e:ed:80:a9:79:79
2020-06-17 22:32:50 INFO  - Controller '1' connected
2020-06-17 22:32:51 INFO  - Serial number: 02600046126702
2020-06-17 22:33:12 DEBUG - Battery type: 1, level: 2
2020-06-17 22:33:19 DEBUG - Battery type: 1, level: 2
2020-06-17 22:33:20 DEBUG - Client lost: 1
2020-06-17 22:33:20 INFO  - Controller '1' disconnected
2020-06-17 22:33:29 DEBUG - Stop signal received
2020-06-17 22:33:29 DEBUG - Closing device...
2020-06-17 22:33:29 INFO  - Dongle power-off
2020-06-17 22:33:29 ERROR - Error in bulk read: LIBUSB_ERROR_OTHER
2020-06-17 22:33:29 DEBUG - Device error, terminating...
2020-06-17 22:33:29 ERROR - Error in bulk read: LIBUSB_ERROR_OTHER
2020-06-17 22:33:29 DEBUG - Device error, terminating...
```

skrrrm, 2020-06-17:
> Wait, so you're getting disconnects on [3a1e5cb](https://github.com/medusalix/xow/commit/3a1e5cb5fea7f099ff936f1ab23e6bf7d209ea8d) but not on [78cdb31](https://github.com/medusalix/xow/commit/78cdb31a735bcc0d1b5ce870585c44ce4b55c657)? That's absolutely impossible.

Yes, so it seems. Albeit, I only tested for 10-15 minuttes. Do you want me to do another test on that commit?

medusalix, 2020-06-17:
> Do you want me to do another test on that commit?

Yes please.

skrrrm, 2020-06-17:
You're right, I've edited the comment for the commit of 78cdb31. I'll continue with the rest.

medusalix, 2020-06-17:
Make sure you're going in the opposite direction then, regarding the commits you're testing. We'll have to find the first one that works reliably.

skrrrm, 2020-06-17:
I see. I thought we went from v0.3 towards v0.4. I'll go down the list instead then.

skrrrm, 2020-06-17:
**Commit 5eca642:** `PASSED` - Only had it fail once after ~35 minutes. Instant reconnect.
```
2020-06-17 23:35:15 INFO  - xow  ©Severin v. W.
2020-06-17 23:35:16 INFO  - Dongle plugged in
2020-06-17 23:35:16 DEBUG - Firmware already loaded, resetting...
2020-06-17 23:35:16 DEBUG - Firmware loaded
2020-06-17 23:35:16 DEBUG - ASIC version: 7632
2020-06-17 23:35:16 DEBUG - Chip id: 7612
2020-06-17 23:35:16 INFO  - Wireless address: 62:45:b4:f2:02:6f
2020-06-17 23:35:16 INFO  - Dongle initialized
2020-06-17 23:35:26 INFO  - Pairing initiated
2020-06-17 23:35:26 INFO  - Pairing initiated
2020-06-17 23:35:26 INFO  - Pairing initiated
2020-06-17 23:35:26 INFO  - Pairing initiated
2020-06-17 23:35:27 DEBUG - Client associating: 7e:ed:80:a9:79:79
2020-06-17 23:35:27 INFO  - Controller '1' connected
2020-06-17 23:35:28 INFO  - Serial number: 02600046126702
2020-06-17 23:35:31 DEBUG - Battery type: 1, level: 2
2020-06-18 00:15:33 DEBUG - Battery type: 1, level: 2
2020-06-18 00:15:41 DEBUG - Client lost: 1
2020-06-18 00:15:41 INFO  - Controller '1' disconnected
2020-06-18 00:15:48 DEBUG - Client associating: 7e:ed:80:a9:79:79
2020-06-18 00:15:48 INFO  - Controller '1' connected
2020-06-18 00:15:48 INFO  - Serial number: 02600046126702
2020-06-18 00:16:08 DEBUG - Battery type: 1, level: 2
2020-06-18 00:16:28 DEBUG - Battery type: 1, level: 2
2020-06-18 00:16:48 DEBUG - Battery type: 1, level: 2
2020-06-18 00:20:48 DEBUG - Battery type: 1, level: 2
2020-06-18 00:22:28 DEBUG - Battery type: 1, level: 2
2020-06-18 00:22:36 DEBUG - Battery type: 1, level: 2
2020-06-18 00:22:37 DEBUG - Client lost: 1
2020-06-18 00:22:37 INFO  - Controller '1' disconnected
^C2020-06-18 00:22:44 DEBUG - Terminating device...
2020-06-18 00:22:44 INFO  - Dongle disconnected
```

skrrrm, 2020-06-21:
Commit 6c2c146:
```
2020-06-21 17:39:41 INFO  - xow  ©Severin v. W.
2020-06-21 17:39:41 INFO  - Dongle plugged in
2020-06-21 17:39:41 DEBUG - Firmware loaded
2020-06-21 17:39:42 DEBUG - ASIC version: 7632
2020-06-21 17:39:42 DEBUG - Chip id: 7612
2020-06-21 17:39:42 INFO  - Wireless address: 62:45:b4:f2:02:6f
2020-06-21 17:39:42 INFO  - Dongle initialized
2020-06-21 17:40:18 INFO  - Pairing initiated
2020-06-21 17:40:18 INFO  - Pairing initiated
2020-06-21 17:40:18 INFO  - Pairing initiated
2020-06-21 17:40:18 INFO  - Pairing initiated
2020-06-21 17:40:20 DEBUG - Client associating: 7e:ed:80:a9:79:79
2020-06-21 17:40:20 INFO  - Controller '1' connected
2020-06-21 17:40:21 INFO  - Serial number: 02600046126702
2020-06-21 17:40:22 DEBUG - Battery type: 1, level: 2
2020-06-21 17:40:42 DEBUG - Battery type: 1, level: 2
2020-06-21 17:40:54 DEBUG - Client lost: 1
2020-06-21 17:40:54 INFO  - Controller '1' disconnected
2020-06-21 17:40:59 DEBUG - Client associating: 7e:ed:80:a9:79:79
2020-06-21 17:40:59 INFO  - Controller '1' connected
2020-06-21 17:40:59 INFO  - Serial number: 02600046126702
2020-06-21 17:41:20 DEBUG - Battery type: 1, level: 2
2020-06-21 17:41:40 DEBUG - Battery type: 1, level: 2
2020-06-21 17:43:00 DEBUG - Battery type: 1, level: 2
2020-06-21 17:44:00 DEBUG - Battery type: 1, level: 2
2020-06-21 17:44:40 DEBUG - Battery type: 1, level: 2
2020-06-21 17:45:00 DEBUG - Battery type: 1, level: 2
2020-06-21 17:45:40 DEBUG - Battery type: 1, level: 2
2020-06-21 17:46:40 DEBUG - Battery type: 1, level: 2
2020-06-21 17:47:21 DEBUG - Client lost: 1
2020-06-21 17:47:21 INFO  - Controller '1' disconnected
2020-06-21 17:47:27 DEBUG - Client associating: 7e:ed:80:a9:79:79
2020-06-21 17:47:27 INFO  - Controller '1' connected
2020-06-21 17:47:27 INFO  - Serial number: 02600046126702
2020-06-21 17:47:47 DEBUG - Battery type: 1, level: 2
2020-06-21 17:50:00 DEBUG - Client lost: 1
2020-06-21 17:50:00 INFO  - Controller '1' disconnected
2020-06-21 17:50:05 DEBUG - Client associating: 7e:ed:80:a9:79:79
2020-06-21 17:50:05 INFO  - Controller '1' connected
2020-06-21 17:50:06 INFO  - Serial number: 02600046126702
2020-06-21 17:50:26 DEBUG - Battery type: 1, level: 2
2020-06-21 17:51:06 DEBUG - Battery type: 1, level: 2
2020-06-21 17:51:26 DEBUG - Battery type: 1, level: 2
2020-06-21 17:51:46 DEBUG - Battery type: 1, level: 2
2020-06-21 17:52:46 DEBUG - Battery type: 1, level: 2
2020-06-21 17:53:26 DEBUG - Battery type: 1, level: 2
2020-06-21 17:53:46 DEBUG - Battery type: 1, level: 2
2020-06-21 17:54:27 DEBUG - Client lost: 1
2020-06-21 17:54:27 INFO  - Controller '1' disconnected
2020-06-21 17:54:27 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:27 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:28 ERROR - Packet for unconnected controller '1'
[... 114 lines cut ...]
2020-06-21 17:54:32 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:32 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:32 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:32 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:32 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:32 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:32 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:32 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:32 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:32 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:32 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:32 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:32 ERROR - Packet for unconnected controller '1'
2020-06-21 17:54:32 ERROR - Packet for unconnected controller '1'
^C2020-06-21 17:54:57 DEBUG - Terminating device...
2020-06-21 17:54:58 ERROR - Error in bulk write: LIBUSB_ERROR_TIMEOUT
2020-06-21 17:54:58 ERROR - Failed to write command
2020-06-21 17:54:58 ERROR - Failed to set LED mode
2020-06-21 17:54:58 ERROR - Failed to turn off LED
2020-06-21 17:54:59 ERROR - Error in bulk write: LIBUSB_ERROR_TIMEOUT
2020-06-21 17:54:59 ERROR - Failed to write command
2020-06-21 17:54:59 ERROR - Failed to set power mode
2020-06-21 17:54:59 ERROR - Failed to turn off radio
2020-06-21 17:54:59 INFO  - Dongle disconnected
```

skrrrm, 2020-06-21:
Commit fd27767:
```
2020-06-21 18:32:05 INFO  - xow  ©Severin v. W.
2020-06-21 18:32:11 INFO  - Dongle plugged in
2020-06-21 18:32:11 DEBUG - Firmware already loaded, resetting...
2020-06-21 18:32:11 DEBUG - Firmware loaded
2020-06-21 18:32:11 DEBUG - ASIC version: 7632
2020-06-21 18:32:11 DEBUG - Chip id: 7612
2020-06-21 18:32:11 INFO  - Wireless address: 62:45:b4:f2:02:6f
2020-06-21 18:32:11 INFO  - Dongle initialized
2020-06-21 18:32:21 DEBUG - Client associating: 7e:ed:80:a9:79:79
2020-06-21 18:32:21 INFO  - Controller '1' connected
2020-06-21 18:32:22 INFO  - Serial number: 02600046126702
2020-06-21 18:32:25 DEBUG - Battery type: 1, level: 2
2020-06-21 18:46:06 DEBUG - Battery type: 1, level: 2
2020-06-21 18:46:11 DEBUG - Client lost: 1
2020-06-21 18:46:11 INFO  - Controller '1' disconnected
2020-06-21 18:46:16 DEBUG - Client associating: 7e:ed:80:a9:79:79
2020-06-21 18:46:16 INFO  - Controller '1' connected
2020-06-21 18:46:17 INFO  - Serial number: 02600046126702
2020-06-21 18:48:18 DEBUG - Battery type: 1, level: 2
2020-06-21 19:13:59 DEBUG - Battery type: 1, level: 2
^C2020-06-21 19:15:37 DEBUG - Terminating device...
2020-06-21 19:15:37 INFO  - Dongle disconnected
```

medusalix, 2020-06-21:
Would you mind opening an issue on [`libusb`](https://github.com/libusb/libusb)'s GitHub page for this problem? It seems like there's a relation between my switch from the async `libusb` API to the sync API in commit 78cdb31a735bcc0d1b5ce870585c44ce4b55c657 and the problems you are encountering. I can't guarantee that they'll be able to help you or if it's something that can be fixed on xow's side.

skrrrm, 2020-06-21:
I can give it a [go](https://github.com/libusb/libusb/issues/746) but we're getting a little out of depth for me. I'm not entirely sure how to phrase the issue other than showing this thread. So please make your voice heard if you have anything to add. 

I'll just stay on v0.3 for now.
Other than that - thanks for the work so far, it is much appreciated.

skrrrm, 2020-06-23:
Tried on the latest version with an USB hub as [suggested](https://github.com/libusb/libusb/issues/746#issuecomment-647186267).

**No-HUB**
```
2020-06-23 15:35:20 INFO  - xow v0.5-10-gea22499 ©Severin v. W.
2020-06-23 15:35:20 INFO  - Waiting for device...
2020-06-23 15:35:20 DEBUG - Opening device...
2020-06-23 15:35:21 DEBUG - Firmware already loaded, resetting...
2020-06-23 15:35:21 DEBUG - Firmware loaded
2020-06-23 15:35:21 DEBUG - ASIC version: 7632
2020-06-23 15:35:21 DEBUG - MAC version: 7662
2020-06-23 15:35:21 DEBUG - Chip id: 7612
2020-06-23 15:35:21 INFO  - Wireless address: 62:45:b4:f2:02:6f
2020-06-23 15:35:21 DEBUG - Channel 1, power: 32
2020-06-23 15:35:21 DEBUG - Channel 6, power: 33
2020-06-23 15:35:21 DEBUG - Channel 11, power: 33
2020-06-23 15:35:21 DEBUG - Channel 36, power: 41
2020-06-23 15:35:21 DEBUG - Channel 40, power: 41
2020-06-23 15:35:21 DEBUG - Channel 44, power: 41
2020-06-23 15:35:21 DEBUG - Channel 48, power: 41
2020-06-23 15:35:21 DEBUG - Channel 149, power: 36
2020-06-23 15:35:21 DEBUG - Channel 153, power: 36
2020-06-23 15:35:21 DEBUG - Channel 157, power: 36
2020-06-23 15:35:21 DEBUG - Channel 161, power: 36
2020-06-23 15:35:21 DEBUG - Channel 165, power: 36
2020-06-23 15:35:21 INFO  - Dongle initialized
2020-06-23 15:35:30 INFO  - Controller '1' connected
2020-06-23 15:35:30 INFO  - Device announced, product id: 02ea
2020-06-23 15:35:30 DEBUG - Firmware version: 3.1.1221.0
2020-06-23 15:35:30 DEBUG - Hardware version: 1284.1.1.1
2020-06-23 15:35:30 INFO  - Serial number: 02600046126702
2020-06-23 15:35:31 DEBUG - Rumble count: 0, duration: 5000, delay: 0
2020-06-23 15:50:27 DEBUG - Rumble count: 0, duration: 5000, delay: 0
2020-06-23 15:50:28 ERROR - Error in bulk write: LIBUSB_ERROR_TIMEOUT
2020-06-23 15:50:28 ERROR - Failed to write command
2020-06-23 15:50:28 ERROR - Failed to set LED mode
2020-06-23 15:50:28 ERROR - Failed to set LED mode
2020-06-23 15:50:28 INFO  - Shutting down...
2020-06-23 15:50:28 ERROR - Failed to remove controller
2020-06-23 15:50:29 DEBUG - Closing device...
```

**USB-HUB:**
```
2020-06-23 16:23:20 DEBUG - Channel 36, power: 41
2020-06-23 16:23:20 DEBUG - Channel 40, power: 41
2020-06-23 16:23:20 DEBUG - Channel 44, power: 41
2020-06-23 16:23:20 DEBUG - Channel 48, power: 41
2020-06-23 16:23:20 DEBUG - Channel 149, power: 36
2020-06-23 16:23:20 DEBUG - Channel 153, power: 36
2020-06-23 16:23:20 DEBUG - Channel 157, power: 36
2020-06-23 16:23:20 DEBUG - Channel 161, power: 36
2020-06-23 16:23:20 DEBUG - Channel 165, power: 36
2020-06-23 16:23:20 INFO  - Dongle initialized
2020-06-23 16:23:48 INFO  - Pairing enabled
2020-06-23 16:23:48 INFO  - Pairing enabled
2020-06-23 16:23:48 INFO  - Pairing enabled
2020-06-23 16:23:50 INFO  - Controller '1' connected
2020-06-23 16:23:50 INFO  - Device announced, product id: 02ea
2020-06-23 16:23:50 DEBUG - Firmware version: 3.1.1221.0
2020-06-23 16:23:50 DEBUG - Hardware version: 1284.1.1.1
2020-06-23 16:23:50 INFO  - Serial number: 02600046126702
2020-06-23 16:23:50 INFO  - Battery level: 2
2020-06-23 16:23:54 DEBUG - Rumble count: 0, duration: 5000, delay: 0
2020-06-23 16:34:16 DEBUG - Rumble count: 0, duration: 5000, delay: 0
2020-06-23 16:34:17 INFO  - Controller '1' disconnected
2020-06-23 16:34:42 INFO  - Shutting down... 
Stopping Xbox One Wireless Dongle Driver...
2020-06-23 16:34:44 ERROR - Error in bulk write: LIBUSB_ERROR_TIMEOUT
2020-06-23 16:34:44 ERROR - Failed to write command
2020-06-23 16:34:44 ERROR - Failed to set LED mode
2020-06-23 16:34:44 ERROR - Failed to turn off LED
2020-06-23 16:34:45 ERROR - Error in bulk write: LIBUSB_ERROR_TIMEOUT
2020-06-23 16:34:45 ERROR - Failed to write command
2020-06-23 16:34:45 ERROR - Failed to set power mode
2020-06-23 16:34:45 ERROR - Failed to turn off radio
2020-06-23 16:34:45 DEBUG - Closing device...
```

**v0.3 No-HUB**
```
2020-06-23 14:35:14 INFO  - xow  ©Severin v. W.
2020-06-23 14:35:20 INFO  - Dongle plugged in
2020-06-23 14:35:20 INFO  - Chip address: 62:45:b4:f2:02:6f
2020-06-23 14:35:20 INFO  - Dongle initialized
2020-06-23 14:35:26 INFO  - Controller '1' connected
2020-06-23 14:35:27 INFO  - Serial number: 02600046126702
2020-06-23 15:00:34 ^C
```

bernardodpc, 2020-07-02:
I'm sorry if I'm making something wrong posting here but I wanted to contribute. I just bought today the dongle and installed version 0.5
I can't play because the controller keeps disconnecting.
did I do something wrong? I just start xow with systemctl beforehand and plug it. when I turn on the controller, it connects successfully but after some time it disconnects.

when I disable xow service and run it directly I get:
2020-07-01 23:47:59 INFO  - xow v0.5-11-g4aa49f2 ©Severin v. W.
2020-07-01 23:47:59 INFO  - Waiting for device...
2020-07-01 23:48:03 INFO  - Wireless address: 62:45:b5:03:8c:ff
2020-07-01 23:48:03 INFO  - Dongle initialized
2020-07-01 23:48:06 INFO  - Controller '1' connected
2020-07-01 23:48:06 INFO  - Device announced, product id: 02ea
2020-07-01 23:48:06 INFO  - Serial number: 02600319502940
2020-07-01 23:48:07 INFO  - Battery level: 3
2020-07-01 23:48:14 ERROR - Error in bulk write: LIBUSB_ERROR_TIMEOUT
2020-07-01 23:48:14 ERROR - Failed to write command
2020-07-01 23:48:14 ERROR - Failed to send controller packet
2020-07-01 23:48:14 INFO  - Shutting down...

my lspci | grep USB gives only one line:
00:14.0 USB controller: Intel Corporation 100 Series/C230 Series Chipset Family USB 3.0 xHCI Controller (rev 31)

I use ubuntu 20.04. I can give more details too.

kakra, 2020-07-02:
If this problem occurs during rumbles, it may be fixed by #100

bernardodpc, 2020-07-02:
> If this problem occurs during rumbles, it may be fixed by #100

hmm let's see if this resolves when it's merged. I tested with jamestown, detroit become human and fifa 19. 
jamestown and detroit become human indeed have rumble but I don't remember fifa having it.
thanks for answering

kakra, 2020-07-03:
> hmm let's see if this resolves when it's merged. I tested with jamestown, detroit become human and fifa 19.
> jamestown and detroit become human indeed have rumble but I don't remember fifa having it.

The fix in #100 is only for disconnects during rumble. The disconnect a few seconds after xow startup probably has a very different cause - maybe libusb as @medusalix already wrote.

bernardodpc, 2020-07-03:
> The fix in #100 is only for disconnects during rumble. The disconnect a few seconds after xow startup probably has a very different cause - maybe libusb as @medusalix already wrote.

yeah. I saw that. thanks for the feedback.
but I'm still clueless about what to do in order to make it work properly. I'll wait for future versions.
I tested version 0.3 and 0.4 and still get disconnects after a few time with LIBUSB_ERROR_TIMEOUT

nakulbende, 2020-07-06:
Hey @medusalix and @skrrrm - I have a similar issue. I am on RPi4, and am using the dongle with retropie. I was actually running v5 without any issue, and I am not sure what broke, but the controller disconnects every few seconds. Saw the post here, and tried the commit [5eca642 ](https://github.com/medusalix/xow/commit/5eca642e67848d5ba5dd42bea4199783848bd152) since it worked for @skrrrm . Unfortunately, it is still disconnecting a lot after working for a day or two. 

Is there a certain commit which I can use issue free? Seems like you could use v3,0 issue free I tried that too - but was not successful.

nakulbende, 2020-07-06:
BTW, here are some logs:

`lspci | grep USB` shows: `01:00.0 USB controller: VIA Technologies, Inc. VL805 USB 3.0 Host Controller (rev 01)`

`lsusb` shows this, when two other controllers and a storage device is plugged in to my raspberry pi USB ports: 

```
Bus 002 Device 002: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 24c6:543a  
Bus 001 Device 004: ID 24c6:543a  
Bus 001 Device 003: ID 045e:02fe Microsoft Corp. 
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Running `xow`: 

```
pi@retropie:~ $ xow
2020-07-06 12:28:47 INFO  - xow v0.3-29-g5eca642 ©Severin v. W.
2020-07-06 12:28:47 INFO  - Dongle plugged in
2020-07-06 12:28:47 INFO  - Wireless address: 62:45:b5:0c:8b:75
2020-07-06 12:28:47 INFO  - Dongle initialized
2020-07-06 12:28:58 INFO  - Controller '1' connected
2020-07-06 12:29:01 INFO  - Controller '2' connected
2020-07-06 12:30:22 INFO  - Controller '2' disconnected
2020-07-06 12:30:26 INFO  - Controller '2' connected
2020-07-06 12:30:31 INFO  - Controller '1' disconnected
2020-07-06 12:30:32 INFO  - Controller '2' disconnected
2020-07-06 12:30:35 INFO  - Controller '1' connected
2020-07-06 12:30:36 INFO  - Controller '2' connected
2020-07-06 12:30:46 INFO  - Controller '2' disconnected
2020-07-06 12:30:46 INFO  - Controller '1' disconnected
2020-07-06 12:30:50 INFO  - Controller '1' connected
2020-07-06 12:30:50 INFO  - Controller '2' connected
2020-07-06 12:31:06 INFO  - Controller '1' disconnected
2020-07-06 12:31:10 INFO  - Controller '1' connected
2020-07-06 12:31:40 INFO  - Controller '2' disconnected
2020-07-06 12:31:44 INFO  - Controller '2' connected
2020-07-06 12:31:46 INFO  - Controller '1' disconnected
2020-07-06 12:31:50 INFO  - Controller '1' connected
2020-07-06 12:31:50 INFO  - Controller '1' disconnected
2020-07-06 12:31:55 INFO  - Controller '1' connected
2020-07-06 12:32:06 INFO  - Controller '1' disconnected
2020-07-06 12:32:11 INFO  - Controller '1' connected
2020-07-06 12:32:30 INFO  - Controller '1' disconnected
```

medusalix, 2020-07-06:
> yeah. I saw that. thanks for the feedback. but I'm still clueless about what to do in order to make it work properly. I'll wait for future versions.

@bernardodpc You could try running xow on a different Linux machine (if you have one at hand). Just to confirm whether or not this is a problem with your USB controller.

@nakulbende Your problem sounds more like issue #57. I can't spot any USB-related errors in your logs.

nakulbende, 2020-07-06:
@medusalix I have this issue when I am right next to it - like 2 feet. I will try on my desktop running Manjaro to get a 2nd reading too :)

nakulbende, 2020-07-06:
Oo, by the way, if I just connect the controller, and leave them, its fine for longer. It is when I start using the controllers while playing games, it starts disconnecting

medusalix, 2020-07-06:
In that case a Wireshark USB capture might helpful.

nakulbende, 2020-07-06:
Interesting, is there a guide on how to get that? Is it like a snooping tool to see what is going on?

kakra, 2020-07-06:
> Oo, by the way, if I just connect the controller, and leave them, its fine for longer. It is when I start using the controllers while playing games, it starts disconnecting

Does it rumble during the disconnect?

nakulbende, 2020-07-06:
I do not have rumble at all - but I did try to use your repo to see if
anything works.

Nakul Bende,

Materials Scientist, Formlabs Inc.

http://nakulbende.com

- Ever tried. Ever failed. Try Again. Fail again. Fail better.

--Samuel Beckett

bernardodpc, 2020-07-06:
> @bernardodpc You could try running xow on a different Linux machine (if you have one at hand). Just to confirm whether or not this is a problem with your USB controller.
>
> @nakulbende Your problem [...]

I tried switching USB ports on my laptop and it gives me the same error like @nakulbende is reporting: if I just leave it alone it doesn't disconnect. It disconnects as soon as I run some game with no rumble. this was my report until yesterday.

a little update here: I tried to connect the dongle on my USB hub and it is working great so far, flawless. I could play jamestown and detroit become human on proton and some native games. it's strange that when connecting directly on my USB ports it doesn't work as expected.

skrrrm, 2020-07-06:
> a little update here: I tried to connect the dongle on my USB hub and it is working great so far, flawless. I could play jamestown and detroit become human on proton and some native games. it's strange that when connecting directly on my USB ports it doesn't work as expected.

> disconnect is rarely caused by s/w unless it cause fault and reboot in the device f/w.
> Am experiencing such issue with some device (most are dwc3 based otg h/w ) and with with some host controler/Cable . It is linked to LPM (Link power management) of USB 3.0 bus. The device h/w if faulty and rarely fail on link training , but it occur so ofen whith LPM aggressive host that it come radily unusable as you start getting exhange.
> You may overcome this problem by disabling PM on the root hub controller (if it work often it does nothing) or try to use a hub in between as they tend to fully filter the power management. it work around the problem.
https://github.com/libusb/libusb/issues/746#issuecomment-647186267

@bernardodpc This might explain your issue and fits with your solution.
***

The consensus I gathered from `libusb` for my particular issue is that my dongle is faulty and the switch of different sync APIs within `xow` triggered the problem. Although they did suggest me to go further via their mailing-list - I haven't got around to that and probably won't since `v3` works without any issues.

medusalix, 2020-07-06:
I don't think your dongle is faulty. It's most likely caused by a buggy driver for your USB controller. Like I said, those problems can be narrowed down by trying xow with a different USB controller on a different PC.