medusalix/xow #91: Program crash after plug audio jack

blythe-ren, 2020-06-18:
Hi,
Recently I've bought a new xbox controller. I am happy to find a solution that makes my linuxmint LXDE can work with this game controller properly through xow program. very appreciate all authors effort.

My controller is wireless version. I didn't test blutooth cause my mainboard didn't support.

After plug headset into controller, I found the xbox-led on controller start blinking.  "ps" command shown xow program disappear

therefore I trigger program manually and program dump message below
==
2020-06-18 14:33:04 INFO  - xow v0.5-6-g9e86c52 ©Severin v. W.
2020-06-18 14:33:04 INFO  - Waiting for device...
2020-06-18 14:33:05 INFO  - Wireless address: 62:45:bd:01:8d:4b
2020-06-18 14:33:05 INFO  - Dongle initialized
2020-06-18 14:33:06 INFO  - Controller '1' connected
2020-06-18 14:33:07 INFO  - Device announced, product id: 02ea
2020-06-18 14:33:07 INFO  - Device announced, product id: 02f6
terminate called after throwing an instance of 'InputException'
  what():  Error adding key code: Invalid argument
Aborted (core dumped)
==
any body can help me ? I've been check that it should be no issue on audio jack

flopglop, 2020-06-18:
Hi.
I can confirm this also happens on manjaro kde, however my controller works out if the box when plugged in, and the linux kernel supports wired xbox controllers out of the box. Try using another usb cable?

medusalix, 2020-06-18:
Headsets are currently not supported, see issue #4. USB connectivity is implemented solely in the kernel.

blythe-ren, 2020-06-18:
yeah, I've been test under win 10 environment. The cable is OK  . Seems my new controller have good function properly in all 3 connectivity way. I also update to the xbox controller to newest driver ..

blythe-ren, 2020-06-18:
`Linux b-pc 4.19.0-9-amd64 #1 SMP Debian 4.19.118-2+deb10u1 (2020-06-07) x86_64 GNU/Linux`

have no idea why usb can't work ..

blythe-ren, 2020-06-18:
> Headsets are currently not supported, see issue #4. USB connectivity is implemented solely in the kernel.

thanks medusalix,
I also want to report a problem. Program exit when I plug out my wireless receiver.

medusalix, 2020-06-18:
> Program exit when I plug out my wireless receiver.

You need to install xow as a system service as described in the README. I wouldn't recommend running xow manually.

blythe-ren, 2020-06-18:
thanks .. I have no question

closed 2020-06-18 by blythe-ren