medusalix/xow #93: At launch I get a LIBUSB_ERROR_ACCESS

PercyJW-2, 2020-06-19:
I installed version 0.5 form the AUR, but when I start xow and then look in the service status it prints
```
● xow.service - Xbox One Wireless Dongle Driver
     Loaded: loaded (/usr/lib/systemd/system/xow.service; enabled; vendor preset: disabled)
     Active: failed (Result: core-dump) since Fri 2020-06-19 14:32:31 CEST; 2s ago
    Process: 122006 ExecStart=/usr/bin/xow (code=dumped, signal=ABRT)
   Main PID: 122006 (code=dumped, signal=ABRT)

systemd[1]: Started Xbox One Wireless Dongle Driver.
xow[122006]: 2020-06-19 14:32:31 INFO  - xow  ©Severin v. W.
xow[122006]: 2020-06-19 14:32:31 INFO  - Waiting for device...
xow[122006]: terminate called after throwing an instance of 'UsbException'
xow[122006]:   what():  Error opening device: LIBUSB_ERROR_ACCESS
systemd[1]: xow.service: Main process exited, code=dumped, status=6/ABRT
systemd[1]: xow.service: Failed with result 'core-dump'.
```

medusalix, 2020-06-19:
Did you do a reboot after installing xow?

PercyJW-2, 2020-06-19:
No, I forgot that. Now it is running

closed 2020-06-19 by PercyJW-2