medusalix/xow #94: Rock band 4 guitar support

abb1995, 2020-06-22:
Looking to get support added for rock band 4 guitars. The controller seems to pair with the dongle just fine, but none of the button inputs are registering. Including buttons that would be shared with a controller like the D-Pad or start/select buttons.

`● xow.service - Xbox One Wireless Dongle Driver
   Loaded: loaded (/etc/systemd/system/xow.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-06-22 15:24:15 CDT; 3min 45s ago
 Main PID: 8432 (xow)
    Tasks: 4 (limit: 4915)
   Memory: 1.1M
   CGroup: /system.slice/xow.service
           └─8432 /usr/local/bin/xow

Jun 22 15:24:15 epsilon systemd[1]: Started Xbox One Wireless Dongle Driver.
Jun 22 15:24:15 epsilon xow[8432]: 2020-06-22 15:24:15 INFO  - xow v0.5-10-gea22499 ©Severin v. W.
Jun 22 15:24:15 epsilon xow[8432]: 2020-06-22 15:24:15 INFO  - Waiting for device...
Jun 22 15:24:16 epsilon xow[8432]: 2020-06-22 15:24:16 INFO  - Wireless address: 62:45:b4:fa:04:bf
Jun 22 15:24:16 epsilon xow[8432]: 2020-06-22 15:24:16 INFO  - Dongle initialized
Jun 22 15:24:24 epsilon xow[8432]: 2020-06-22 15:24:24 INFO  - Controller '1' connected
Jun 22 15:24:24 epsilon xow[8432]: 2020-06-22 15:24:24 INFO  - Device announced, product id: 4161
Jun 22 15:24:24 epsilon xow[8432]: 2020-06-22 15:24:24 INFO  - Battery level: 3`

Even on windows however, I have to use a special driver to get this controller to work properly: https://github.com/artman41/guitarsniffer

I imagine things like the strum and whammy bars will need custom configuration. Not sure what kind of packet logs would be needed for this, so if someone could point me in the right direction for that, I can get that info.

medusalix, 2020-06-27:
The packet logs are already available on that GitHub repository you've linked. I don't have much time right now to implement support for it, maybe some time in the future.

medusalix, 2022-04-27:
The new driver now supports the Rock Band 4 Stratocaster. You can check it out [here](https://github.com/medusalix/xone).