medusalix/xow #97: Connecting headphones to controller causes driver to crash

Fisk24, 2020-06-26:
I am using an xbox one s controller and it has a 3.5mm audio jack on it.
When i plugged my headphones in the driver instantly crashed

● xow.service - Xbox One Wireless Dongle Driver
     Loaded: loaded (/etc/systemd/system/xow.service; enabled; vendor preset: disabled)
     Active: failed (Result: signal) since Fri 2020-06-26 01:43:48 CDT; 8s ago
    Process: 101098 ExecStart=/usr/local/bin/xow (code=killed, signal=ABRT)
   Main PID: 101098 (code=killed, signal=ABRT)

Jun 26 01:43:28 Esys-001 xow[101098]: 2020-06-26 01:43:28 INFO  - Pairing disabled
Jun 26 01:43:28 Esys-001 xow[101098]: 2020-06-26 01:43:28 INFO  - Controller '1' connected
Jun 26 01:43:29 Esys-001 xow[101098]: 2020-06-26 01:43:29 INFO  - Device announced, product id: 02ea
Jun 26 01:43:29 Esys-001 xow[101098]: 2020-06-26 01:43:29 INFO  - Serial number: 09VO0988525013
Jun 26 01:43:29 Esys-001 xow[101098]: 2020-06-26 01:43:29 INFO  - Battery level: 2
Jun 26 01:43:47 Esys-001 xow[101098]: 2020-06-26 01:43:47 INFO  - Device announced, product id: 02f6
Jun 26 01:43:47 Esys-001 xow[101098]: terminate called after throwing an instance of 'InputException'
Jun 26 01:43:47 Esys-001 xow[101098]:   what():  Error adding key code: Invalid argument
Jun 26 01:43:48 Esys-001 systemd[1]: xow.service: Main process exited, code=killed, status=6/ABRT
Jun 26 01:43:48 Esys-001 systemd[1]: xow.service: Failed with result 'signal'.

medusalix, 2020-06-26:
Duplicate of #91. Won't be fixed until audio support is implemented.

closed 2020-06-26 by medusalix

Fisk24, 2020-06-27:
Ok sure, but couldn't we just have it not crash in the mean time

medusalix, 2020-06-27:
That's not possible right now. If you ignore the accessory announcement and don't crash, the controller will just keep spamming your logs with those packets. The driver is deliberately crashing because it can't continue its normal operation.