medusalix/xow #98: Steam's "Turn off controller" chord not working

Merrit, 2020-06-27:
**Describe the bug**
The Steam guide button chord has an option for "Turn off controller" (default top button, eg Y or Triangle)

With xow nothing happens using this chord. All other chords are working, for example I have some for:
- Mute audio
- Toggle redshift/night color.
- Move mouse / click
- Alt-Tab

The "Turn off controller" chord works as expected when I tested with my PS4 controller.

**Steps to reproduce the behavior:**

- Setup chord in Steam -> Settings -> Controller -> Guide button chord configuration
- Call the chord (for me Guide button + Y)
- Nothing happens

**Expected behavior**:
- Expected the controller to power off

I tried to check the usb packets, however I am not familiar with wireshark and the documentation is a bit vague:

[image]

<details><summary>Wireshark capture (collapsed)</summary>
<p>

```
No.	Time	Source	Destination	Protocol	Length	Info
1	0.000000	host	1.6.0	USB	64	GET DESCRIPTOR Request DEVICE
2	0.000092	1.6.0	host	USB	82	GET DESCRIPTOR Response DEVICE
3	-0.000002	host	1.6.0	USB	64	GET DESCRIPTOR Request DEVICE
4	0.000984	host	2.1.0	USB	64	GET DESCRIPTOR Request DEVICE
5	0.000135	8.2.1	host	USB	71	URB_INTERRUPT in
6	0.000999	2.1.0	host	USB	82	GET DESCRIPTOR Response DEVICE
7	0.000089	1.6.0	host	USB	82	GET DESCRIPTOR Response DEVICE
8	0.000138	host	1.1.0	USB	64	GET DESCRIPTOR Request DEVICE
9	0.000137	host	1.1.0	USB	64	GET DESCRIPTOR Request DEVICE
10	0.000154	1.1.0	host	USB	82	GET DESCRIPTOR Response DEVICE
11	0.000155	1.1.0	host	USB	82	GET DESCRIPTOR Response DEVICE
12	0.001751	host	3.1.0	USB	64	GET DESCRIPTOR Request DEVICE
13	0.001763	3.1.0	host	USB	82	GET DESCRIPTOR Response DEVICE
14	0.000158	host	8.2.1	USB	64	URB_INTERRUPT in
15	0.000985	host	2.1.0	USB	64	GET DESCRIPTOR Request DEVICE
16	0.000999	2.1.0	host	USB	82	GET DESCRIPTOR Response DEVICE
17	0.002467	host	4.1.0	USB	64	GET DESCRIPTOR Request DEVICE
18	0.001752	host	3.1.0	USB	64	GET DESCRIPTOR Request DEVICE
19	0.002480	4.1.0	host	USB	82	GET DESCRIPTOR Response DEVICE
20	0.001764	3.1.0	host	USB	82	GET DESCRIPTOR Response DEVICE
21	0.002468	host	4.1.0	USB	64	GET DESCRIPTOR Request DEVICE
22	0.003317	host	5.3.0	USB	64	GET DESCRIPTOR Request DEVICE
23	0.002480	4.1.0	host	USB	82	GET DESCRIPTOR Response DEVICE
24	0.004395	5.3.0	host	USB	82	GET DESCRIPTOR Response DEVICE
25	0.003318	host	5.3.0	USB	64	GET DESCRIPTOR Request DEVICE
26	0.004428	host	5.2.0	USB	64	GET DESCRIPTOR Request DEVICE
27	0.004749	5.2.0	host	USB	81	GET DESCRIPTOR Response DEVICE[Malformed Packet]
28	0.004396	5.3.0	host	USB	82	GET DESCRIPTOR Response DEVICE
29	0.004773	host	5.1.0	USB	64	GET DESCRIPTOR Request DEVICE
30	0.004781	5.1.0	host	USB	82	GET DESCRIPTOR Response DEVICE
31	0.004429	host	5.2.0	USB	64	GET DESCRIPTOR Request DEVICE
32	0.004749	5.2.0	host	USB	81	GET DESCRIPTOR Response DEVICE[Malformed Packet]
33	0.004774	host	5.1.0	USB	64	GET DESCRIPTOR Request DEVICE
34	0.004782	5.1.0	host	USB	82	GET DESCRIPTOR Response DEVICE
35	0.005805	host	7.1.0	USB	64	GET DESCRIPTOR Request DEVICE
36	0.006346	host	8.3.0	USB	64	GET DESCRIPTOR Request DEVICE
37	0.005263	host	6.1.0	USB	64	GET DESCRIPTOR Request DEVICE
38	0.007140	8.3.0	host	USB	82	GET DESCRIPTOR Response DEVICE
39	0.005813	7.1.0	host	USB	82	GET DESCRIPTOR Response DEVICE
40	0.005271	6.1.0	host	USB	82	GET DESCRIPTOR Response DEVICE
41	0.007176	host	8.2.0	USB	64	GET DESCRIPTOR Request DEVICE
[... 779 lines cut ...]
I ran xow from terminal, and there is no output when calling the chord:

```
2020-06-26 20:40:09 INFO  - xow v0.5-8-gb3370b3 ©Severin v. W.
2020-06-26 20:40:09 INFO  - Waiting for device...
2020-06-26 20:40:09 INFO  - Wireless address: 62:45:b4:ec:3b:23
2020-06-26 20:40:09 INFO  - Dongle initialized
2020-06-26 20:40:13 INFO  - Pairing enabled
2020-06-26 20:40:13 INFO  - Pairing enabled
2020-06-26 20:40:13 INFO  - Pairing enabled
2020-06-26 20:40:16 INFO  - Controller '1' connected
2020-06-26 20:40:16 INFO  - Device announced, product id: 02d1
2020-06-26 20:40:18 INFO  - Battery level: 0
```

**Software**
Manjaro KDE
xow-git 127-1

**Hardware**
Controller: Model 1537

Not sure if this is actually a problem on xow's end.
I don't know how xow works, though my guess would be that Steam is sending a signal that isn't being passed to the controller maybe?
If there is anything further I can do to assist troubleshooting, let me know! :slightly_smiling_face:

medusalix, 2020-06-27:
First of all, thank you for the detailed bug report. Very much appreciated.

> Not sure if this is actually a problem on xow's end.
I don't know how xow works, though my guess would be that Steam is sending a signal that isn't being passed to the controller maybe?

I actually don't know how that chord is implemented. Maybe @slouken can shed some light on this? It's definitely something that would have to be added to xow, but I can't find any documentation for that feature.

Merrit, 2020-06-28:
The only thing I found that seemed remotely related was this: https://partner.steamgames.com/doc/features/steam_controller/getting_started_for_devs

Seems to be more about adding controller support to your games though. Not sure where else would be good to look for such documentation. Maybe the Steam for Linux github? https://github.com/ValveSoftware/steam-for-linux

slouken, 2020-06-28:
For the PS4, Steam is sending a USB packet directly to the controller to turn it off. With xow, Steam doesn't talk directly to the controller, so it isn't able to directly turn it off.

I've added the ability to signal xow to turn off controllers in the version shipped with the Steam Link hardware, but I haven't hooked this up to Steam's turn off controller functionality. It's just used when the Steam Link goes into sleep mode.

medusalix, 2020-06-28:
> I've added the ability to signal xow to turn off controllers in the version shipped with the Steam Link hardware, but I haven't hooked this up to Steam's turn off controller functionality.

It would also require a way for Steam to notify xow which controller it would like to turn off.