medusalix/xow #102: OpenRC support

AlwaysLivid, 2020-07-04:
Hey, I was wondering whether getting `xow` to work with the OpenRC service manager would be possible.

Thanks!

medusalix, 2020-07-06:
It's certainly possible, but I don't have any experience writing `init` scripts.
If you'd like to create such scripts, I'd suggest putting them into a repository and I'll add a link to the README.

kakra, 2020-07-06:
OpenRC does not support user services... I think xow was converted to running with user privileges only recently? So it would need to become an xdg autostart item - which is not compatible with systemd-user service... You can either use one or the other.

medusalix, 2020-07-06:
Nah, it's still a system-service right now. I've realized that user-services have too many drawbacks to be a good alternative.

kakra, 2020-07-06:
Then it should be very easy to create an OpenRC service, the format is pretty simple. But xow probably needs a real log file writer so logs to not vanish in /dev/null. I banned OpenRC from my system when I migrated to systemd (by masking /etc/init.d from package installation in Gentoo), so I can't really help here.

medusalix, 2020-08-04:
@AlwaysLivid Are you still interested in creating an `init` file yourself?

AlwaysLivid, 2020-08-04:
> @AlwaysLivid Are you still interested in creating an `init` file yourself?

In all honesty, I'm still trying to get the hang of Gentoo and I barely know anything about OpenRC, as well as a bunch of side-projects on my plate. I'll have to pass for the time being, sorry about that.

AlwaysLivid, 2020-08-04:
Just not particularly sure if I'm able to make the time for it.

kakra, 2020-08-04:
> > ... Are you still interested in creating an `init` file yourself?
> 
> In all honesty, I'm still trying to get the hang of Gentoo and I barely know anything about OpenRC, as well as a bunch of side-projects on my plate. I'll try to look into it.

Gentoo is in the process of making systemd a first-class citizen (but not a replacement). Portage currently unifies session manager support useflags from elogind and systemd and will make either of both a requirement. After this, switching from or to systemd should be mostly plug & play. So you may want to wait and not even bother with learning OpenRC, and just stick with systemd.

You could also enable systemd just now by switching to a systemd profile, but it may require you to tweak one or another useflag and maybe uninstall or mask a package currently. I'm using systemd in Gentoo since over 5 years without any major problems, and without any problems at all since at least 2 years.

medusalix, 2020-08-04:
@AlwaysLivid You can join the [Discord server](https://discord.gg/FDQxwWk) if you want.
@kakra Would you mind enabling the Server Widget in the server settings? I'd like to add a Discord badge to xow's README.

AlwaysLivid, 2020-08-04:
> Gentoo is in the process of making systemd a first-class citizen (but not a replacement). So you may want to wait and not even bother with learning OpenRC, and just stick with systemd.
> 
> I'm using systemd in Gentoo since over 5 years without any major problems, and without any problems at all since at least 2 years.

I'll admit that this is my first year using Gentoo and that I deliberately chose to use OpenRC for more complicated and superseding reasons that are go beyond the scope of this discussion, and I'd assume that others have done the same. I see your point and I completely understand how people are essentially systemd-ifying OpenRC in order to keep it compatible with more recent software.

But this is really just a matter of getting an Xbox One controller to work on a system with slightly different characteristics and not forcing people to just meet specific conditions (that should otherwise be irrelevant) to do the same. I'll see to it, if possible.

115100, 2020-08-06:
```sh
#!/sbin/openrc-run

command=/bin/xow
pidfile=/var/run/xow.pid
command_background=true
command_user="nobody:nobody"
name="xow"

depend() {
        keyword -lxc
}
```

Works for me. Ideally, you want a dedicated user to run xow under instead of `nobody`, but the guru overlay's ebuild doesn't create a user and I can't be bothered to maintain my own ebuild.

kapsh, 2020-08-11:
Delivered in #114.

Basically it works, customizing user, environment and logging supported too. 
```
❯ rc-service xow start
xow              | * Caching service dependencies ... [ ok ]
xow              | * Starting xow ... [ ok ]

~ root@hedgehog
❯ cat /var/log/xow.log 
2020-08-11 14:27:27 INFO  - xow v0.5-17-g0b3b7f3 ©Severin v. W.
2020-08-11 14:27:27 INFO  - Waiting for device...
2020-08-11 14:27:28 INFO  - Wireless address: 62:45:b5:0d:d8:63
2020-08-11 14:27:28 INFO  - Dongle initialized
```
The only issue I found that after suspend2ram `xow` exits with 
```
2020-08-11 13:46:42 ERROR - Error in bulk read: LIBUSB_ERROR_NO_DEVICE
2020-08-11 13:46:42 INFO  - Shutting down...
``` 
and service reported as `crashed`. Don't know a way to restart it automatically, cheers for heads-up.

kapsh, 2020-08-12:
Managed to automate restarts by using `supervise-daemon` instead of `start-stop-daemon` (current openrc default), works for me after suspend.