medusalix/xow #105: Configurable dead zone

a-kring, 2020-07-13:
I found myself changing the `controller.cpp` file to reduce flatness so that I can play Elite Dangerous (running on Linux via Steam Play). Would you consider making the dead zone a runtime setting?

medusalix, 2020-07-15:
No, that's out of scope for this driver. You can use `evdev-joystick` to set the deadzones separately for each controller, there's a link to the wiki page in the README.

a-kring, 2020-07-15:
Plugging the gamepad using a USB cable with the default Linux driver doesn't add dead zones. It only has them when connected via Bluetooth using the xpad driver, and when using xow. It still has a small dead zone on Wine now that I reduced the flatness in `controller.cpp`, but it is a much smaller one than the one it had before, which makes games that require continuous precision a lot more usable. I'll play with `evdev-joystick`.

Would you consider making the default dead zone in your driver the same as when connecting the gamepad via a USB cable?

kakra, 2020-07-15:
You should maybe post the values you're expecting because 4095 and 255 look pretty standard to me, that are the values used in the source code.

a-kring, 2020-07-16:
I'll paste some values from a couple of gamepads connected via multiple methods below.

## Scuf Prestige - USB

These are the numbers I get when my Scuf Prestige (modified Xbox One controller) is plugged directly to the USB port:

```
[akring@nebula ~]$ evdev-joystick --showcal /dev/input/by-id/usb-Microsoft_Controller_3032363030363035373732383235-event-joystick
Supported Absolute axes:
  Absolute axis 0x00 (0) (X Axis) (value: 674, min: -32768, max: 32767, flatness: 128 (=0.39%), fuzz: 16)
  Absolute axis 0x01 (1) (Y Axis) (value: 2551, min: -32768, max: 32767, flatness: 128 (=0.39%), fuzz: 16)
  Absolute axis 0x02 (2) (Z Axis) (value: 0, min: 0, max: 1023, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x03 (3) (X Rate Axis) (value: -168, min: -32768, max: 32767, flatness: 128 (=0.39%), fuzz: 16)
  Absolute axis 0x04 (4) (Y Rate Axis) (value: 1179, min: -32768, max: 32767, flatness: 128 (=0.39%), fuzz: 16)
  Absolute axis 0x05 (5) (Z Rate Axis) (value: 0, min: 0, max: 1023, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x10 (16) (Hat zero, x axis) (value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x11 (17) (Hat zero, y axis) (value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
```

As you can see there, the fuzz is just 16 and flatness is 128. I haven't modified anything. This is out-of-the-box on Fedora 32.

## Xbox Elite - USB

Similar values are visible when I plug in my Xbox Elite controller (first edition) via USB:

```
[akring@nebula ~]$ evdev-joystick --showcal /dev/input/by-id/usb-Microsoft_Controller_7EED84EADC7B-event-joystick
Supported Absolute axes:
  Absolute axis 0x00 (0) (X Axis) (value: 1366, min: -32768, max: 32767, flatness: 128 (=0.39%), fuzz: 16)
  Absolute axis 0x01 (1) (Y Axis) (value: 965, min: -32768, max: 32767, flatness: 128 (=0.39%), fuzz: 16)
  Absolute axis 0x02 (2) (Z Axis) (value: 0, min: 0, max: 1023, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x03 (3) (X Rate Axis) (value: 1071, min: -32768, max: 32767, flatness: 128 (=0.39%), fuzz: 16)
  Absolute axis 0x04 (4) (Y Rate Axis) (value: -1641, min: -32768, max: 32767, flatness: 128 (=0.39%), fuzz: 16)
  Absolute axis 0x05 (5) (Z Rate Axis) (value: 0, min: 0, max: 1023, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x10 (16) (Hat zero, x axis) (value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x11 (17) (Hat zero, y axis) (value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
```

## Scuf Prestige - Bluetooth

Now these numbers show the same parameters that xow uses: flatness at 4095 and fuzz at 255.

This is using the default Bluetooth driver from Fedora 32. I am pretty sure it is xpad.

```
[akring@nebula devices]$ evdev-joystick --s /dev/input/event22
Supported Absolute axes:
  Absolute axis 0x00 (0) (X Axis) (value: 35316, min: 0, max: 65535, flatness: 4095 (=6.25%), fuzz: 255)
  Absolute axis 0x01 (1) (Y Axis) (value: 35518, min: 0, max: 65535, flatness: 4095 (=6.25%), fuzz: 255)
  Absolute axis 0x02 (2) (Z Axis) (value: 32278, min: 0, max: 65535, flatness: 4095 (=6.25%), fuzz: 255)
  Absolute axis 0x05 (5) (Z Rate Axis) (value: 32629, min: 0, max: 65535, flatness: 4095 (=6.25%), fuzz: 255)
  Absolute axis 0x09 (9) (Accelerator) (value: 0, min: 0, max: 1023, flatness: 63 (=6.16%), fuzz: 3)
  Absolute axis 0x0a (10) (Brake) (value: 0, min: 0, max: 1023, flatness: 63 (=6.16%), fuzz: 3)
  Absolute axis 0x10 (16) (Hat zero, x axis) (value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x11 (17) (Hat zero, y axis) (value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
```

I'll continue posting numbers for xpadneo after a reboot.

a-kring, 2020-07-16:
## Scuf Prestige - Bluetooth (xpadneo)

These are parametres for my Scuf Prestige connected via Bluetooth using the xpadneo driver.

```
[akring@nebula modprobe.d]$ evdev-joystick --s /dev/input/event22
Supported Absolute axes:
  Absolute axis 0x00 (0) (X Axis) (value: 2314, min: -32768, max: 32767, flatness: 4095 (=12.50%), fuzz: 255)
  Absolute axis 0x01 (1) (Y Axis) (value: 2916, min: -32768, max: 32767, flatness: 4095 (=12.50%), fuzz: 255)
  Absolute axis 0x02 (2) (Z Axis) (value: 0, min: 0, max: 1023, flatness: 63 (=6.16%), fuzz: 3)
  Absolute axis 0x03 (3) (X Rate Axis) (value: -686, min: -32768, max: 32767, flatness: 4095 (=12.50%), fuzz: 255)
  Absolute axis 0x04 (4) (Y Rate Axis) (value: 0, min: -32768, max: 32767, flatness: 4095 (=12.50%), fuzz: 255)
  Absolute axis 0x05 (5) (Z Rate Axis) (value: 0, min: 0, max: 1023, flatness: 63 (=6.16%), fuzz: 3)
  Absolute axis 0x10 (16) (Hat zero, x axis) (value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x11 (17) (Hat zero, y axis) (value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
```

Same as xow: flatness: 4095 and fuzz: 255. I believe they were copied from xpad initially.

kakra, 2020-07-16:
That looks strange because the standard way HID configures these values results in 4095/255 and 63/3 (the calculation is `(maximum-minimum) << 4` and `... <<8`. So with USB, the device itself probably pushes these value, or the USB drivers uses different values. Those values look more like what the `joydev` API uses for old-style joysticks. HID drivers usually don't "invent" these values but derive them from the generic HID driver in the kernel (usually hard-coded but it's no decision out of the blue).

We could argue if 128/16 are good values - the Xbox One S sticks have quite some idle clearance/backlash (before the centering springs apply force) which fits between 2000-4000 from my tests. I'll try to set those values in out `xpadneo` driver and see how that works out. So maybe 2047/31 would already be good values? That would better match the idle zone of the sticks.

I cannot speak for `xow`, tho, if such a change would be incorporated.

medusalix, 2020-07-16:
I've adapted the `fuzz` and `flat` values from xpadneo (see issue #32). If the changes work out well for xpadneo, I'll change them for xow too.

kakra, 2020-07-16:
So we were looking at the HID generic driver and you looked at xpadneo. ;-)

Flatness is the deadzone before the joystick returns any values except 0, we can probably half that, this needs some experiments. Fuzz is the difference it needs before the joystick returns a new value, we can probably turn that way down, it should affect the experienced smoothness the most.

But getting precision around the deadzone is a bit more complicated. Usually, we probably need to scale the range from (deadzone..max) to (0..max) - this is almost a change I probably don't want to do. Or at least not now until I got the sensitivity curves in. But I'll experiment with that, too.

@-akring Where do you see the missing precision? During movement or around the deadzones? Which is most problematic? Getting the deadzone too low can result in phantom movement - we already see that when just tilting the controller without touching the sticks.

a-kring, 2020-07-16:
@kakra this is the process by which I figured something was wrong:

I installed Elite Dangerous (using Steam Play) and xow to move my gaming stuff to Linux. After this I noticed that my ship didn't seem respond in the way I was used to from Windows. I figured that parameters would be different on Linux and I needed to get used to it, but after weeks it wasn't working. I thought wireless interference or even lag might be to blame.

After doing some informal testing I ended up in https://gamepad-tester.com/ and noticed how difficult it was to draw a circle with my gamepad there. That explained why it was so hard to use the Full Spectrum Scanner in Elite Dangerous, as well as to provide fine inputs required for combat without flight assist: the dead zone is too large, and I cannot develop enough muscle memory to figure out exactly where it starts, as I must give my spaceship tiny precise inputs for milliseconds if I want to keep on target.

I tried for the first time connecting the gamepad with a cable, and it was glorious. The difference was huge. My hit ratio with the Guardian Gauss Cannon, a very hard to master weapon, went from about one in ten to about one in three, and on occasion it would get as high as three in five, when very focused. I could hold the reticle on target longer.

The Full Spectrum Scanner when using a wired connection is also much easier to use. This is an interface that requires using the gamepad thumb stick to move a reticle around a sphere pointing at things with a high degree of precision. With xow and default settings I cannot move the reticle diagonally with any degree of precision.

I ended up fudging only the flatness parameter in xow, but didn't mess with fuzz. And it got much closer to the wired experience. I'll fudge the fuzz down too next. In fact, I'll do the same in xpadneo and test that one too, although my Bluetooth connection is not ideal.

Another detail to consider: even when removing the dead zone entirely by setting flatness to zero and dropping it in the game UI, there is still a small dead zone present. I think that Wine has its own dead zone. I need to also test some native Linux games like X-Plane to see if this dead zone is still present.

Still my recommendation would be to go for the same values as the USB driver, and allow users to make the dead zone larger if they so require, either in-game or by using `evdev-joystick` or `jscal`, instead of keeping a very large dead zone that hurts precision as much as it helps prevent phantom movement. After all most games allow you to add dead zones, and Steam does too via its Joystick API. Removing the dead zone on the other hand is a lot more painful. In my experience the easiest way is to do it directly in the driver source code instead of messing around with udev rules, `jscal` and `evdev-joystick`.

a-kring, 2020-07-16:
This is a great article on the subject of gamepad dead zones. Steam joystick configuration can do all this, but at the expense of some input lag. In Windows it is about 1ms, but I don't think Steam on Linux is so fortunate: https://www.gamasutra.com/blogs/JoshSutphin/20130416/190541/Doing_Thumbstick_Dead_Zones_Right.php

kakra, 2020-07-16:
The fuzz parameter should make the smoothness of moving the stick around. But getting out of the deadzone is a jump of 4096 units. You made the jump small by shrinking the deadzone but that may not be the underlying problem, I think the big value jump out of the deadzone is the biggest problem then. I'll try to smooth that out in xpadneo, by the same time we can probably shrink the deadzone a little. Thanks for the article, will read it later. My current planning doesn't involve lags but I'll have to look at the article.

a-kring, 2020-07-16:
Yes, I'll reduce the fuzz in my settings too and see how it performs. In any case if it performs like wired it is perfect for me.

The article suggests that different strategies for dead zones should be used depending on the type of application. I completely agree with that sentiment. A shooter or a simulator would benefit from a small dead zone while the same might not the case for a fighting game or isometric MMORPG.

What this implies is that the approach taken by the USB driver of exposing the entire range of the thumb stick and letting the application decide how to handle its dead zones is probably a very good way to do things. As for the fuzz factor: a small value that doesn't typically jitter when you are not trying to change the force applied should be good. I understand that cheaper lower-quality gamepads may benefit from having a higher number there. The default fuzz chosen for the USB driver is ideal for both of my gamepads.

kakra, 2020-07-16:
It looks like the original Xbox controller already de-jitters the values reported so we can turn the fuzz down. But I'll implement that as a quirk flag in xpadneo so we can mark clone-devices with the bigger fuzz/deadzone easily and automatically.

a-kring, 2020-07-16:
I wonder where do these allegedly standard fuzz: 255 and flatness: 4095 come from in the default Bluetooth driver that Fedora 32 provides out-of-the-box. This is what I found on the [Linux kernel tree](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/input/joystick/xpad.c?h=v5.7.8&id=7977a464dcddd95c2b27de275a5f13f503d87e32#n1586):

```
static void xpad_set_up_abs(struct input_dev *input_dev, signed short abs)
{
	struct usb_xpad *xpad = input_get_drvdata(input_dev);

	switch (abs) {
	case ABS_X:
	case ABS_Y:
	case ABS_RX:
	case ABS_RY:	/* the two sticks */
		input_set_abs_params(input_dev, abs, -32768, 32767, 16, 128);
		break;
	case ABS_Z:
	case ABS_RZ:	/* the triggers (if mapped to axes) */
		if (xpad->xtype == XTYPE_XBOXONE)
			input_set_abs_params(input_dev, abs, 0, 1023, 0, 0);
		else
			input_set_abs_params(input_dev, abs, 0, 255, 0, 0);
		break;
	case ABS_HAT0X:
	case ABS_HAT0Y:	/* the d-pad (only if dpad is mapped to axes */
		input_set_abs_params(input_dev, abs, -1, 1, 0, 0);
		break;
	default:
		input_set_abs_params(input_dev, abs, 0, 0, 0, 0);
		break;
	}
}
```

This is fuzz: 16, flatness: 128 as in USB drivers. I'll keep digging to see where the large constants really come from.

By the way @kakra, have you observed any issues on games that you played due to phantom movement caused by a small deadzone when using a gamepad with default USB drivers? I'm trying to understand if this is not something that by convention is dealt with at the application or input library as opposed to the driver.

kakra, 2020-07-16:
I never really used the default drivers. Back the time when I got the gamepad, support for it was abysmal. At some time I found xpadneo, contacted the author, sponsored a controller for improving the driver and received a properly working driver within a few days. \o/

About the big values: It's in `hid-input.c` and its derived from the HID descriptor by using the logical minimum and maximum, and shifting the bits right. It's a one-fits-all calculation:
```c
// static void hidinput_configure_usage(struct hid_input *hidinput, struct hid_field *field,
//				     struct hid_usage *usage)
		if (field->application == HID_GD_GAMEPAD || field->application == HID_GD_JOYSTICK)
			input_set_abs_params(input, usage->code, a, b, (b - a) >> 8, (b - a) >> 4);
		else	input_set_abs_params(input, usage->code, a, b, 0, 0);
```

Fuzz introduces a lag of one frame in `input.c` (one frame has no defined time, it depends on the report rate of the device, it's actually 0ms for the controller over Bluetooth because it reports values to us multiple times in sequence for no new input):
```c
static int input_defuzz_abs_event(int value, int old_val, int fuzz)
{
	if (fuzz) {
		if (value > old_val - fuzz / 2 && value < old_val + fuzz / 2)
			return old_val;

		if (value > old_val - fuzz && value < old_val + fuzz)
			return (old_val * 3 + value) / 4;

		if (value > old_val - fuzz * 2 && value < old_val + fuzz * 2)
			return (old_val + value) / 2;
	}

	return value;
}
```

a-kring, 2020-07-16:
Now that is pretty cool. I saw that code today doing the bit shifting but didn't pay enough attention to figure out that's where the magic was happening.

The default fuzz parameter of 255 seems to be good enough. You get that the total range of 65535 divided by 255 yields 257. I don't think that the thumb stick has a higher resolution.

This seems to respond why Bluetooth and USB parameters are different. When connected over USB the gamepad seems to be using the specific xpad driver instead of the generic HID input driver that the device uses when connected over Bluetooth.

So to be clear there are four _different_ gamepad drivers on Linux for the Xbox One gamepad:

- xpad when connected over USB.
- Generic HID gamepad over Bluetooth.
- xpadneo when installed manually and using a Bluetooth connection.
- xow userspace driver via uinput when using the USB dongle and Wi-Fi Direct.

The fuzz and flatness values for xpadneo and xow were taken from the generic HID driver instead of the more specific xpad driver, which is why the Xbox Elite and Xbox One controllers feel dull when used with these drivers as opposed to the precise and responsive feel it gets when connected via USB.

I was ready to blame the mushy feeling on wireless link problems. I'm glad I dug and found that the cause of my problems are just these constants.

I forked both xow and xpadneo repositories and changed the constants there to match xpad. I recommend using these as default values for both xpadneo and xow, or at the very least have the option to set them via runtime configuration instead of having to fumble with udev and evdev incantations.

I'll check how Windows behaves with the official drivers and report my findings.

a-kring, 2020-07-16:
I'm on Windows right now, connected the gamepad over USB and headed to the Gamepad Tester or Edge. It is clearly the same behavior as xpad, as there is no dead zone and movement is buttery smooth.

I think this completes a compelling case to switch defaults for both xpadneo and xow to the same parameters used by xpad:

1. Consistency: gamepad responds equally when connected via USB and when used wirelessly, although not when using generic HID Bluetooth gamepad support (yet, this seems to be a good change to do on Linux).
2. Precision: availability of full range of motion offered by gamepad sensors. Ability to accurately move reticles or vehicles in directions other than straight up-down or left-right.
3. Consistency: gamepad responds to inputs exactly like it does on Windows.
4. Flexibility: allow specific games to [use custom dead zones that match its genre](https://www.gamasutra.com/blogs/JoshSutphin/20130416/190541/Doing_Thumbstick_Dead_Zones_Right.php) instead of forcing all to use a mostly sub-optimal cross-shaped dead-zone.
5. Precision: can use gamepad to control robots and RC devices accurately (gamepads are not only for games).

kakra, 2020-07-17:
I played around with it a while. The solution is apparently not so straight forward. While the `fuzz` parameter is processed right in the HID layer, the `flatness` parameter seems not - at least I didn't find code where it is used. That means, upper layers will see `evdev` values without any flatness applied but they can query the proposed flatness. So if something isn't working to your expectations, that's probably applied in one of the upper layers. Since you say that Wine is affected but native Linux games, too, I suppose SDL is applying `flatness` when reading `evdev` devices. Wine implements an SDL bus, and it should probably take care to kill flatness values from the interface when exposing the devices so the joystick becomes passed through to games without any dead zones applied - most games seem to implement that on their own.

But there's also `joydev`, and this kernel driver sits on top of HID devices - and IT DOES apply the flatness parameters to the joystick - and it uses the naïve approach of axial dead zones.

So just using a dead zone of 128 is still wrong because you want to apply no dead zone at all. To fix passing through to games and Wine which use SDL, the dead zone should be set to 0. May @medusalix could tell if the Windows GIP bus exposes values with or without a dead zone. I suspect it's more like a raw interface without dead zones, and defuzzing seems to be done in the controller firmware already (by looking at the raw HID reports in `btmon`). We would just apply a fuzz value to lower the amount of events generated during movement.

But now, the kernel `joydev` driver sees no dead zones and since this driver implemented that from the beginning, we can expect that games using this API also expect `joydev`. BTW: The HTML5 gamepad API - at least in Chrome - uses the `joydev` API for reading axes but uses `evdev` to figure out which axis is what (this creates a whole mess of its own as it also implies certain behavior based on the product ID).

So what can we do about it to make a consistent view across different APIs? My current approach is setting a dead zone of 0 as this seems most correct. But this leaves the `joydev` API unfiltered. So I'm currently implementing my own dead zone handler - and I already figured out that going below flatness ~3000 creates phantom movement when the stick is idle or snaps back to the center. Also I'm seeing that my left thumb stick needs a much bigger flatness values (3500 vs 2000) probably because it's worn out much more. So we absolutely need a flatness value to make all APIs happy, and 4095 seems to be a not too bad choice. But we can implement it more intelligently as outlined in the link you posted. The only problem I am currently having is that I'm not allowed to use FP math in the kernel driver, and all the examples rely on vector operations that require at least `sqrt()` to work. There's an integer implementation in the kernel I'd probably rather not use because it's not a constant time function, and I need to rescale the formulas for a range of 0..32678 because they depend on normalized values, that's 0..1, and that's not really useful with integers. I probably need to implement a LUT.

BTW: The resolution of the stick seems to be far greater than just 255 units, so a small fuzz value still makes sense. The original value seems to be meant for old school analog resistor-based joysticks that counted a register while a capacitor depleted - which is very inaccurate especially on multitask systems and faster CPUs. Also, fuzz does not divide the axis into discrete values, it's a smoothing algorithm over a sliding window of the last two values that also filters out too small deltas. So you actually can still see all 65535 possible values even with high fuzz parameters, the stick will "just" lag behind 1 or 2 report frames.

I still need to figure out which layer exactly applies the flatness for a final decision. I cannot rule out the kernel currently but I suspect that SDL is doing this. If that is true, SDL is doing it wrong and that's your problem.

a-kring, 2020-07-17:
I did some further testing last evening against X-Plane 11, a Linux-native game. With my fork of xow that reduces flatness to the same small value as xpad, the game runs without any apparent deadzone as expected. It's very smooth and responds to even the slightest input.

While running on Wine, Elite Dangerous still has a small deadzone, but I'm pretty sure this is happening at a higher layer, either the game itself or Wine. I am not using Steam gamepad support, which adds its own gamepad customization. I do know that [Wine can add deadzones](https://wiki.winehq.org/Useful_Registry_Keys):

```
      |  +->DefaultDeadZone
      |  |   [Sets a default value for every joystick axis dead zone. Should be a value between 0 and 10000.
      |  |    The value represents the percentage of the axis which will seen as dead (applications will think of it as being centered).
      |  |    For example 3000 will extend the dead zone to only the first 30% of the axis, while 0 will use the whole axis extension.]
```

I'm too stupid to find the registry file that my Steam installation is using so that I can check the value though.

You can see how the problem may compound when the kernel driver has its own deadzone, then on top of that Wine, and on top of that the game itself adds one: you get a very large deadzone. This is likely why Elite Dangerous was near unplayable for me on Linux when using xow, xpadneo or the default Bluetooth driver.

I do see your experience with the left stick being a little bit off-center. My thumb sticks are off too by a few percent all the time. This is the case also in Windows with the official Microsoft drivers. It is true that Bluetooth HID, xow, and xpadneo by default will center my sticks when I'm not using them in any app where I test them owing to the large 4095 dead zones.

It is though my assessment based on this discussion that the gamepad driver is not the place to add a deadzone, although if a driver author may insist on enhancing the driver by adding a deadzone, it would have to be done correctly, and by correctly I mean a Scaled Radial Dead Zone [as explained by Josh Sutphin](https://www.gamasutra.com/blogs/JoshSutphin/20130416/190541/Doing_Thumbstick_Dead_Zones_Right.php). I still strongly believe this to be something that should be done at the highest level, close to the game, rather than at the kernel driver. Microsoft seems to agree as evidenced by its Windows drivers.

Here is a screen capture of the Gamepad Tester showing my Scuf Prestige gamepad input using the wired xpad driver on Linux:

## Scuf Prestige - Neutral - USB - Linux

It is not perfectly centered, but that is perfectly fine.

[image]

a-kring, 2020-07-17:
## Scuf Prestige - Neutral - USB - Windows

It's exactly the same as in Windows.

[image]

a-kring, 2020-07-17:
## Xbox Elite - Neutral - USB - Windows

This is my Xbox Elite controller, wired on Windows and in a neutral position. Again no dead zone.

[image]

a-kring, 2020-07-17:
## Stadia Controller - Neutral - USB - Windows

The Stadia controller has a square deadzone. The minimum input required to get out of the deadzone is a bit more than 10% on the X _or_ Y axis. Once out of the dead zone, both axes register precise motion, so it is a hybrid between the circular and the horrible cross-shaped dead zone. An advantage of this type of dead zone is that it doesn't require any floating point calculations.

[image]

a-kring, 2020-07-17:
While testing the Stadia controller on Linux, I found that by default it has that same ugly cross-shaped dead zone that the HID input drivers add. After I disable it using `evdev-joystick` and `jscal`, there is still a small square dead zone at about ~10% that seems to be implemented at the hardware level. It behaves in the same way that it does on Windows.

kakra, 2020-07-17:
This is another case of different layers doing different things they should not do. SDL should probably not mess with the axes and button mapping, as different drivers with different mappings sometimes expose the same controller UID to SDL. Instead, that should be fixed in the drivers itself - they need to expose a commonly agreed standard.

Now we have the next one: `evdev` exposes a flatness property but doesn't apply it (I tried in `evtest` and can see tiny movements within the dead zone, in contrast to `joydev` where I don't see it), so upper layers can decide how to use it. But then there's `joydev` which does this already at the kernel level by using the `evdev` property, or some hard-coded values for non-HID joysticks. Then there's SDL which probably also does it, and you never know for sure if it uses `joydev` or `evdev`. The Browser API uses `joydev`, so you should see the flatness applied. It only falls back to `evdev` if no associated `joydev` device is found - which is bad since both APIs have a different exposure of the dead zone.

I'll look at SDL next to see if it applies the dead zone. If it doesn't, that's probably a good thing. In that case, Wine/Proton should also not do this.

But if I implement a dead zone in xpadneo while telling upper layers that our device has/needs no dead zone (by setting `flatness` to 0), everything should be fine. The driver will use the correct 2D plane implementation. Whatever upper layer looks at the flatness parameter should just do nothing with the values, and we have a consistent view in `joydev` and `evdev`. I think I will push my branch in a few hours so you could test it.

a-kring, 2020-07-17:
Excellent!

I think that Google nailed with the Stadia controller in this department. they implemented a 10% dead zone as a small square (no floating point arithmetic needed). Input value jumps to 10% immediately after the stick leaves the dead zone, which means that if an upper layer library or program adds its own 5% dead zone, it will not add to the 10% dead zone that is already there. It doesn't lead to smooth small inputs when the dead zone is cleared though, but it is still a big improvement over cross-shaped dead zones. If a game has a matching 10% dead zone, it will work nicely with that controller.

kakra, 2020-07-17:
Yeah, I experimented with a square dead zone but it had some strange behavior... Since a round dead zone cannot be implemented with at least one call to `sqrt()`, my current implementation just pretends that I can subtract the values squared (without calculating the roots first) to just calculate a linear scale from it, but the result actually forms circular movement back into a square. I'd like to move this discussion over to the issue report in xpadneo so not to hijack this any longer.

kakra, 2020-07-17:
> I think that Google nailed with the Stadia controller in this department. they implemented a 10% dead zone as a small square

I tried the Stadia controller: It fooled you, it has a resolution of only 256 positions for the thumb sticks. That makes a very tiny flatness value of 5%. The dead zone seems axial when looking at the output of `evtest`. But the idle point is still very tight on that, time will show how it works when the sticks are worn out.

kakra, 2020-07-17:
@medusalix We settled on the following solution: The defaults were changed to use 25% smaller dead zones and a fuzz value smaller by a factor of 8, for the thumb sticks. This is already a huge improvement. The triggers do not need dead zone handling at all. Additionally, there's now an option to disable the dead zone completely which puts the controller on par with Windows when it comes to precision. This is due to some short-comings in the code path from Wine through SDL down to the Linux device, there's currently no way for Wine games to manage dead zones on its own (but they expect that and thus dead zones are applied twice). So the current solution is to disable dead zone handling at the driver level. Since that messes with joydev, it's a switch that is off by default.

x4mers, 2020-07-18:
>  Additionally, there's now an option to disable the dead zone completely which puts the controller on par with Windows when it comes to precision. So the current solution is to disable dead zone handling at the driver level. Since that messes with joydev, it's a switch that is off by default.

Is this controllable by the "Set Launch Options" section in Steam, to let it be controlled on a game by game basis like Mesa does for LLVM/ACO, or it's hard set at boot?

kakra, 2020-07-18:
@x4mers First, let's clean up some possible confusion: The quoted solution in your comment is only implemented in xpadneo as of yet. xow does not have it yet.

To answer your question, simple answer: no. Complicated answer: maybe. Maybe means, that this needs support from Proton. Proton would need to have a way of controlling weather it wants to receive input events with dead zones from the driver. Currently it cannot do this because SDL does not support this. OTOH, on the other hand, it could do this in a reverse way: You could turn dead zones off in the driver, and Proton would need support to apply dead zones itself depending on some condition set (and it would need to do that in different drivers). There's a way to set dead zones in the registry so you could already force dead zones per game maybe. But this may not apply to every driver a game could use. Explicitly, xinput games must receive input events without dead zones as this is clearly documented by MS to work that way. I'm not sure if this is also true for dinput, and games using the old winejoystick.sys driver can probably set the dead zones themselves. So if you're playing only Proton games, you should be absolutely fine with using no dead zones at all at the driver level. Most games that are Windows but also have a native Linux version, probably also have their own dead zone handler. Only Linux games using SDL (without having Windows in mind as the primary platform) or using the `joydev` interface may expect a dead zone, and this would be missing if turned off at the driver level.

Also, this can only be controlled at driver load time as we cannot change the dead zones after the device was detected. So for xow that would need a restart of xow to control it, for xpadneo that would need reload of the module, so you could be creative and find a way to restart xow / reload the module at launch time of the game.

a-kring, 2020-07-18:
By the way, when I mean that the Stadia controller implements a 10% deadzone in hardware, this is how I tested:

Using the [Gamepad Tester](https://gamepad-tester.com/) site on Windows you can check yourself. It's pretty evident. I could try recording a video. As soon as you pass 10% deflection on any axis you get a bit above 10% input on that axis, and a value under 10% for the other axis. The only way to get input below 10% on any axis is when the other axis has above 10% deflection. This applies to both thumb sticks.

Linux does apply the dreaded cross-shaped axial dead zones by default. I disabled them using `evdev-joystick` and `jscal`, and then tested to the same results that I got on Windows. I _assume_ it is done by hardware and not by any sort of software driver, considering that no other gamepad driver for Linux behaves in this manner.

kakra, 2020-07-18:
I used `evtest` which isn't graphical but the numbers that were returned do not suggest that a square dead zone is applied, it seems axial. But my observation may come just from the fact that I had a hard time to get a fine grained value between 0 and 255 anyways, it seemed more like it jumps by values of at least 2 to 4 along both axes, if you double that into both directions, we are already near the dead zone value. I almost never use the Stadia controller, so the idle point is still extremely tight on mine. I could imaging that the controller maybe uses some auto-calibration, or that we use just different firmware versions. IOW, I didn't see evidence of a dead zone in `evtest`, and Linux claimed to use a dead zone of 5% - so my dead zone was about the size of the fuzz I was seeing. But 5% also does not support your observation. So, no idea why we see a difference.

a-kring, 2020-07-19:
Maybe there are differences between versions of the Stadia controller.

I set mine to run this script when it is plugged in:

```sh
#!/bin/sh

jscal -s 8,1,0,128,128,4227201,4227201,1,0,128,128,4227201,4227201,1,0,128,128,4227201,4227201,1,0,128,128,4227201,4227201,1,0,127,127,4227201,4194176,1,0,127,127,4227201,4194176,0,0,0,0 /dev/input/by-id/usb-Google_LLC_Stadia_Controller_rev._A_99040YCAC2FY98-if01-joystick

evdev-joystick --evdev /dev/input/by-id/usb-Google_LLC_Stadia_Controller_rev._A_99040YCAC2FY98-if01-event-joystick --deadzone 0
```

This is the corresponding udev rules file:

```
SUBSYSTEM=="input", ATTRS{name}=="Google LLC Stadia Controller rev. A", ACTION=="add", RUN+="/bin/sh /usr/local/bin/stadia-jscal.sh"
```

If I move the stick up, the first reading from Gamepad Tester comes at around `-0.11026`. If I move it down I get a reading at `0.12598`, sometimes at around `0.13385`. If I keep pushing the reading changes at intervals of roughly 0.005 until it reaches either 1.00000 or -1.00000. The X axis gives small readings immediately after the Y axis reads anything above 0.10000. Same for the Y axis with respect to X. That is: as soon as one axis clears 0.10000, the other axis gives precise readings, even below 0.10000, and it doesn't seem to _magnetize_ to the axis line like it does when it has an axial deadzone configured. This is the string that I see on the Gamepad Tester when the controller is connected: "Google LLC Stadia Controller rev. A (STANDARD GAMEPAD Vendor: 18d1 Product: 9400)".

kakra, 2020-07-19:
This is getting off-topic, we should not continue discussing the Stadia controller here. I made a Discord server for xpadneo, we can move over there: https://discord.gg/6RkrZzg

kakra, 2020-08-17:
This may be obsolete by https://bugzilla.libsdl.org/show_bug.cgi?id=5241

kakra, 2020-12-17:
This will be fixed in the next SDL2 version, no more adjustments needed: https://bugzilla.libsdl.org/show_bug.cgi?id=5241#c17

closed 2021-01-25 by medusalix