medusalix/xow #106: How to figure out event device name?

peci1, 2020-07-15:
Hi, I'd like to programatically detect the name of the event device. With e.g. Logitech F710, there is a record in `/dev/input/by-id/`, but this driver leaves no such clue... I've only found directory `/sys/devices/virtual/input/input89/event6`, which ends in `event6` which happens to be the name of the event device. Is this a correct way to detect the event device? Or is there a better one?

peci1, 2020-07-15:
If I connect the gamepad via microusb, there is:

```
$ ll /dev/input/by-id/
drwxr-xr-x 2 root root 120 čec 15 02:18 ./
drwxr-xr-x 4 root root 540 čec 15 02:18 ../
lrwxrwxrwx 1 root root   9 čec 15 02:18 usb-Microsoft_Controller_3039363430373633383532393436-event-joystick -> ../event6
lrwxrwxrwx 1 root root   6 čec 15 02:18 usb-Microsoft_Controller_3039363430373633383532393436-joystick -> ../js0
```

kakra, 2020-07-15:
You can use `udevadm test <PATH>` to extract all sorts of information as environment variables. Does that help?

But yes, the final `event*` matches - it's a globally (in scope of your system) unique name generated at udev detection time but it is not persistent across reboots (although it may look like because your system doesn't change a lot).

medusalix, 2020-07-15:
It's actually pretty simple with Python:

```python3
import evdev

devices = [evdev.InputDevice(path) for path in evdev.list_devices()]

for device in devices:
	print(device.path, device.name, device.info)
```

Output:

```
/dev/input/event24 Xbox One Wireless Controller bus: 0003, vendor 045e, product 02ea, version 0408
```

peci1, 2020-07-15:
Wow, that's super simple :) Is there a way to map these event devices to corresponding `/dev/input/jsX`? My use-case is to let the user provide path to the `/dev/input/jsX` and then automatically search for the related event device.

medusalix, 2020-07-15:
Then you'd want to use `pyudev` instead:

```python3
from pyudev import Context, Devices

context = Context()
device = Devices.from_device_file(context, '/dev/input/js0')

for child in device.parent.children:
    if child.sys_name.startswith('event'):
        print(child.sys_number)
```

peci1, 2020-07-15:
Great, thanks for the help!

closed 2020-07-15 by peci1