medusalix/xow #108: xow breaking steam proton games while running

avatar4ik3, 2020-07-19:
I just realized that when xow is active it brakes games. Game starts to freeze lag or just not work any good when if xow is not active everything is ok. I'm noob in linux, and i could misunderstand something.
i'm using Pop! os 20.04

kakra, 2020-07-19:
You could run `atop -w stats.atop` before starting the game, then play the game reproducing the problem, then stop atop and post the result file `stats.atop`. You could repeat the same for running without xow and post another file. Maybe we can find a difference there.

Warning: atop collects a lot of system performance data about from running processes. If you feel like not posting it, feel free to read the atop results yourself and compare the differences: `atop -r stats.atop -ARps`.

avatar4ik3, 2020-07-19:
I switched system to KDE neon but the problem still here. I also do what you recommended but i don't understand what should i looking for in those files. So i attached them.File Wired.atop was logging game(?) without any troubles and file Wireless.atop did opposite.
[Рабочий стол.zip](https://github.com/medusalix/xow/files/4944872/default.zip)
 I also attached screenshots for wired and wireless actions

Wired
[image]

Wireless 
[image]

kakra, 2020-07-20:
In wireless mode, there seems to be a lot more blocking on syscalls (when loading the samples into atop, you can press "t" to walk through the time intervals of 10s to find peak usages):
```
cpu | sys      32%  | user     68% | 
```

I cannot see such a situation for the wired mode. The rest of the system usage looks identical. So maybe there's some lock fighting on a single kernel resource. Or it is locking too much within wineserver, wireless shows higher CPU usage for winedevice.exe. wineserver is single-threaded, so if any syscall blocks for too long, wine will lag.

Also, the game looks like it's running on only 2 cores mostly. Does this problem persist with other games or a different Proton version? Which libusb version are you using?

@medusalix Is there anything that could cause high sys time in the kernel? Maybe due to accessing libusb?

avatar4ik3, 2020-07-20:
I'v checked l.a.noire and Bioshock infinite and they work fine with xow. I belive something wrong with project cars 2 and not with xow, but i only checked 3 games.

I'v also checked game cpu behavior in project cars 2 via DXVK_HUD and noticed that lag starts only when car speed is changing rapidly.

[wireless.zip](https://github.com/medusalix/xow/files/4948890/wireless.zip)
Also I checked wireless.atop again and noticed : cpu frequency not stable and when it is low wineserver takes more cpu power
And pCARS2AVX.exe reqieres > 100% of cpu wtf?

i'm using libusb-1.0-0-dev and i tried libusb-1.14(with manjaro) but it didn't help

kakra, 2020-07-20:
More than 100% just means the the games runs on multiple threads, this is most common for DXVK games. There's difference: Linux reports CPU usage per core, Windows reports it across all cores. That's probably because Linux from the beginning supported massive-multicore-systems much better, and a usage value per core is just more readable. So if you are seeing a process with >100% CPU usage in Linux, that's multi-threaded across multiple cores. In Windows you'll never see such a value: A game would need to use 100% of all cores in a 64-core system to report 100% usage. Or from the other perspective: Games which use 3-4 cores in Windows probably hover around 4-5% usage which is not a very useful value, while in Linux they report 300-400%.

wineserver is a single-threaded process with a global lock that acts as a relay for kernel-level calls in Wine to Linux. If it uses too much CPU, a lot of expensive relay work is going on and the game blocks on the thread which initiated that work (directly or indirectly).

You may want to report your findings with Project CARS 2 to the Proton issue tracker, as it points to a problem that may be solved in Wine. Since other games you tested don't show that behavior, it's probably not to blame on xow. The atop results support that observation if we can exclude libusb as the problem.

kakra, 2020-07-20:
> Also I checked wireless.atop again and noticed : cpu frequency not stable and when it is low wineserver takes more cpu power

Yes, that's expected: wineserver is probably having quite constant work to do, so when CPU frequency goes down, it's relative usage goes up. But wineserver doesn't do busy waiting in most places, so its CPU usage doesn't scale linearly with the work put on it, actual work and waiting is done at lower level (towards the Linux kernel), it just relays work and waits for the result. But it only relays one work at a time and waits for it because it's single threaded, all queued work waits on wineserver before it is processed. So if you are seeing high CPU usage in wineserver, it's actually not waiting a lot and does a good job.

medusalix, 2020-07-24:
> But that wouldn't explain why it only hits in the wireless case.

Might be another symptom of my poorly implemented force feedback handler. The controller probably rumbles a lot when you accelerate in Project Cars :smile:.

avatar4ik3, 2020-07-24:
Seems like it, because if i choose input for wheel(and continue usung xbox one controller) and set rumble(AKA force feedback) to 0 all ok. Problem solved i guess

medusalix, 2020-07-24:
@avatar4ik3 Can you test if PR #100 solves your issues?

avatar4ik3, 2020-07-24:
how should i test it?

avatar4ik3, 2020-07-24:
I mean how could i even download PR and build it

kakra, 2020-07-24:
I think you can just append .zip to the URL.

avatar4ik3, 2020-07-24:
I just downloaded @kakra forked repository and instlaled it. And all ok - game running as intended. So thanks guys.

closed 2020-07-24 by avatar4ik3

kakra, 2020-07-24:
@medusalix So +1 point for #100, I guess