medusalix/xow #109: Light on the dongle turns on, but controller won't connect (Raspberry Pi 4)

adwd33, 2020-07-24:
Prior to installing xow, the light on the dongle wouldn't light up at all when plugging into any of the Pi's USB ports. (I didn't even realize it was a light at first; it was only when I tried setting it up on my Windows PC that it came on.) After installing xow, the light DID come on, which gave me hope, but I haven't managed to get further than that.

When pressing the button on the dongle when it was attached to my PC, the light was blinking and only stayed on once it had paired with my controller. When it's plugged into the Pi, it turns on and stays on; I can't get it to start blinking.

Pressing the button on the dongle and trying to pair it to the controller like I did on my PC didn't work. Sending the SIGUSR1 signal in lieu of pressing the button on the dongle also didn't work. What should I try next? What logs should I try to obtain?

The Xbox One controller I'm using is the one that came with my Oculus Rift. From what I've read, going by the design of the plastic, it's one of the early ones that doesn't use Bluetooth, hence why I can't simply pair it with the Raspberry Pi directly.

Thank you ever so much for any help you can offer!

grant2258, 2020-08-03:
this worked fine for me on the pi. All i done was send the SIGUSR1 pressed the wync button and pressed the xbox one button and i was good to go.

medusalix, 2020-08-04:
> What logs should I try to obtain?

@adwd33 The logs from the system journal would be a good start (you can obtain them with `journalctl`). If those don't show any obvious issues, a Wireshark packet capture will be required.