medusalix/xow #111: Using controller causes games to pause or enter inventory screen every few seconds.

Nikorasu, 2020-07-30:
I'm having a strange issue when I try using my controller for games, which doesn't happen if I use a keyboard..
Once I get into gameplay, every few seconds either the game pauses itself, or enters an inventory screen, depending on the game. I've tried remapping the controls from within those games, but that doesn't seem to fix it.. And I'm relatively inexperienced with Linux, so I'm not exactly sure what to check..
I'm on Linux Mint 20, using an Xbox One wireless controller, Model 1708. And I think dongle model # says 1713, but it's almost too tiny to make out and has Chinese writing in/near it.
Maybe I'm just missing some simple configuration or setting to fix this.. but I just don't know. If necessary, I'll be happy to check logs or something that outputs button presses.

x4mers, 2020-07-30:
> .., every few seconds either the game pauses itself, or enters an inventory screen, depending on the game. I'm relatively inexperienced with Linux, so I'm not exactly sure what to check..I'm on Linux Mint

On Mint, run software manager from your start menu, and use the top right search box to look for jstest & install jstest-gtk.

It will be in the administration section in your start menu after install. Run that and look at the boxes down the right side of the interface, numbered 0-10. They should turn black as you hit each button. Judging by what's happening to you, I would suspect button 6 or 7 to be faulty. Are any of them flickering black/white without hitting a button?

Try putting some pressure/flex on the controller housing while moving the sticks around(like you would while playing a game). Do any of the boxes flicker black while doing that?

Nikorasu, 2020-07-30:
Ok, I've done that, and so far nothing appears to be faulty in there. No flickering buttons, 6 and 7 are working fine and don't activate when I'm not pressing them.. Even if I'm a little rough on the controller, it's not causing anything to misbehave.

kakra, 2020-07-30:
Do you use rumble? The behavior may result from intermittent disconnects, and PR #100 may then solve this for you.

Nikorasu, 2020-07-30:
Nope, I haven't had a chance to use rumble for anything yet, tho I dono if it's still enabled by default or something. Maybe I can disable it to rule it out.. hm

x4mers, 2020-07-30:
Do the games in question work fine, if you have the same controller plugged in via USB cable instead of wireless?

x4mers, 2020-07-30:
> Unfortunately no, the issue still happens, even if I unplug the wireless dongle, and plug the controller in directly with a usb cable.

While plugged in, it uses the kernel xpad driver not xow, so this is not an issue with the xow project.

 I'd try it wired on a different computer if you have one available to you. If you don't, try wired on different USB ports to see if it works on any. If you have available USB 1.1/2.0 ports on the back of your PC and wired passes there, plug the wireless adapter in there.

Nikorasu, 2020-07-30:
Well, darn this is so hard to narrow down.. I'm fairly certain the controller works fine, as I'm able to use it in Windows 10 without issue, as well as when paired with other devices like an Oculus Go.
So I'm not sure where the issue actually is.. Is there some setting in Linux that could be interfering? I tried running the games in with gamemode enabled and got the same issue. :/
I wonder if I need to monitor something while playing one of these games to see where the command that's pausing it is originating from, but I'm not exactly sure how to go about doing that..

Nikorasu, 2020-07-31:
For now, I assume my issue isn't with this driver, but rather a compatibility issue with certain games I was testing on. I switched to testing it on a couple native linux games, and those seem to be working MUCH better.. I also switched the dongle to a USB3 port, and that also seems to have improved things for at least one game that was seeing this issue, oddly.
So should I just close this issue?

x4mers, 2020-07-31:
If all your native Linux games, as far as controller goes, work flawlessly (and only then), and Proton ones don't then it may possibly be a proton issue to discuss there. However, you should close this issue as it's clearly not a xow issue.

 Mint 20 is recent. Upgrade or clean install? Have you checked their support forums to see if others are having problems with controllers?

Xow has issues with some USB 3 controllers' handling by the OS. That's why I mentioned to try 1.1/2.0 ports earlier. I would suggest, since your controller doesn't work wired either, to proceed via wired with troubleshooting posts to other support venues.

Good luck.

closed 2020-07-31 by Nikorasu

kakra, 2020-07-31:
Just one more idea: You should try disabling the Steam configuration support for your controller, I think that's most easily done from big picture mode. Just remove the check mark from your Xbox controller and it will no longer be passed through the controller configuration layer of Steam. Maybe that solves your problem.