medusalix/xow #112: Dongle works but it is messing my system: LIBUSB_ERROR_BUSY

luissantos27, 2020-08-05:
I have installed the dongle and it works with no problem. However, I have other systems that use Serial communication that stopped working: 

`18:30:34.317 [3876] [E] Device-85E0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)`

Torrenator, 2020-08-30:
I also seem to be running into this issue. I can not figure out what is causing the device to be busy. I have not manually installed any of the other xboxdrv or xpad packages. 

Running Arch Linux on kernel 5.8.4

Torrenator, 2020-08-30:
I fixed it for me. Still don't exactly know what is going on. But after a restart of my system it still would not work the first time. But then it gave me the Mt76Exception error. After following the steps as described on the readme page (take out the dongle and restart xow service), the service would start and pick up the dongle as expected.

Sorry for any inconvenience about me jumping to conclusions to fast, should have restarted first.

matdave, 2022-02-13:
> take out the dongle and restart xow service
derp, that fixed it for me