medusalix/xow #118: Model 1708 Connects to Wireless Dongle, Disconnects When Headphones ArePlugged In

arlenreyb, 2020-09-08:
I'm using the 2016 model controller (1708). I purchased the controller and usb adapter a few days ago, so they're new. Following the instructions, and using the terminal to put the dongle in pairing mode (the button on the adapter doesn't do anything), it will connect to the controller and pair. The controller will work fine, until you plug in any standard headphones in to the headphone jack (note; I'm not using a fancy headset or snything, just a standard pair of in-ear headphones and the headphone jack on the controller).

Once you plug in headphones, the controller will permanently disconnect (after about 3 seconds). Removing the headphones does nothing, it remains disconnected. Turning the controller off and back on does nothing, it will not reconnect whether the headphones are plugged in or not. Stopping/restarting the service, and turning the controller off and back on is the only way to reconnect the controller (it doesn't forget its pairing when it disconnects, but the service itself needs to be restarted).

Additionally, paring will not work at all if headphones are already plugged in; it will never find the controller.

Bluetooth connection works, but as the controller's instruction manual will tell you, audio through the controller will NOT work if you connect via Bluetooth. Audio will only pass through if you use the USB adapter, and I've confirmed that it works on my Windows 10 system, so I'm sure the issue is not with the controller, adapter, or headphones.

The main page for the project mentions that "Audio Jack" is "Working" in model 1697, but no mention to the headphone jack or to anything audio related is made anywhere else, neither confirming that it works nor explaining that it's not supported, so I'm not entirely sure if this driver even offers audio-through-usb-adapter functionality in the first place.

Using Ubuntu 20.04, and I've tested multiple times, using multiple USB ports. It's a consistent issue on my system.
Any ideas, or suggestions for places to look for more information? I'm still a bit new to Linux, so you might have to be specific.

transcengopher, 2020-09-27:
It may be either the same as, or a consequence of, #4.

Anyhow, I've got what looks to be the same issue, and this is what gets logged after I plug anything into the jack on controller:

```
Sep 27 21:27:23 user xow[11593]: 2020-09-27 21:27:23 INFO  - Battery level: 2
Sep 27 21:31:38 user xow[11593]: 2020-09-27 21:31:38 INFO  - Device announced, product id: 02f6
Sep 27 21:31:38 user xow[11593]: 2020-09-27 21:31:38 DEBUG - Firmware version: 3.1.1221.0
Sep 27 21:31:38 user xow[11593]: 2020-09-27 21:31:38 DEBUG - Hardware version: 1284.1.1.1
Sep 27 21:31:38 user xow[11593]: terminate called after throwing and instance of 'InputException'
Sep 27 21:31:38 user xow[11593]:   what():  Error adding key code: Invalid argument
Sep 27 21:31:38 user systemd[1]: xow.service: Main process exited, code=killed, status=6/ABRT
Sep 27 21:31:38 user systemd[1]: xow.service: Failed with result 'signal'.
```

adonisd, 2020-09-28:
Same. However am not sure audio is compatible
Looking at README.MD the only one that implicitly declares audio compatibility is model 1697

Anyhow, same exact error as @transcengopher :
```

Sep 28 14:39:21 adonis-desktop systemd[1]: Started Xbox One Wireless Dongle Driver.
Sep 28 14:39:21 adonis-desktop xow[43932]: 2020-09-28 14:39:21 INFO  - xow v0.5-16-ga3d2b01 ©Severin v. W.
Sep 28 14:39:21 adonis-desktop xow[43932]: 2020-09-28 14:39:21 INFO  - Waiting for device...
Sep 28 14:39:21 adonis-desktop xow[43932]: 2020-09-28 14:39:21 INFO  - Wireless address: 62:45:b4:e6:26:ad
Sep 28 14:39:21 adonis-desktop xow[43932]: 2020-09-28 14:39:21 INFO  - Dongle initialized
Sep 28 14:39:21 adonis-desktop xow[43932]: 2020-09-28 14:39:21 INFO  - Controller '1' connected
Sep 28 14:39:23 adonis-desktop xow[43932]: 2020-09-28 14:39:23 INFO  - Controller '2' connected
Sep 28 14:39:24 adonis-desktop xow[43932]: 2020-09-28 14:39:24 INFO  - Device announced, product id: 02ea
Sep 28 14:39:24 adonis-desktop xow[43932]: 2020-09-28 14:39:24 INFO  - Serial number: 02SH0542655736
Sep 28 14:39:38 adonis-desktop xow[43932]: 2020-09-28 14:39:38 INFO  - Device announced, product id: 02f6
Sep 28 14:39:38 adonis-desktop xow[43932]: terminate called after throwing an instance of 'InputException'
Sep 28 14:39:38 adonis-desktop xow[43932]:   what():  Error adding key code: Invalid argument
Sep 28 14:39:38 adonis-desktop systemd[1]: xow.service: Main process exited, code=killed, status=6/ABRT
Sep 28 14:39:38 adonis-desktop systemd[1]: xow.service: Failed with result 'signal'.
```

relkochta, 2020-10-23:
I can confirm the same issue on Gentoo Linux.

`2020-10-23 09:14:23 INFO  - xow v0.5-17-gaf5b9d4 ©Severin v. W.
2020-10-23 09:14:23 INFO  - Waiting for device...
2020-10-23 09:14:24 INFO  - Wireless address: 62:45:bd:03:eb:d7
2020-10-23 09:14:24 INFO  - Dongle initialized
2020-10-23 09:14:32 INFO  - Controller '1' connected
2020-10-23 09:14:33 INFO  - Device announced, product id: 02ea
2020-10-23 09:14:33 INFO  - Serial number: 02ED0050922029
2020-10-23 09:14:36 INFO  - Battery level: 3
2020-10-23 09:15:26 INFO  - Device announced, product id: 02f6
terminate called after throwing an instance of 'InputException'
  what():  Error adding key code: Invalid argument
`

transcengopher, 2020-11-07:
I've come bearing some debugging news.

The error about "adding key code" is thrown from `InputDevice::addKey`, and the `code` seems to always be `0x13c`, which is the `BTN_MODE` and is added among the first keys when input device is initialized right after announcement.

Since logs always seem to have two "Device announced" statements (see mine and @relkochta snippets), I can only assume there is some kind of double initialization going on.

Oh, and each announcement has different `deviceId` and `productId` too.

kakra, 2020-11-08:
@transcengopher Interesting observation. Does this happen when you plug the headphone jack?

transcengopher, 2020-11-08:
@kakra, yes, there is second announcement when the jack is plugged in.
I also tested by tuning the gamepad off, plugging the jack in and then turning it on like that - there are still two "Device announced" events in logs.

kakra, 2020-11-08:
Yeah, that supports my research I've done for another project where I've read by coincidence that the audio part of the connection is exposed as a separate device. xow should probably just skip initializing input mappings on that device. Modern kernels complain when you do such "non-sense".

transcengopher, 2020-11-08:
@kakra, well, hey - if audio stuff is exposed on as another device, it should simplify the driver architecture in the sense that if audio is handled by userspace service, then each service gets its own device to work with.
I'll try to look into how to dump both events and see if there's anything concrete like device capabilities reported to make skipping announcements more targeted.

Might take a while as I'm a Java dev and don't remember a lot from how modern C works.

kakra, 2020-11-08:
@transcengopher It may still be an "input device" in that sense that it creates input events like "plugged" and "unplugged". Monitor hot-plugging and audio jack plugging in Linux is generally implemented that way: Such devices create an input device which has only those events. Trying to map anything else may fail on modern kernels.

Also, it may be a composite device: The Linux input layer supports input devices to consist of multiple sub-devices, e.g. the Xbox Elite 2 controller exposes itself as a gaming device, a keyboard, and a "consumer control" device (for the Xbox logo button and some non-gaming misc functions) when connected via Bluetooth. When you try to "double map" an input event on the components, the kernel may return that error. So - without having looked at the code - xow should probably map events only matching the component that was detected.

It may need some code re-design to register it as a proper composite input device, I'm sure xow currently exposes only one single device but when it detects the second component, it re-runs the whole input mapping. It should probably just be filtered by PID for now.

In xpadneo I'm currently working on splitting the single device node into multiple sub-devices properly. This also works around some quirks in libinput which may detect such composite devices exposed as a single device wrongly as a touch pad, other software like SDL makes similar assumptions, or may see unexpected key-codes, then being unable to bind them to some actions (e.g., I think RetroPi is such a software which doesn't expect keyboard events from a gaming device and thus you're not able to bind them). For simplicity, we currently do not split the components in xpadneo although we probably should.

kakra, 2020-11-08:
I'd probably start here:
https://github.com/medusalix/xow/blob/af5b9d477706874797ba1bece0282d00225b742b/controller/controller.cpp#L115

Just put a switch-case statement around it querying the PID detected. I'll leave it as an exercise to you to figure out how to properly get the correct PID there. ;-) I'm currently lacking some time.

A better solution could be to split that into two classes. Actually, the abstraction could need some improvement here and as far as I remember, that's on the todo list of @medusalix.

kakra, 2020-11-08:
> well, hey - if audio stuff is exposed on as another device, it should simplify the driver architecture in the sense that if audio is handled by userspace service, then each service gets its own device to work with.

You cannot properly register real ALSA devices from user space. You'd need to write a plugin for every audio solution out there which creates a lot of unwanted dependencies (pulseaudio, libalsa), and also needs a lot of complicated IPC for communicating with a single main service from the various clients. And if that's not enough, you'd need to handle the service going away under your feet (e.g., because it crashed). Such events may crash your applications or whole desktop. IOW, you'd need to reinvent the wheel. For proper audio support, the driver has to be implemented in the kernel which creates a whole lot of other headache, e.g. there's no driver for the dongle (well, there is but not for handling the GIP protocol) and there's a proprietary firmware blob needed which has uncleared licensing issues.

transcengopher, 2020-11-09:
@kakra, I'd personally go for https://github.com/medusalix/xow/blob/af5b9d477706874797ba1bece0282d00225b742b/controller/controller.cpp#L62

And when product id is `02f6`, init a `SoundDevice` (dummy entity for now). Though I'm not certain different models announce same product ids, so this might only work for my controller model and no other.

kakra, 2020-11-09:
You can get a list of known device IDs from hid-ids.h in der Linux kernel. They seem to be identical.

kakra, 2020-11-09:
But maybe it's better to work with an inverse logic and skip 0x2f6 for InputDevice, and accept all other PIDs that may emerge. I'd not expect the audio PID to change between models...

mabasic, 2020-11-15:
I am also having this issue, maybe I can help? 

I have some experience in c and cpp years ago, but zero experience in building drivers. 

If someone would point me in the right direction I could probably help somehow. 

I've already skimmed through this issue thread, but I will read it more detailed later.

I'm currently having different issues, once I solve them I will get back to this.

kakra, 2020-11-16:
There are already links to the code above.

transcengopher, 2020-11-21:
Another bit of debugging news:
Once I stubbed the announcement of device by `productId: 0x02f6` (just logged&returned from the function), the drives stopped crashing on headphone plug. However, the log gets spammed with device announements at that point, which suggests that having the headphones (I think this is without mic) just plugged in constantly produces events that match condition below:
```
        frame->command == CMD_ANNOUNCE &&
        frame->length == sizeof(AnnounceData) &&
        data.size() >= sizeof(AnnounceData)
```

@kakra I don't even know what to make of this. Does it mean controller announces a sound packet or something?

kakra, 2020-11-21:
It may need some ACK sent back to the controller. I don't think this is properly reverse engineered yet.

medusalix, 2020-11-21:
Sorry for the silence on my part, it's been a busy time for me lately...

This is unfortunately a known issue that requires a little bit of effort to solve correctly.
Filtering the product id is not actually feasible, since the headset will just keep sending the announcements until you respond with the audio initialization command. Properly fixing this would require me to implement most of the audio stuff which I've determined not to be worth it for this driver.

> I'll try to look into how to dump both events and see if there's anything concrete like device capabilities reported to make skipping announcements more targeted.

The device identification command does exactly that (xow currently skips this).

transcengopher, 2021-01-20:
@medusalix , can you elaborate on why is it not worth fixing this? From looking at the code, I don't see how it might be too far gone (but then again, I only have a pretty vague idea of what needs to be done to address it).

If this is just the case of too much time commitment, can you maybe give me some pointers, so I can try making a stab at it in my forked repo? I'm quite new to the whole ecosystem and torn between various people saying this and that about whether programs should support ALSA, Pulse or something else entirely.

medusalix, 2021-03-11:
@arlenreyb @transcengopher This issue should be fixed by commit 9d5a11b4e8610b65581c6794069ef2e70fcefbd9. Any packets from accessories are now ignored to prevent xow from crashing when plugging in your headphones.

closed 2022-03-20 by medusalix