medusalix/xow #119: "libusb_hotplug_event" has not been declared

southside94, 2020-09-11:
Hello,

first sorry for bad english.

I installed a new raspbian on a Pi4.

When I am typing the command "make BUILD=RELEASE" the issue is coming.
I've update the whole system including libusb in every version.

I have the new "xBox wireless adapter for windows 10".

if you need more information please let me know and how to get the informations

[image]

kapsh, 2020-09-18:
@southside94 probably your system is missing libusb header files. Have you installed package libusb-dev (and everything else listed in https://github.com/medusalix/xow/blob/master/README.md#prerequisites)?

southside94, 2020-09-18:
This is what i have installed now

[image]

PhoneixS, 2021-05-17:
It happens to me too. I'm using ubuntu and I have also installed libusb-dev. @southside94 do you got it ok?