medusalix/xow #121: Controller keeps disconnecting, ultimately crashing xow due to LIBUSB_ERROR_NO_DEVICE

filipdutescu, 2020-09-20:
**Platform:** Arch Linux x64
**Kernel:** 5.8.8.arch1-1
**Version:** both `[xow](https://aur.archlinux.org/packages/xow/)` and `[xow-git](https://aur.archlinux.org/packages/xow-git/)` from the AUR
**Controller:** Xbox One Wireless Controller for Windows (or PC, however they brand it)

Was working no problems yesterday, today starting having this error, restarting xow, installing the git version and restarting the laptop did not resolve the issue. I run xow from the terminal (the service is disabled) and the following output is presented before crashing:

```console
2020-09-20 21:47:12 INFO  - xow v0.5-16-ga3d2b01 ©Severin v. W.
2020-09-20 21:47:12 INFO  - Waiting for device...
2020-09-20 21:47:24 INFO  - Wireless address: *removed by me, dunno if it matters or not*
2020-09-20 21:47:24 INFO  - Dongle initialized
2020-09-20 21:48:12 INFO  - Pairing enabled
2020-09-20 21:48:12 INFO  - Pairing enabled
2020-09-20 21:48:12 INFO  - Pairing enabled
2020-09-20 21:48:12 INFO  - Pairing enabled
2020-09-20 21:48:17 INFO  - Pairing disabled
2020-09-20 21:48:18 INFO  - Controller '1' connected
2020-09-20 21:48:18 INFO  - Device announced, product id: 02ea
2020-09-20 21:48:18 INFO  - Serial number: 02600074016931
2020-09-20 21:48:19 INFO  - Battery level: 2
2020-09-20 21:48:27 INFO  - Controller '1' disconnected
2020-09-20 21:48:33 INFO  - Controller '1' connected
2020-09-20 21:48:34 INFO  - Device announced, product id: 02ea
2020-09-20 21:48:34 INFO  - Serial number: 02600074016931
2020-09-20 21:48:35 INFO  - Battery level: 2
2020-09-20 21:48:54 INFO  - Controller '1' disconnected
2020-09-20 21:49:10 INFO  - Controller '1' connected
2020-09-20 21:49:11 INFO  - Device announced, product id: 02ea
2020-09-20 21:49:11 INFO  - Serial number: 02600074016931
2020-09-20 21:49:15 INFO  - Controller '1' disconnected
2020-09-20 21:49:23 ERROR - Error in bulk read: LIBUSB_ERROR_NO_DEVICE
2020-09-20 21:49:23 INFO  - Shutting down...
2020-09-20 21:49:23 ERROR - Error in bulk read: LIBUSB_ERROR_NO_DEVICE
2020-09-20 21:49:23 ERROR - Error in bulk write: LIBUSB_ERROR_NO_DEVICE
2020-09-20 21:49:23 ERROR - Failed to write command
2020-09-20 21:49:23 ERROR - Failed to set LED mode
2020-09-20 21:49:23 ERROR - Failed to turn off LED
2020-09-20 21:49:23 ERROR - Error in bulk write: LIBUSB_ERROR_NO_DEVICE
2020-09-20 21:49:23 ERROR - Failed to write command
2020-09-20 21:49:23 ERROR - Failed to set power mode
2020-09-20 21:49:23 ERROR - Failed to turn off radio
2020-09-20 21:49:23 ERROR - Error releasing interface: LIBUSB_ERROR_NO_DEVICE
```
I did not know you encourage the usage of it as a service, will try it out, but I thought maybe this is something you should look into, in case it is also a service problem.

kakra, 2020-09-20:
What does `dmesg` say? Maybe move over to a different USB port. For me, the dongle only works reliably in a USB hub by Amazon Basics, all the others failed (while they worked some kernel versions before). Also, USB ports tend to degrade electronically over time, especially if plugging devices in and out often. This may actually matter for some hardware (or cable lengths).

filipdutescu, 2020-09-20:
I never had a problem with that port before, as I have been using it for the dongle for about half a year, with most of it being in conjunction with xow. I ran it as a service and got the following message now:

```console
Sep 20 22:05:34 xow[430]: 2020-09-20 22:05:34 INFO  - Controller '1' disconnected
Sep 20 22:05:41 xow[430]: 2020-09-20 22:05:41 INFO  - Controller '1' connected
Sep 20 22:05:42 xow[430]: 2020-09-20 22:05:42 INFO  - Device announced, product id: 02ea
Sep 20 22:05:42 xow[430]: 2020-09-20 22:05:42 INFO  - Device announced, product id: 02ea
Sep 20 22:05:42 xow[430]: terminate called after throwing an instance of 'InputException'
Sep 20 22:05:42 xow[430]:   what():  Error adding key code: Invalid argument
```

`dmesg` is a foreign language to me, but here is what I found:

```console
[  222.353985] usb 1-3: new high-speed USB device number 7 using xhci_hcd
[  222.495261] usb 1-3: New USB device found, idVendor=045e, idProduct=02fe, bcdDevice= 1.00
[  222.495267] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  222.495271] usb 1-3: Product: XBOX ACC
[  222.495274] usb 1-3: Manufacturer: Microsoft Inc.
[  222.495277] usb 1-3: SerialNumber: 321522
[  222.650893] usb 1-3: reset high-speed USB device number 7 using xhci_hcd
[  222.867166] audit: type=1100 audit(1600628687.699:85): pid=1716 uid=0 auid=4294967295 ses=4294967295 msg='op=PAM:authentication grantors=pam_rootok acct="filip" exe="/usr/bin/su" hostname=? addr=? terminal=? res=success'
[  222.867353] audit: type=1101 audit(1600628687.699:86): pid=1716 uid=0 auid=4294967295 ses=4294967295 msg='op=PAM:accounting grantors=pam_unix acct="filip" exe="/usr/bin/su" hostname=? addr=? terminal=? res=success'
[  222.868320] audit: type=1103 audit(1600628687.699:87): pid=1716 uid=0 auid=4294967295 ses=4294967295 msg='op=PAM:setcred grantors=pam_rootok acct="filip" exe="/usr/bin/su" hostname=? addr=? terminal=? res=success'
[  222.868513] audit: type=1105 audit(1600628687.699:88): pid=1716 uid=0 auid=4294967295 ses=4294967295 msg='op=PAM:session_open grantors=pam_unix acct="filip" exe="/usr/bin/su" hostname=? addr=? terminal=? res=success'
[  223.171192] audit: type=1106 audit(1600628688.003:89): pid=1716 uid=0 auid=4294967295 ses=4294967295 msg='op=PAM:session_close grantors=pam_unix acct="filip" exe="/usr/bin/su" hostname=? addr=? terminal=? res=success'
[  223.171220] audit: type=1104 audit(1600628688.003:90): pid=1716 uid=0 auid=4294967295 ses=4294967295 msg='op=PAM:setcred grantors=pam_rootok acct="filip" exe="/usr/bin/su" hostname=? addr=? terminal=? res=success'
[  231.659277] input: Xbox One Wireless Controller as /devices/virtual/input/input35
[  246.104689] input: Xbox One Wireless Controller as /devices/virtual/input/input36
[  277.632503] input: Xbox One Wireless Controller as /devices/virtual/input/input37
[  278.124560] audit: type=1701 audit(1600628742.956:91): auid=4294967295 uid=64499 gid=64499 ses=4294967295 pid=430 comm="xow" exe="/usr/bin/xow" sig=6 res=1
[  278.133248] audit: type=1334 audit(1600628742.963:92): prog-id=15 op=LOAD
[  278.133316] audit: type=1334 audit(1600628742.963:93): prog-id=16 op=LOAD
[  278.133880] audit: type=1130 audit(1600628742.966:94): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@0-1954-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  278.427897] audit: type=1131 audit(1600628743.259:95): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=xow comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
[  278.471148] audit: type=1131 audit(1600628743.303:96): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@0-1954-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  278.592515] audit: type=1334 audit(1600628743.423:97): prog-id=16 op=UNLOAD
[  278.592520] audit: type=1334 audit(1600628743.423:98): prog-id=15 op=UNLOAD
[  294.731951] usb 1-3: USB disconnect, device number 7
[  307.019438] audit: type=1101 audit(1600628771.849:99): pid=1995 uid=1000 auid=1000 ses=2 msg='op=PAM:accounting grantors=pam_permit,pam_time acct="filip" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[  307.019581] audit: type=1110 audit(1600628771.849:100): pid=1995 uid=0 auid=1000 ses=2 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[  307.020842] audit: type=1105 audit(1600628771.853:101): pid=1995 uid=0 auid=1000 ses=2 msg='op=PAM:session_open grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[  307.029914] audit: type=1106 audit(1600628771.863:102): pid=1995 uid=0 auid=1000 ses=2 msg='op=PAM:session_close grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[  307.029958] audit: type=1104 audit(1600628771.863:103): pid=1995 uid=0 auid=1000 ses=2 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[  371.600033] audit: type=1101 audit(1600628836.429:104): pid=2019 uid=1000 auid=1000 ses=2 msg='op=PAM:accounting grantors=pam_permit,pam_time acct="filip" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[... cut ...] ev/pts/0 res=success'
```

I switched the port and it seems to work, but I did notice something weird, which from the issues you had might be important: the problems start when I start Sekiro: Shadows Die Twice with Steam Proton. The controller disconnects a few times, sometimes as soon as I start the game, other times after a bit.

Hope this helps. In any case, if the issue were the usb port, any advice on what to do in the future, as I am using a laptop, with a limited number of ports? Cheers!

kakra, 2020-09-20:
That may be related to the rumble protocol, try https://github.com/medusalix/xow/pull/100

filipdutescu, 2020-09-20:
Thanks a lot for the tips! Will try the suggested fixes, hopefully they work. 

Any ETA on merging the PR or when that version will be released?

kakra, 2020-09-20:
Not sure about an ETA... @medusalix may want to go for a different approach a redesign the driver architecture first. But I'll keep the PR updated on request or when I stumble across important updates. Just comment on the PR in that case. Let me know if it fixes things for you.

kakra, 2020-09-20:
FWIW, that PR is actually crafted from a working implementation I did for xpadneo, based on findings by @medusalix while experimenting with the rumble protocol. It successfully fixed the problem in xpadneo and I helped him out to make an identical fix for xow.

filipdutescu, 2020-09-20:
Thanks a lot. Will try tomorrow and see if anything changes.

filipdutescu, 2020-09-21:
@kakra seems like #100 fixed it, as far as I can tell, no problems even on the initial port. Dunno what you did, but you did great for me, as there was no disconnect at all. Hopefully @medusalix will merge your PR, it seems like a great improvement!

filipdutescu, 2020-09-21:
@kakra also, could you link this issue to your PR, to keep it in sync with the current fix?

kakra, 2020-09-21:
I'll put a close tag into the commit message - if that's what you mean.

filipdutescu, 2020-09-21:
I guess that should work, but in the side bar of the PR from GH UI, there is a section named "Linked Issues" (under assignees, labels etc.).

kakra, 2020-09-21:
Yeah but that function is working a little bit unfortunate... If this gets closed, it might close the PR, too. That function links the status of all issues together - this may not always be what we want. IMO, that should only be used for duplicates but here we want one issue to close another but not vice-versa. What linking actually does is: If one closes, it closes all the others, too. There's no single origin of status change.

kakra, 2020-09-21:
But I wonder why it doesn't say "may be fixed by #100" just below the title... It should to that.

filipdutescu, 2020-09-21:
Ah I see, that makes sense. Did not know that is how it worked, thanks for letting me know. That fact that, besides a "mention" in a commit, literally nothing happened is strange.

kakra, 2020-09-21:
Well, if you merge a PR, Github will look for tags in the commit descriptions. If one says "Closes:" or "Fixes:", it will close the references issue through the merge commit. It will also link each commit in the conversation time line. The sidebar link will link the status of reports into one: Change one, it changes the others. But a commit tagline will work just one way. So I prefer that.

Usually, when a commit tagline says "Fixes:", that should show up in the title as "May be fixed by PR ..." - but maybe that was just a UI experiment of Github (I activated some of those).

filipdutescu, 2020-10-18:
@kakra problem started appearing again, not entirely sure why. maybe the last kernel upgrade (for me 5.9.1-arch1-1) changed something?

kakra, 2020-10-18:
Your log shows that systemd dumped core - that should usually not happen. Do you see other processes doing that? Take a look at `coredumpctl`.

filipdutescu, 2020-10-19:
`coredumpctl` has the followin output:
```bash
Sun 2020-10-18 22:58:13 EEST    454 64499 64499   6 error     /usr/local/bin/xow
```

However, when trying to recreate the issue today, I could not do so, even after installing the current release of `xow`, rather than your fork.

kakra, 2020-10-19:
So you are seeing a different issue... Maybe some timing thing? I've updated my fork lately to use udev for loading the xow service, you should no longer have it enabled in systemd: `systemctl disable xow.service`. Then, with my current fork, it should load the service when you plug the dongle. However, I did not try that myself yet because I'm mostly only using xpadneo and Bluetooth connection.

filipdutescu, 2020-10-19:
I am now using this version, not the fork and today I did not have any problems whatsoever, so I have no clue. Your fix did however work, when this version failed and I did not updated it since you first told me to try it out, so it shouldn't be something you did. Might be something regarding timing, as on Windows I did not encounter problems.

medusalix, 2021-01-24:
The fixed rumble implementation is now included in `master`.

closed 2021-01-24 by medusalix