medusalix/xow #122: Xbox 360 and Xbox One Controller HID Report Descriptors

DJm00n, 2020-10-05:
While looking at original `xboxgip.sys` (version 10.0.18362.267) and how it works I found `gipControllerHidDescriptor` var that this driver uses for its HID Physical Device Object. It was [`HID_DESCRIPTOR`](https://docs.microsoft.com/windows-hardware/drivers/ddi/hidport/ns-hidport-_hid_descriptor) struct.
Stripped header and decoded it via [RDD! HID Report Descriptor Decoder](https://github.com/abend0c1/hidrdd) tool. Maybe could be userfull in this or other projects:
```cpp

//--------------------------------------------------------------------------------
// Report descriptor data in hex (length 262 bytes)
//--------------------------------------------------------------------------------

// 05010905 A101A100 09300931 150027FF FF000095 02751081 02C0A100 09330934
// 150027FF FF000095 02751081 02C00501 09321500 26FF0395 01750A81 02150025
// 00750695 01810305 01093515 0026FF03 9501750A 81021500 25007506 95018103
// 05091901 290A950A 75018102 15002500 75069501 81030501 09391501 25083500
// 463B0166 14007504 95018142 75049501 15002500 35004500 65008103 A102050F
// 09971500 25017504 95019102 15002500 91030970 15002564 75089504 91020950
// 66011055 0E26FF00 95019102 09A79102 65005500 097C9102 C0050109 80A10009
// 85150025 01950175 01810215 00250075 07950181 03C00506 09201500 26FF0075
// 08950181 02C0      

//--------------------------------------------------------------------------------
// Decoded Application Collection
//--------------------------------------------------------------------------------

/*
05 01        (GLOBAL) USAGE_PAGE         0x0001 Generic Desktop Page 
09 05        (LOCAL)  USAGE              0x00010005 Game Pad (Application Collection)  
A1 01        (MAIN)   COLLECTION         0x01 Application (Usage=0x00010005: Page=Generic Desktop Page, Usage=Game Pad, Type=Application Collection)
A1 00          (MAIN)   COLLECTION         0x00 Physical (Usage=0x0: Page=, Usage=, Type=) <-- Error: COLLECTION must be preceded by a USAGE <-- Warning: USAGE type should be CP (Physical Collection)
09 30            (LOCAL)  USAGE              0x00010030 X (Dynamic Value)  
09 31            (LOCAL)  USAGE              0x00010031 Y (Dynamic Value)  
15 00            (GLOBAL) LOGICAL_MINIMUM    0x00 (0)  <-- Info: Consider replacing 15 00 with 14
27 FFFF0000      (GLOBAL) LOGICAL_MAXIMUM    0x0000FFFF (65535)  
95 02            (GLOBAL) REPORT_COUNT       0x02 (2) Number of fields  
75 10            (GLOBAL) REPORT_SIZE        0x10 (16) Number of bits per field  
81 02            (MAIN)   INPUT              0x00000002 (2 fields x 16 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
C0             (MAIN)   END_COLLECTION     Physical 
A1 00          (MAIN)   COLLECTION         0x00 Physical (Usage=0x0: Page=, Usage=, Type=) <-- Error: COLLECTION must be preceded by a USAGE <-- Warning: USAGE type should be CP (Physical Collection)
09 33            (LOCAL)  USAGE              0x00010033 Rx (Dynamic Value)  
09 34            (LOCAL)  USAGE              0x00010034 Ry (Dynamic Value)  
15 00            (GLOBAL) LOGICAL_MINIMUM    0x00 (0) <-- Redundant: LOGICAL_MINIMUM is already 0 <-- Info: Consider replacing 15 00 with 14
27 FFFF0000      (GLOBAL) LOGICAL_MAXIMUM    0x0000FFFF (65535) <-- Redundant: LOGICAL_MAXIMUM is already 65535 
95 02            (GLOBAL) REPORT_COUNT       0x02 (2) Number of fields <-- Redundant: REPORT_COUNT is already 2 
75 10            (GLOBAL) REPORT_SIZE        0x10 (16) Number of bits per field <-- Redundant: REPORT_SIZE is already 16 
81 02            (MAIN)   INPUT              0x00000002 (2 fields x 16 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
C0             (MAIN)   END_COLLECTION     Physical 
05 01          (GLOBAL) USAGE_PAGE         0x0001 Generic Desktop Page <-- Redundant: USAGE_PAGE is already 0x0001
09 32          (LOCAL)  USAGE              0x00010032 Z (Dynamic Value)  
15 00          (GLOBAL) LOGICAL_MINIMUM    0x00 (0) <-- Redundant: LOGICAL_MINIMUM is already 0 <-- Info: Consider replacing 15 00 with 14
26 FF03        (GLOBAL) LOGICAL_MAXIMUM    0x03FF (1023)  
95 01          (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields  
75 0A          (GLOBAL) REPORT_SIZE        0x0A (10) Number of bits per field  
81 02          (MAIN)   INPUT              0x00000002 (1 field x 10 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
15 00          (GLOBAL) LOGICAL_MINIMUM    0x00 (0) <-- Redundant: LOGICAL_MINIMUM is already 0 <-- Info: Consider replacing 15 00 with 14
25 00          (GLOBAL) LOGICAL_MAXIMUM    0x00 (0)  <-- Info: Consider replacing 25 00 with 24
75 06          (GLOBAL) REPORT_SIZE        0x06 (6) Number of bits per field  
95 01          (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields <-- Redundant: REPORT_COUNT is already 1 
81 03          (MAIN)   INPUT              0x00000003 (1 field x 6 bits) 1=Constant 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
05 01          (GLOBAL) USAGE_PAGE         0x0001 Generic Desktop Page <-- Redundant: USAGE_PAGE is already 0x0001
09 35          (LOCAL)  USAGE              0x00010035 Rz (Dynamic Value)  
15 00          (GLOBAL) LOGICAL_MINIMUM    0x00 (0) <-- Redundant: LOGICAL_MINIMUM is already 0 <-- Info: Consider replacing 15 00 with 14
26 FF03        (GLOBAL) LOGICAL_MAXIMUM    0x03FF (1023)  
95 01          (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields <-- Redundant: REPORT_COUNT is already 1 
75 0A          (GLOBAL) REPORT_SIZE        0x0A (10) Number of bits per field  
81 02          (MAIN)   INPUT              0x00000002 (1 field x 10 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
15 00          (GLOBAL) LOGICAL_MINIMUM    0x00 (0) <-- Redundant: LOGICAL_MINIMUM is already 0 <-- Info: Consider replacing 15 00 with 14
25 00          (GLOBAL) LOGICAL_MAXIMUM    0x00 (0)  <-- Info: Consider replacing 25 00 with 24
75 06          (GLOBAL) REPORT_SIZE        0x06 (6) Number of bits per field  
95 01          (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields <-- Redundant: REPORT_COUNT is already 1 
81 03          (MAIN)   INPUT              0x00000003 (1 field x 6 bits) 1=Constant 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
05 09          (GLOBAL) USAGE_PAGE         0x0009 Button Page 
19 01          (LOCAL)  USAGE_MINIMUM      0x00090001 Button 1 Primary/trigger (Selector, On/Off Control, Momentary Control, or One Shot Control)  
[... 111 lines cut ...]
  uint8_t  : 4;                                      // Pad
                                                     // Collection: CA:GamePad CP:SystemControl
  uint8_t  GD_GamePadSystemControlSystemMainMenu : 1; // Usage 0x00010085: System Main Menu, Value = 0 to 1
  uint8_t  : 7;                                      // Pad
                                                     // Collection: CA:GamePad
  uint8_t  GEN_GamePadBatteryStrength;               // Usage 0x00060020: Battery Strength, Value = 0 to 255
} inputReport_t;

//--------------------------------------------------------------------------------
// Physical Interface Device Page outputReport (Device <-- Host)
//--------------------------------------------------------------------------------

typedef struct
{
                                                     // No REPORT ID byte
                                                     // Collection: CA:GamePad CL:
  uint8_t  PID_GamePadDcEnableActuators : 4;         // Usage 0x000F0097: DC Enable Actuators, Value = 0 to 1
  uint8_t  : 4;                                      // Pad
  uint8_t  PID_GamePadMagnitude[4];                  // Usage 0x000F0070: Magnitude, Value = 0 to 100
  uint8_t  PID_GamePadDuration;                      // Usage 0x000F0050: Duration, Value = 0 to 255, Physical = Value in 10⁻² s units
  uint8_t  PID_GamePadStartDelay;                    // Usage 0x000F00A7: Start Delay, Value = 0 to 255, Physical = Value in 10⁻² s units
  uint8_t  PID_GamePadLoopCount;                     // Usage 0x000F007C: Loop Count, Value = 0 to 255
} outputReport_t;
```
https://gist.github.com/DJm00n/07e1b7bb21643725e53b16f45e0e7022

kakra, 2020-10-05:
What does "looking at xboxgip.sys" mean? Did you extract information directly from the binary? That's probably legally problematic...

DJm00n, 2020-10-05:
> What does "looking at xboxgip.sys" mean? Did you extract information directly from the binary? That's probably legally problematic...

Same HID descriptor could be extracted via USB sniffer software tool (wireshark etc).