medusalix/xow #123: Dongle heating too much when not in use

DatCrazyDelphox, 2020-10-07:
In the past, there was an update that made the dongle enter in a low-power state when the controller was disconnected on when waiting for a controller to connect

But since these new updates, the dongle isnt entering in this state and it heats up pretty bad when not in use.

Arch Linux x86_64 - linux-tkg-upds 5.8.13
Dongle model 1790

DatCrazyDelphox, 2020-10-07:
Here's my journalctl
seems like xow isnt powering down the dongle when the controller is disconnected

```
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - Opening device...
out 07 16:32:25 Archie systemd-udevd[1470954]: could not find module by name='xow_blacklist'
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - Firmware loaded
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - ASIC version: 7632
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - MAC version: 7662
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - Chip id: 7613
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 INFO  - Wireless address: 62:45:b5:04:40:fc
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - Channel 1, power: 32
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - Channel 6, power: 34
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - Channel 11, power: 35
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - Channel 36, power: 46
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - Channel 40, power: 46
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - Channel 44, power: 45
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - Channel 48, power: 45
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - Channel 149, power: 43
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - Channel 153, power: 43
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - Channel 157, power: 43
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - Channel 161, power: 43
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 DEBUG - Channel 165, power: 43
out 07 16:32:25 Archie xow[1470402]: 2020-10-07 16:32:25 INFO  - Dongle initialized
out 07 16:32:37 Archie xow[1470402]: 2020-10-07 16:32:37 INFO  - Controller '1' connected
out 07 16:32:37 Archie xow[1470402]: 2020-10-07 16:32:37 INFO  - Device announced, product id: 02ea
out 07 16:32:37 Archie xow[1470402]: 2020-10-07 16:32:37 DEBUG - Firmware version: 4.8.1923.0
out 07 16:32:37 Archie xow[1470402]: 2020-10-07 16:32:37 DEBUG - Hardware version: 1284.1.1.1
out 07 16:32:37 Archie xow[1470402]: 2020-10-07 16:32:37 INFO  - Serial number: 02600209312844
out 07 16:32:37 Archie xow[1470402]: 2020-10-07 16:32:37 INFO  - Battery level: 2
out 07 16:32:50 Archie xow[1470402]: 2020-10-07 16:32:50 INFO  - Controller '1' disconnected
```

ddragic, 2020-10-10:
I don't think xow ever supported powering down the dongle when there are no controllers connected. Windows driver does this, the timeout for which can be controlled with selective suspend idle timeout in xbox wireless adapter properties (in device manager).

xow only turns off the dongle when it exits with systemctl stop ... or ctrl+c if running from command line.

If this actually was implemented at some point then I must have missed it...

NoXPhasma, 2020-10-31:
As far as I know xow never went to low-power state. Only change I can remember which have been done so far was to disable the LED when no controller is connected.