medusalix/xow #125: Does xow require libusb-1.0.23?  (re: errors with libusb-1.0.20)

sofakng, 2020-10-14:
Does xow require libusb-1.0.23?

I'm troubleshooting a problem running this on a MiSTer (DE10-Nano + ARM) board.  If I compile libusb-1.0.23 it works fine, but MiSTer includes libusb-1.0.20 and it doesn't work.  Just to verify, I re-compiled libusb-1.0.20 with debug symbols and it produces the following segmentation fault:

```
(gdb) run
Starting program: /root/xow
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/libthread_db.so.1".
2020-10-14 17:49:59 INFO  - xow v0.5-16-ga3d2b01 ©Severin v. W.
[New Thread 0xb6cfe450 (LWP 1689)]
2020-10-14 17:49:59 INFO  - Waiting for device...
2020-10-14 17:50:05 DEBUG - Opening device...

Thread 1 "xow" received signal SIGSEGV, Segmentation fault.
0xb6f9b0f4 in do_sync_bulk_transfer (dev_handle=0x37600, endpoint=<optimized out>, buffer=0x531b8 "", length=14344,
    transferred=transferred@entry=0x0, timeout=timeout@entry=1000, type=type@entry=2 '\002') at sync.c:182
182     sync.c: No such file or directory.

```

Here is the backtrace:
```
(gdb) bt
#0  0xb6f9b0f4 in do_sync_bulk_transfer (dev_handle=0x37600, endpoint=<optimized out>, buffer=0x531b8 "",
    length=14344, transferred=transferred@entry=0x0, timeout=timeout@entry=1000, type=type@entry=2 '\002')
    at sync.c:182
#1  0xb6f9b2e8 in libusb_bulk_transfer (dev_handle=<optimized out>, endpoint=<optimized out>, data=<optimized out>,
    length=<optimized out>, transferred=transferred@entry=0x0, timeout=timeout@entry=1000) at sync.c:257
#2  0x00019a82 in UsbDevice::bulkWrite (this=0x37670, endpoint=endpoint@entry=4 '\004', data=...) at dongle/usb.cpp:141
#3  0x0001369a in Mt76::loadFirmwarePart (this=this@entry=0xbefff9d0, offset=offset@entry=524288,
    start=start@entry=..., end=end@entry=...) at dongle/mt76.cpp:1529
#4  0x00013aaa in Mt76::loadFirmware (this=this@entry=0xbefff9d0) at dongle/mt76.cpp:1464
#5  0x000167f6 in Mt76::Mt76 (this=0xbefff9d0, usbDevice=...) at dongle/mt76.cpp:819
#6  0x00018b6a in Dongle::Dongle (this=0xbefff9d0, usbDevice=...) at dongle/dongle.cpp:24
#7  0x00012804 in main () at xow.cpp:88
```

If anybody can verify that libusb-1.0.23 is definitely required it will help me communicate with the MiSTer team and ask them to upgrade the included libusb package.

kakra, 2020-10-14:
This looks mostly like a bug in libusb to me. You could look at the git log of libusb or run a git bisect of the two versions to find a commit that breaks xow, then see if that commit is actually about such an issue.

sofakng, 2020-10-15:
You are correct and I've found the bug in the older versions of libusb.

However, the fix is very simple and I've submitted a pull request to enable compatibility with these older versions.

#126

kakra, 2020-10-16:
This should be closed, then. @medusalix @sofakng

sofakng, 2020-10-16:
@kakra I'll close the issue as this has now been fixed.  Thanks again!

closed 2020-10-16 by sofakng