medusalix/xow #127: XOW not working on the latest stable manjaro release.

Walnut-dev, 2020-10-19:


transcengopher, 2020-11-07:
@Walnut-dev, your log seems very similar to what me and some other people have been getting in #118 (in fact, down to product ids in announcement events).

I suspect this might be due to controller itself, not the kernel modules.

Walnut-dev, 2020-11-08:
@transcengopher, the problem is due to a missing kernel module called uinput. The kernel module seems loaded on most zen-kernels, which can be found here https://github.com/zen-kernel/zen-kernel

closed 2020-11-08 by Walnut-dev

kakra, 2020-11-08:
This doesn't make sense: If it works without headphones plugged, it would still use uinput and would throw the same error. However, there may be a difference between kernel versions where older version accept that "invalid argument".

transcengopher, 2020-11-09:
@Walnut-dev, in lines between "Device announced 02ea" and "Device announced 02f6" the driver writes a great deal of stuff into the `uinput` channel , which works fine. This suggests you do have the module, and problem was in something else.