medusalix/xow #128: Possible to disable built-in mt76x2u driver without modprobe.conf?  (re: embedded linux)

sofakng, 2020-10-19:
Is it possible to disable the mt76x2u driver when it's compiled directly into the kernel and not a module?

I'm trying to get this working on MiSTer which uses an embedded linux distribution but it has the mt76x2u driver built directly into the kernel and not as a kernel module.

It looks possible to disable certain kernel drivers passing a kernel parameter (initcall_blacklist=xx) but I can't seem to get this to work for the mt76x2u driver.

I'm also using the older "bulkier" Xbox Adapter which I believe does require blacklisting but I'm hoping an alternative might exist?

medusalix, 2020-10-23:
> It looks possible to disable certain kernel drivers passing a kernel parameter (initcall_blacklist=xx) but I can't seem to get this to work for the mt76x2u driver.

I believe this has to be implemented for each driver individually. AFAIK the `mt76` doesn't support this option.

> I'm also using the older "bulkier" Xbox Adapter which I believe does require blacklisting but I'm hoping an alternative might exist?

Unfortunately, even the newer Xbox wireless adapter requires blacklisting on recent kernels (5.x something).

I'm currently working on a kernel module that will eventually replace `xow` entirely (hence my sudden inactivity regarding xow's issues). A proper announcement will follow.

sofakng, 2020-10-23:
@medusalix Thanks very much for the information and your hard work on this driver.  That's very exciting news that you are working on a kernel module which sounds like the easiest and best solution for end-users!  

I'm looking forward to see what you have to announce!

closed 2020-10-23 by sofakng