medusalix/xow #129: R3 button do not work

Esunmanta, 2020-11-09:
All works very good but R3 button dont. I dont know why. Since i have virtual here i use the devices when i run moonlight, parsec or steamlink and stop using the device when i quit session being able to use xbox Wireless with retropie again. Use device active drivers for Windows. 

I know many People do not have this software so you cannot use R3. I think there is a confict with R3 and guide button. Maybe mapping we can solve this?

Thank you so much.

kakra, 2020-11-09:
From xpadneo development, I found that some games start to mix R3 (if you mean the right thumb stick button by that), Guide and Select if you're presenting 11 buttons instead of 10. The fix is easy: The guide button should not be part of the button map of the gamepad, that only supports 10 buttons originally. I'm not sure if xow could solve this currently unless it reports the guide button remapped to an `KEY_*` event instead of `BTN_MENU`. But as far as I know, such a change makes the guide button useless to RetroPi.

Esunmanta, 2020-11-09:
Excuse my ignorance, I'm quite a newbie.

Yes we are talking about the same button: Right thumb stick. I use retropie and tried leaving the guide button unassigned but it didn't work.

Could you tell me how can i remmap buttons in xow? I don't mind losing the guide button.

Thank you so much for helping.

kakra, 2020-11-10:
Yeah, as I suspected. This line in xow should probably be changed to use `KEY_HOME` instead:
https://github.com/medusalix/xow/blob/af5b9d477706874797ba1bece0282d00225b742b/controller/controller.cpp#L196

And of course, this line, too:
https://github.com/medusalix/xow/blob/af5b9d477706874797ba1bece0282d00225b742b/controller/controller.cpp#L101

Then, the device exposes just 10 joystick buttons which is compatible with expectations of games. The Xbox controller has 10 buttons, not 11. The Guide button is handled completely separate in Windows, and even the HID descriptor shows it as a secondary HID application (consumer control).

medusalix, 2021-01-08:
@kakra I'll probably stick to `BTN_MODE` at least until you implement subdevices for xpadneo. I don't want to break any existing stuff.

closed 2022-03-20 by medusalix