medusalix/xow #132: Error setting up axis: Invalid argument

woelrutie, 2020-11-16:
Hello,

I have an official Microsoft XBOX Wireless Dongle and a new XBOX Series X controller (with the share button), and I am unable to get the controller to remain connected. Below is the output of sudo xow/xow:

```
2020-11-15 19:41:24 INFO  - xow v0.5-17-gaf5b9d4 ©Severin v. W.
2020-11-15 19:41:24 INFO  - Waiting for device...
2020-11-15 19:41:29 INFO  - Wireless address: 62:45:bd:09:37:56
2020-11-15 19:41:29 INFO  - Dongle initialized
2020-11-15 19:41:30 INFO  - Controller '1' connected
2020-11-15 19:41:30 INFO  - Device announced, product id: 0b12
terminate called after throwing an instance of 'InputException'
  what():  Error setting up axis: Invalid argument
Aborted
```
Is anyone able to help me out with this issue?

medusalix, 2020-11-21:
Sounds like a kernel issue to be honest. The driver is failing at the `UI_ABS_SETUP` ioctl (which is handled by the `uinput` kernel module). Are you using a kernel older than 4.5?

woelrutie, 2020-12-09:
I was using a Ubuntu LXC container which seemed to complicate things. I have since switched to a Ubuntu virtual machine and had no trouble getting the controller to connect.

closed 2020-12-09 by woelrutie