medusalix/xow #133: Works With Series X Controller

cow-killer, 2020-11-16:
Hey guys, not really an issue, but just wanted to let y'all know `xow` works with the Series X controller:
[image]
Took a little bit of fiddling to get it to work nicely; had to switch USB port from 2.0 to 3.0 and press the receiving button on the dongle a couple of times to finally get it to connect, but it's working just fine now. Rumble works too.

Only thing is the `Share` button is not detected. On `jstest` 10 buttons show up and `Share` would be the 11th.

kakra, 2020-11-16:
You shouldn't hide the product id... It identifies the model and is important for development.

Besides that: xow shouldn't show the guide nor the share button in jstest. Both buttons should be exposed as consumer controller buttons or keyboard events, otherwise games may get confused. Also look if right thumb, select, and back are correctly shown in jstest because MS may have inserted the share button in an unexpected location of the protocol.

cow-killer, 2020-11-16:
Product ID is `0b12`.
`Back` is working as button 6.
`Start` is button 7.
The `Guide` button is actually working as button 8.
And right-stick click is button 10.

transcengopher, 2020-11-17:
@cow-killer does it have an audio jack? Do you have headphones/headset you can plug in there?

Can you do that for me please? The driver would amlost certainly crash when you do (it's a known thing), I'm just interested in what `product id` gets reported when controller announces that it now has audio capabilities.

cow-killer, 2020-11-17:
Inserting headphones didn't make any difference. The `id` stayed the same and the program kept running. Though, I didn't get any new headphone options in my sound settings.

transcengopher, 2020-11-18:
@cow-killer, interesting. When I plug headphones into my gamepad, there is a second message in log reading `Device announced, product id: XXX`, and then the driver crashes. Possibly your model works differently then.
Are you plugging just the headphones (i.e. is there a microphone)?

cow-killer, 2020-11-19:
Yup, just the headphone port. I mean, I'd try with another pair of headphones, but I have only one right now.

medusalix, 2020-11-21:
I'd like to get the share button working before I merge #130.
It would be extremely helpful if you could provide me with a USB packet capture of the controller (either wired or wireless) where you press the share button.

cow-killer, 2020-11-22:
> It would be extremely helpful if you could provide me with a USB packet capture of the controller (either wired or wireless) where you press the share button.

Assuming I'm getting the packet from Wireshark, what protocol/interface should I be specifically looking for?

Claybie, 2020-11-23:
For what it's worth it may be useful to be aware of [this issue](https://answers.microsoft.com/en-us/xbox/forum/all/xbox-series-x-controller-keeps-disconnecting/16d9f462-57c0-4b6c-93fd-b03711e632ee) even though it is not directly related to linux or xow. I do not use the controller with a console but can attest that the disconnects happen for me in both Win 10 and Manjaro when left idle.

This isn't really an issue in games like Dark Souls II where if the pad disconnects, the game will recognize it again almost immediately. Playing though, for example, Dragon Quest XI S demo in Proton-GE, when the pad disconnected the game would not register it again unless restarted. FYI

medusalix, 2020-11-23:
> Assuming I'm getting the packet from Wireshark, what protocol/interface should I be specifically looking for?

You mean the interface to select in Wireshark to capture packets from? You'd have to find the one that corresponds to the USB device (lots of incoming packets as soon as you plug in the dongle/controller).
The protocol itself is proprietary (just USB packets), so I need the whole packet capture to find the share button packet manually.

kakra, 2020-11-23:
> Playing though, for example, Dragon Quest XI S demo in Proton-GE, when the pad disconnected the game would not register it again unless restarted. FYI

This may be a bug in some Proton versions that has since been fixed. I think I've seen a commit for that lately, and I think Proton-GE lists a fix for it in the latest version. At least it works for me again.

OTOH, there may be a situation where the controller disconnects without the kernel taking note of it, unless the controller appears again on the input bus. It looks like the kernel now announces the controller as a new device while the old connection is still existing, and only then the old connection disconnects. The game may thus see the controller as a new input device, and not the originally disconnected controller.

cow-killer, 2020-11-23:
You can try this, though, I got no idea if it captured the Share button or not (tested with USB cable).
https://drive.google.com/file/d/1_n4Lr7EWpBtyXgcMMgFXjh9RQuCEEdLR/view?usp=sharing

ataffanel, 2021-02-21:
This might be a bit late but I captured the share button on linux. I filtered the pcap to only keep the report that shows pressing/releasing share and pressing/releasing A (so this results in 4 packets). The capture was done over USB when running jstest-gtk to get the gamepad to output data.

[xbs_controller_share_press-release_A_press-release_4p.zip](https://github.com/medusalix/xow/files/6017960/xbs_controller_share_press-release_A_press-release_4p.zip)

I looks like an input report (0x20), there seems to be 32bits of 0 after stickRightY and then there is the share button bit. There is also a lot of data after. This version of the gamepad supports remapping like the Elite controllers but the extra data does not look like the "unmapped" inputs (I saw in gip.cpp that it could be a reason for the input report to be bigger)

kakra, 2021-02-21:
The extra data may be 127 keyboard bits as the controller supports mapping keyboard events. In HID mode (Bluetooth), the extra data in the reports actually map to keyboard events.