medusalix/xow #135: Problems connecting xbox controller to steam client

JenkooTheLegitC, 2020-11-27:
xow.service is running fine for me, the controller seem to be connected to the system. But, steam does not recognize the controller. Anyone that knows how to fix this?

[image]

[image]

xsgb, 2020-12-04:
Hi, I can't try by myself but someone by chat reported me that it worked after unchecking "Xbox Configuration Support".

JenkooTheLegitC, 2020-12-12:
Unchecked the "Xbox configuration support". Killed the steam daemon and repaired the controller. Still no connection. 

[image]

closed 2022-03-20 by medusalix