medusalix/xow #139: Working with Model 1914 on Raspberry Pi OS

victor-sm, 2020-12-06:
I was able to successfully pair the 1914,1790 model with the current xow version and all buttons and joysticks were working, except for the share button. The Steam Link App also recognized the controller correctly.

Running Raspberry Pi OS I was missing some packages and needed to execute the following commands to make it work:

`sudo apt-get install cabextract`
`apt install libusb-1.0-0-dev build-essential`

Cageman1984, 2020-12-06:
I wanted to mention I also have the xbox one wireless gamepad (model 1914) with the usb receiver on a raspberry pi with raspberry pi OS and moonlight. Also installed xow. It seems to work well.

Like victor, my share button doesn't work and I was missing the same packages.

The steam link app recognized the gamepad as xinput, instead of xbox controller. However I don't use the steam link app. I use moonlight instead. 

Doesn't matter, it works great.

Any things I should know about (automatic) updating?

victor-sm, 2020-12-07:
@Cageman1984 Before I connected the controller to the Raspberry, I had it connected to the Windows 10 machine where Steam was running as a host. There the controller was recognized correctly and after switching to the Raspberry I think Steam "remembered" and recognized it again correctly as a xbox controller.

githubuserdefault, 2020-12-15:
Got it working also. Using steam link; recognized as xinput.
Has either of you gotten the sound working via the 3.5mm headphone plug?

Cageman1984, 2020-12-17:
> Has either of you gotten the sound working via the 3.5mm headphone plug?

Haven't tried that yet.

I noticed there are some button mismaps. The right stick press is recognized as the guide button. The left stick press is recognized as the right stick press. The left stick press itself seems absent...

kakra, 2020-12-19:
> I noticed there are some button mismaps

This is due to the share button which MS inserted at and odd location. xow probably needs to remap it to a different bit at the end of the button bitmap. You should try updating the firmware of the controller first, tho. In the past I've seen MS fixing mappings through firmware updates.

Cageman1984, 2021-01-12:
I solved the button mismap problem by disabling the xbox game bar in windows settings.

toccata10, 2021-04-13:
I've got exactly the same setup (1914 gamepads, 1790 dongle) on a pi4 with raspbian which is used as a steamlink connecting to a win10 pc.
It works mostly fine, but I have some disconnections (specially when idle, but not only). I updated the firmware a few days ago with a win10. No difference.
The gamepads sometimes can stay connected for quite a long time even if I don't touch them, but sometimes, the light starts blinking. I'm wondering if you can share some details on your installation (used usb port, distance, if you put your dongle in a special position). I can't say if it's game dependent as I'm mostly playing Lego-lotr for now. 
Strange behaviour (and a bit frustrating).
And many thanks for the xow project by the way.

Cageman1984, 2021-04-20:
Hi toccata. We've got about the same setup indeed. Only i have a Pi3B.

My connection stays well, most of the time. On rare occasions the problem you describe occur, but I'm never really sure if it's a powersave function or something like that.

I think there are not a lot of specific details I can give, but I'll try.

My dongle is directly plugged in the upper left port at the moment, but they have been plugged in different ports as well. The distance to the controller is about two meters, no obstructions. Recently I've switched from regular AA batteries to a Microsoft recharcheable pack. This does not seem to have any (negative) effect.

When i just got the controller and tried it for the first time, I used the dongle with an USB extension cable. This resulted in poor performance. 

That's about all the info I can think of.

toccata10, 2021-04-20:
Hi Cageman,
Thanks for your feedback. I'm a bit further away and have to use the usb extension cable (otherwise it's behind my amplifier and it's worse). For now, as I use this pi to stream steam from my win10 bedroom computer, I've started using "virtual here": there's a free version which works well (limitation is just 1 device, but I don't need more). It transmits the usb signal over ethernet: everything is as if the dongle was directly connected to the win10 pc.
So far, it's very reliable: no disconnection. Maybe there's a "secret sauce" in win10 to avoid disconnection.
The only drawback is that I have to launch the steam link with a mouse (as the gamepad is "connected" to the remote computer).