medusalix/xow #143: 1914 controller - triggers are not managed like buttons

Franck-Berry, 2020-12-19:
I tested xow with the 1914 controller. It is recognized, but I am not sure that all buttons are handled correctly.

Triggers are not seen as buttons:
Left trigger: AXIS2+
Right trigger: AXIS5+

By the way, I don't know if it matters, but the button numbers are different when I use the controller with a USB cable (without xow).

FYI, I am using Raspberry Pi OS Lite.

kakra, 2020-12-19:
Triggers are actually axes. The Linux input subsystem doesn't know of buttons that can be pressed just half way, so they are axes. It's up to SDL or other higher layers to interpret those as buttons and make up a proper interpretation of the value.

The controller has a different mapping in USB mode: It follows native Windows mapping in that case.

If you're seeing this axis order, it's actually correct for Linux HID: LX,LY,LT,RX,RY,RT. Joydev games may see a wrong mapping as they expect LX,LY,RX,RY,LT,RT - but usually there's no modern game still using that API. SDL2 ships a database of controller mappings to compensate for differences between models and present a consistent view to games.

But since games only see the USB VID/PID of the dongle, xow should expose a consistent mapping across all models. LX,LY,LT,RX,RY,RT looks right to me. The new Xbox One X|S controllers (not sure if that's model 1914) may insert the share button at an unexpected location which will mess up what SDL2 guesses as the button mapping, you'll probably see wrong mappings for LS,RS,Start,Select,Share in that case. If that mapping is wrong, xow needs to fix it but the axes look correct to me, and if that messes with your games, you need to get it fixed in SDL.

USB and dongle mode will have different VID/PID values, so SDL would see different gamepad models and apply different mappings.

Franck-Berry, 2020-12-19:
Thank you for this very detailed answer! Unfortunately, I am just starting to discover the subject of controllers with Linux. So it's going to take me some time and research to fully analyze / understand the answer. Among other things, the notion of axis order does not mean anything to me at this moment. I'll look ! ;-)

Regarding the controller model, I confirm that this is the very last version released by Microsoft. (Actually, I didn't check if this was really the 1914 version).

It's RetroPie that communicates to me the mapping between the buttons during the configuration of the controller. It gives me the following correspondence (if that can help):
>d-pad up (HAT 0 UP)
>d-pad down (HAT 0 DOWN)
>d-pad left (HAT 0 LEFT)
>d-pad right (HAT 0 RIGHT)
>start (BUTTON 7)
>select (BUTTON 6)
>Button A / east (BUTTON 1)
>Button B / south (BUTTON 0)
>Button X / north (BUTTON 3)
>Button Y / west (BUTTON 2)
>left shoulder (BUTTON 4)
>right shoulder (BUTTON 5)
>left trigger (AXIS 2+)
>right trigger (AXIS 5+)
>left thumb (BUTTON 9)
>right thumb (BUTTON 10)
>Left analog up (AXIS 1-)
>Left analog down (AXIS 1+)
>Left analog left (AXIS 0-)
>Left analog right (AXIS 0+)
>Right analog up (AXIS 4-)
>Right analog down (AXIS 4+)
>Right analog left (AXIS 3-)
>Right analog right (AXIS 3+)

The Xbox button is seen as BUTTON8.

Once the controller is configured in RetroPie, it seems to behave fine in games. (I'm not a gamer. I'm just doing an install for a friend. I'm not testing games enough to be sure everything is working properly. Besides, I don't have any libRetro games that use triggers.)

I just have a problem with the OpenMSX emulator (which in its latest version uses SDL2) because I need to define a mapping between controller buttons and keyboard keys (F1, F2, F3 ...). OpenMSX does not use LibRetro. So I have to configure the assignment of each button of the controller. With OpenMSX, I can establish a mapping between a "button" and a keyboard key. I don't know how to do it for an "axis". This is what motivated me to declare an issue here.

kakra, 2020-12-19:
Most software should accept axes as button presses if you assign it to a binary input. It should register an input at 50% range. If it doesn't work, you can still try anti-micro which can translate controller input, either by remapping, or by converting it to mouse and keyboard.

Franck-Berry, 2021-01-31:
Kakra was right.

I finally took some time to work on this topic. Starting from a blank configuration, I get exactly the same setting whether the controller is connected by a cable or by the Microsoft wireless connector.

For example, in both cases, the right trigger corresponds to AXIS 5+.

I probably had an old "driver" that was wrongly left on my first attempt.

I close this "false" issue.

closed 2021-01-31 by Franck-Berry