medusalix/xow #144: Model 1904 wireless dongle timeouts/disconnects when idle

jeffkeller87, 2020-12-20:
**Model Number (as printed on box):** "1904, 1790"
**`xow` version:** 0.5

The `xow` driver works flawlessly with this controller model as long as the controller is not idle for too long (~30s). This usually isn't a problem when gaming, since the user is often sending inputs from the controller more than once per 30 seconds.

However, if left to idle, the wireless connection to the dongle will disconnect, and reconnect a few seconds later. Not a huge deal, but is somewhat annoying when resuming play after a cut scene or load screen. Sometimes I just flick one of the analog sticks to keep the connection alive as a workaround.

```
$ lspci | grep USB
03:00.0 USB controller: Advanced Micro Devices, Inc. [AMD] X370 Series Chipset USB 3.1 xHCI Controller (rev 02)
26:00.0 USB controller: ASMedia Technology Inc. ASM2142 USB 3.1 Host Controller
29:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Matisse USB 3.0 Host Controller
```
I have tried plugging the controller into:

* A USB port on the back motherboard (I assume this is the first controller listed above)
* A USB hub, which in turn is plugged into the back of the motherboard
* A front-panel USB port (I assume this is the third controller listed above)

The only other USB header in use on my motherboard is from my NZXT X62 Kraken AIO. I assume this is the ASMedia controller listed above.

Some other logs that are useful as a sanity check:

```
$ sudo systemctl status xow
● xow.service - Xbox One Wireless Dongle Driver
     Loaded: loaded (/etc/systemd/system/xow.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2020-12-20 14:18:49 EST; 4min 11s ago
   Main PID: 5905 (xow)
      Tasks: 4 (limit: 38366)
     Memory: 2.0M
     CGroup: /system.slice/xow.service
             └─5905 /usr/local/bin/xow

Dec 20 14:22:06 COOKIE-MONSTER xow[5905]: 2020-12-20 14:22:06 INFO  - Controller '1' connected
Dec 20 14:22:07 COOKIE-MONSTER xow[5905]: 2020-12-20 14:22:07 INFO  - Device announced, product id: 0b12
Dec 20 14:22:07 COOKIE-MONSTER xow[5905]: 2020-12-20 14:22:07 INFO  - Serial number: 09711987454032
Dec 20 14:22:08 COOKIE-MONSTER xow[5905]: 2020-12-20 14:22:08 INFO  - Battery level: 2
Dec 20 14:22:33 COOKIE-MONSTER xow[5905]: 2020-12-20 14:22:33 INFO  - Controller '1' disconnected
Dec 20 14:22:42 COOKIE-MONSTER xow[5905]: 2020-12-20 14:22:42 INFO  - Controller '1' connected
Dec 20 14:22:42 COOKIE-MONSTER xow[5905]: 2020-12-20 14:22:42 INFO  - Device announced, product id: 0b12
Dec 20 14:22:42 COOKIE-MONSTER xow[5905]: 2020-12-20 14:22:42 INFO  - Serial number: 09711987454032
Dec 20 14:22:43 COOKIE-MONSTER xow[5905]: 2020-12-20 14:22:43 INFO  - Battery level: 2
Dec 20 14:22:58 COOKIE-MONSTER xow[5905]: 2020-12-20 14:22:58 INFO  - Controller '1' disconnected

```

```
$ dmesg -w
# ...
[ 1756.079810] input: Xbox One Wireless Controller as /devices/virtual/input/input67
[ 1779.732696] input: Xbox One Wireless Controller as /devices/virtual/input/input68
[ 1826.222612] input: Xbox One Wireless Controller as /devices/virtual/input/input69
[ 1864.723501] input: Xbox One Wireless Controller as /devices/virtual/input/input70
[ 1886.227131] input: Xbox One Wireless Controller as /devices/virtual/input/input71
[ 1916.229892] input: Xbox One Wireless Controller as /devices/virtual/input/input72
[ 1947.973437] input: Xbox One Wireless Controller as /devices/virtual/input/input73
[ 1983.711712] input: Xbox One Wireless Controller as /devices/virtual/input/input74
[ 2008.183596] input: Xbox One Wireless Controller as /devices/virtual/input/input75
[ 2032.042565] input: Xbox One Wireless Controller as /devices/virtual/input/input76
[ 2078.121966] input: Xbox One Wireless Controller as /devices/virtual/input/input77
[ 2129.935380] input: Xbox One Wireless Controller as /devices/virtual/input/input78
[ 2170.485511] input: Xbox One Wireless Controller as /devices/virtual/input/input79
[ 2194.651219] input: Xbox One Wireless Controller as /devices/virtual/input/input80
[ 2216.052670] input: Xbox One Wireless Controller as /devices/virtual/input/input81
[ 2236.020467] input: Xbox One Wireless Controller as /devices/virtual/input/input82
[ 2287.936289] input: Xbox One Wireless Controller as /devices/virtual/input/input83
[ 2307.699296] input: Xbox One Wireless Controller as /devices/virtual/input/input84
[ 2338.930774] input: Xbox One Wireless Controller as /devices/virtual/input/input85
[ 2359.615096] input: Xbox One Wireless Controller as /devices/virtual/input/input86
[ 2383.269234] input: Xbox One Wireless Controller as /devices/virtual/input/input87
```

Please let me know if there is any other information I can provide. I know there is a section in the README that mentions USB incompatibilities. I think I have successfully avoided an ASMedia USB controller, but maybe AMD controllers are also incompatible?

I will be doing another PC build in a few weeks which will take me away from this X370 motherboard and onto a B550. Will report back then as well.

jeffkeller87, 2020-12-20:
Could it be that even though I have tried using the USB 2.0 ports in a couple locations, that because the underlying USB controller on the motherboard is USB 3.x, that I am actually running into the power management issues described in the README?

medusalix, 2020-12-20:
@jeffkeller87 It doesn't look like the problem is caused by your USB controller at all. It seems like the newer Series X controller has that exact same issue on Windows too. 
There's a PR (#142) that apparently fixes those issues. You might want to give it a try.

jeffkeller87, 2020-12-20:
Thanks for the tip @medusalix. Unfortunately, that PR did not fix the issue for me. I commented my results there.

yoyossef, 2020-12-22:
I confirm this issue with my Xbox Series Controller (white one), one thing different though is that it's sometimes not reconnecting the controller after it gets disconnected.

Sometimes I'm able to reconnect it by pushing the Xbox button and it's not that big of a deal, but after a few times my controller won't turn on again, the led immediately turns off (instead of repeatedly blinking) as I stop pushing the Xbox button.

And the only solution to that is for me to put out the batteries, and I have to wait for the controller to reset itself for ~2-3 minutes, similar to waiting for a BIOS battery juice to run out it feels like.

It feels very unstable and unreliable overall unfortunately, I'd be glad to help in any way.

jeffkeller87, 2020-12-27:
So this issue has seemingly gone away for me.... the only material change I can think of has been changing the ports in which my USB hub is connected to the motherboard _and also rebooting_. Previously, I was just restarting the `xow` service between testing different ports. So something for others to try.

alkazar, 2020-12-31:
@jeffkeller87 are you using this fix https://github.com/medusalix/xow/pull/142 or a release build?

I am also experiencing this issue. The timeout doesn't seem to be consistent. I have seen as short as 15 seconds and as long as 45 seconds.

I am using xow 0.5.

jeffkeller87, 2020-12-31:
@alkazar I am back to using the 0.5 release build and the issue has still not returned for me.

Aside from what has already been mentioned, the only other thing I can think of that could have contributed to the fix was that I booted into my Windows partition. I did not use the controller while in Windows, but I suppose it is possible that a firmware update was applied by Windows Update automatically?

When I was having the issue, I did notice the timeout could be inconsistent as well, 15-45 seconds sounds about right.

alkazar, 2020-12-31:
Switching USB ports did not seem to help. I am using the dongle I bought for use with the Xbox One S controller. I wonder if the dongle needs a firmware upgrade to work with the new controllers properly.

TheCraiggers, 2021-01-10:
Switching entire computers hasn't helped, never mind USB ports. Meanwhile, my older xbox controller (no share button, microUSB) works just fine. As medusa said, people are [having problems in windows as well](https://bgr.com/2021/01/06/xbox-series-x-controller-bug-disconnect-fix/) so I don't think this is even a xow issue. 

However, I'm worried that any fix released from MS will be difficult / impossible to apply without a windows computer or xbox game console, as it looks like it may involve a firmware update for the controllers. I don't know of a way to update them on Linux.

kakra, 2021-01-10:
@TheCraiggers You can download an eval ISO of Windows 10 directly from Microsoft, install it into a VM, and run the Xbox accessories app from there to update the gamepad. You should do that with the dongle, tho, because the controller will reboot into a firmware download mode which would disconnect it from USB (and thus disable USB pass-thru to the VM) - using the dongle, the USB connection will be kept alive as only the controller reboots.

jeffkeller87, 2021-01-12:
> I will be doing another PC build in a few weeks which will take me away from this X370 motherboard and onto a B550. Will report back then as well.

Reporting back with more info now that I'm on an MSI B550 Tomahawk motherboard:

```
$ lspci | grep USB
02:00.0 USB controller: Advanced Micro Devices, Inc. [AMD] Device 43ee
2d:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Matisse USB 3.0 Host Controller
```

I had to re-pair the controller with the dongle and then proceeding to try every USB port on the back of the motherboard (rebooting in between) and found that the lone USB 2.0 port (not counting the BIOS flashback port) gave me the best results.

[image]

Interestingly, I was still experiencing disconnects when using the dongle with this port, but they relatively infrequent. I turned the controller on and just let it sit without any input to see what happened. Disconnects happened at ever increasing intervals until the interval was long enough that the controller powered down due to inactivity (30 minutes I think).

Repeating the test, I still had disconnects but they continued to happen fairly infrequently.

**TLDR; My earlier experience where I claimed the disconnects had stopped may have just been a case where the disconnects were happening quite far apart. Far enough apart that input activity from normal use (playing a game) would keep the connection alive.** Not great, but a lot less annoying than a disconnect every 20-30 seconds on inactivity.

```
$ dmesg -w
[   48.492762] input: Xbox One Wireless Controller as /devices/virtual/input/input34
[   77.173289] input: Xbox One Wireless Controller as /devices/virtual/input/input35
[   98.071499] input: Xbox One Wireless Controller as /devices/virtual/input/input36
[  290.283890] input: Xbox One Wireless Controller as /devices/virtual/input/input37
[  376.708101] input: Xbox One Wireless Controller as /devices/virtual/input/input38
[ 1250.907928] input: Xbox One Wireless Controller as /devices/virtual/input/input39
# controller shuts down due to inactivity around 1850s
# I turn the controller back on
[ 2325.424310] input: Xbox One Wireless Controller as /devices/virtual/input/input40
[ 2525.107803] input: Xbox One Wireless Controller as /devices/virtual/input/input41
[ 2743.736879] input: Xbox One Wireless Controller as /devices/virtual/input/input42
[ 2779.577636] input: Xbox One Wireless Controller as /devices/virtual/input/input43
[ 2894.985140] input: Xbox One Wireless Controller as /devices/virtual/input/input44
```

kakra, 2021-01-12:
Wasn't there something in the past that mentioned Wifi channel hopping of the dongle and how the controller may disconnect then? Do you have a lot of 5 GHz Wifi around? Maybe try turning 5 GHz off in your Wifi network for a test...

jeffkeller87, 2021-01-13:
In my case there are no 5 GHz WiFi networks in the vicinity, only two 2.4 GHz networks, and I am still experiencing intermittent disconnects.

medusalix, 2021-01-15:
> Or is it 2.4 GHz that the dongle is using? @medusalix Do you know?

@kakra I'm pretty sure the firmware defaults to 5 GHz, although xow is configuring both 2.4 and 5 GHz channels. There's a `Enable2Dot4GHz` string in the binary of Microsoft's GIP driver, so I'm pretty sure there's a way to force the dongle to use 2.4 GHz (I'd really like to know how to enable it).

kakra, 2021-01-15:
I wonder if it could be disassembled, tho that's probably not allowed by the TOS supplied with the driver.

movisman, 2021-01-23:
Just to confirm, I see this exact issue on my MiSTer (linux). I had chance to test an Xbox Series X controller (1914) with it, and it works great except for this idle disconnect/reconnect thing which is quite bizarre. My firmware is fully up to date on the controller side of things. Mine disconnects and reconnects by itself in 3-4 seconds, I don't need to power it back on or anything like that. Timing can vary but seems to be around 25-30 seconds initially when left idle.

My Xbox One controllers (1708) do not exhibit this behaviour - they have worked robustly on RetroPie for a long while now. They also work great on my MiSTer.

For reference, I am using the slim wireless dongle.

There are a lot of threads all over the place about a similar issue with the controller disconnecting for people, and MS are still working on it - but it is not resolved yet. I hope this is the same issue, although curiously, when using the dongle in Windows, it doesn't seem to do the disconnect/reconnect thing - but I haven't tested in detail.

medusalix, 2021-01-24:
We'll have to wait until Microsoft releases a firmware update.

movisman, 2021-01-24:
No worries. I haven't managed to make it happen in Windows yet, but only did a little more testing today. Certainly seems like an MS thing though, so will hang tight and test again further once they release a firmware update for the controller. It's very easy to replicate on my MiSTer, intervals between each idle desync/resync can vary but ultimately it still happens on every boot and the first desync/resync is "usually" only after 30 seconds or so of idle time.

Thanks!

Bloodaxe95, 2021-08-11:
I'm not seeing any of the disconnects when I'm connected to a Windows PC, and I've updated the controller to the latest firmware. This issue is still present, and it's annoying the crap out of me. Is there anything a non-coder can do to help finding and publishing a solution to this?

kakra, 2021-08-11:
Maybe it could be worth it capturing the USB traffic from connect until the unexpected disconnect, then do the same in Windows, and compare both. If Windows does something different, it probably sends intermittent commands to the dongle which Linux doesn't - or reacts to some other events that xow does not support/care about.

You should probably not touch any controls while capturing the USB traffic to keep the protocol noise low. But I cannot help with how to correctly capture the traffic in both OS - but I'm pretty sure @medusalix already posted it somewhere (maybe in Discord only).

Pieman711, 2021-09-01:
Just to confirm I'm getting exactly the same issue. 1914 controller with the wireless dongle. I'm playing through steamlink and the disconnect/reconnect cause a load of lag and glitches that can often stop the whole game. 

I'll try and look through the discord, but if someone can point me to a bit of a guide on how to monitor USB traffic, or anything else useful, then I'll have a stab at it and feedback my findings.

djolker, 2021-09-06:
Having a very similar experience to what's being described here.

I use a 1537 and 1914 controller with my 1790 Wireless Adapter

The 1537 works very well for the most part (doesn't disconnect early), but the 1914 will start to disconnect cycle after about 15 to 30 seconds (sometimes even less) of idle time, or occasionally just straight up in the middle of usage (very rarely for that though)

I've noticed the issue is worse when they are both attached to the wireless adapter, even causing the 1537 model to disconnect cycle occasionally. 

Happy to assist in any way

samAnnisBrown, 2021-09-10:
+1

Same issue on latest release of Rasbian.  Using 1914 with wireless dongle since Bluetooth seemed impossible (pairing issues).

Dongle works a charm outside the connection issues. Same 15-30 sec idle disconnect.  Wouldn’t bother me that much except I’m using with RetroPie, and the hotkey stops working after a disconnect, meaning i can no longer save states or exit emulators without a reboot.

Extremely frustrating and keen for a fix or workaround if someone finds one!

silvannn, 2021-09-14:
Hi, I have the same issue. 1914 is constantly disconnecting and connecting but...
I am just starting my journey with RetroPie and instead of installing it by myself I used ready ISO image which I downloaded from arcadepunk.com. I have checked two images created by RickDangerous for RaspberryPi 4 and on one of them my dongle and controller is working perfectly. No idle disconnection. On second image I have that issue. In both cases I installed xow drivers the same way. So now I have pendrive with retropie where controller is working fine and ssd with retropie where I am experiencing this issue. I am a noob and I do not know how to check differences between those versions but it looks like there is no problem with xow but with operation system itself. Some settings needs to be changed or something is missing. My 1914 is updated and working perfect on PC.

djolker, 2021-09-15:
I'm curious if maybe there's two separate issues at play here that look very similar:

1. The controller is idle in which case bluetooth low energy mode trys to turn on and causes the disconnect.
2. Something unrelated to BLE causing disconnects mid use 

The title of this thread has to do with the controller timeing out when it idles, but it seems like a few of us are having disconnect problems in the middle of use.

I've noticed even my older controller will begin rapidly disconnecting and reconnecting after about an hour or two of constant play

OriginLegend, 2021-09-17:
Not sure if this issue is related to one I am facing (seems like it is).

For me the issue was the empty-packet received by the driver results in an automatic disconnect:
https://github.com/medusalix/xow/blob/master/dongle/dongle.cpp#L272

As a result, I added a build-flag to disable this behavior: EMPTY_PACKET_DISCONNECT
Pull-Request: https://github.com/medusalix/xow/pull/185

If you want to test it make sure you clean your workspace and rebuild with this flag:
make clean
make BUILD=RELEASE EMPTY_PACKET_DISCONNECT=0

Let me know if this resolves the issues for ya'll as well

samAnnisBrown, 2021-09-18:
Thanks @OriginLegend - initial testing didn’t solve this for me.

- Stopped and disable xow service
- Cleaned workspace in existing xow dir
- Cloned your fork in new dir
- Rebuilt with EMPTY_PACKET_DISCONNECT=0 flag
- Reinstalled and started service

After this I needed a reboot before controllers would connect.  Stillhad a disconnect within 30 seconds of idle.

Very basic testing - might spend more time on it tonight.

OriginLegend, 2021-09-18:
@samAnnisBrown Do you have a model 1914 controller as well?
I am trying to experiment with other solutions which might help, but so far I don't have any new things to test (all changes I have experimented with cause instability - crashing, etc).

Will try to take a look on Monday too when I have more time and see if there is anything else I can try to make this work.

samAnnisBrown, 2021-09-19:
A little more testing - basically just uninstalls, reboots, and rebuilds in various orders to see if anything stuck.  Still getting disconnects @OriginLegend.

I’ve the 1790 dongle and 1914 controllers.

Thanks for your efforts - happy to test anything you might come up with.  The install and testing I’m good with, but the dev is a bit out of my depth.

samAnnisBrown, 2021-10-30:
Not aware of conventions myself @OriginLegend , just made the perhaps incorrect assumption it would be in `master`.  

Reporting back after a few days use, and this is definitely solved.  I've now had several long play sessions without a single drop out.  Thanks so much for the fix.

Pieman711, 2021-10-31:
I've used it for a good few hours, including having two controllers attached for co-op. No disconnects whatsoever. Seems to have nailed it, thanks everyone!

To install it using SamAnnisBrowns suggestion of

git clone -b frequent-disconnects https://github.com/OriginLegend/xow.git

But otherwise followed the instructions as above. 

Good work team!

OriginLegend, 2021-10-31:
I am going to make one more change to the code to not require the additional build-flag (since this does seem to be a universal fix for all people with this controller).

I am assuming that the product id everyone is using is `0b12` so I will automatically enable the fix for this id.
But I will also leave in the build-flag for different controllers in-case they need this as well.

Logs showing the product id from original issue poster (matches with my own):
```
Dec 20 14:22:06 COOKIE-MONSTER xow[5905]: 2020-12-20 14:22:06 INFO  - Controller '1' connected
Dec 20 14:22:07 COOKIE-MONSTER xow[5905]: 2020-12-20 14:22:07 INFO  - Device announced, product id: 0b12
Dec 20 14:22:07 COOKIE-MONSTER xow[5905]: 2020-12-20 14:22:07 INFO  - Serial number: 09711987454032
Dec 20 14:22:08 COOKIE-MONSTER xow[5905]: 2020-12-20 14:22:08 INFO  - Battery level: 2
```

XirXes, 2021-10-31:
> I am assuming that the product id everyone is using is `0b12` so I will automatically enable the fix for this id. But I will also leave in the build-flag for different controllers in-case they need this as well.

I just tested this branch with a 1708 on the new beta Bluetooth LE firmware 5.11.3118.0 and its finally working correctly again. It reports as `02ea` which I'm pretty sure is the the same ID from the old firmware version.

XirXes, 2021-11-02:
I let it run for a bit over 30 minutes, and let it power off from inactivity. Let me know if you need any other captures on this firmware.
https://drive.google.com/file/d/18HpBU6tkL1qVd8MP9ng4U07NZGAqjemt/view?usp=sharing

25minutes, 2021-11-05:
@OriginLegend 
I'm trying to compile your code with static libraries. I have searched the web and tried to modify Makefile by adding "-static", then it failed to compile with a lot of "undefined reference" errors. Below are the changes I made:

xow: $(OBJECTS)
	$(CXX) $(LDFLAGS) -o $@ $^ $(LDLIBS) -static

%.o: %.cpp
	$(CXX) $(CPPFLAGS) $(CXXFLAGS) -c -o $@ $< -static

As I have limited skills on coding, I could work this out. I'd appreciate if anyone experienced could give me some advice.

Untz64, 2021-12-23:
Mine was seemingly working fine on @OriginLegend's branch until recently. Now after about a minute of use the service crashes similarly to @XirXes, but I'm not on the same firmware. I rebuilt as `DEBUG` to see if there was any more relevant info, see below:

```
Dec 23 22:39:35 retropie xow[1510]: 2021-12-23 22:39:35 INFO  - Battery level: medium
Dec 23 22:39:55 retropie xow[1510]: 2021-12-23 22:39:55 INFO  - Controller '2' connected
Dec 23 22:39:56 retropie xow[1510]: 2021-12-23 22:39:56 INFO  - Device announced, product id: 0b12
Dec 23 22:39:56 retropie xow[1510]: 2021-12-23 22:39:56 DEBUG - Firmware version: 5.9.2709.0
Dec 23 22:39:56 retropie xow[1510]: 2021-12-23 22:39:56 DEBUG - Hardware version: 1032.1.1.1
Dec 23 22:39:56 retropie xow[1510]: terminate called after throwing an instance of 'InputException'
Dec 23 22:39:56 retropie xow[1510]:   what():  Error adding key code: Invalid argument
Dec 23 22:39:56 retropie systemd[1]: xow.service: Main process exited, code=killed, status=6/ABRT
Dec 23 22:39:56 retropie systemd[1]: xow.service: Failed with result 'signal'.
```

Not sure why it's seeing the same controller as a "second" controller.

emilsz, 2022-01-10:
Can second this. The "frequent-disconnects" fix by @OriginLegend was sort of working for me, but the driver was crashing occasionally. Decided to upgrade the controller firmware using Windows 10 Xbox Accessories. After the upgrade, leaving the controller inactive for about a minute, xow shows that a new controller is connected and immediately crashes.
Controller firmware version: 5.9.2709.0
Will see if I can downgrade the firmware in the mean time.

C0nvert, 2022-02-20:
Same Problem for me

> Mine was seemingly working fine on @OriginLegend's branch until recently. Now after about a minute of use the service crashes similarly to @XirXes, but I'm not on the same firmware. I rebuilt as [...]

reapingbearpig, 2024-02-02:
Hello all!!! im here to breath life into this old thread!!! im having some related issues to this where it connects then disconnects a few seconds later and repeats this spiral infinitely. i ran command `{sudo systemctl status xow}` and receives thin back.. im not sure where to go from here. any help understanding this error is appreciated.

```
xow.service - Xbox One Wireless Dongle Driver
     Loaded: loaded (/etc/systemd/system/xow.service; enabled; preset: enabled)
     Active: failed (Result: core-dump) since Thu 2024-02-01 22:23:28 EST; 8min ago
   Duration: 13min 25.314s
    Process: 1358 ExecStart=/usr/local/bin/xow (code=dumped, signal=ABRT)
   Main PID: 1358 (code=dumped, signal=ABRT)
        CPU: 20ms

```