medusalix/xow #149: Using Controller Buttons should keep the screen unlocked

StefanLobbenmeier, 2021-01-04:
Basically just like when I play with keyboard, I would expect pressing buttons on the controller to keep the screen from unlocking.
But when I play (wine) games with this controller, the screen locks after ~5 minutes since there has been no input.

Is it possible for xow to do this? I do not think of this as high priority as starting games through Lutris allows me to deactivate the screen saver anyway.

kakra, 2021-01-04:
There has been a similar request to xpadneo, and we found that this should actually be solved by the games itself: https://github.com/atar-axis/xpadneo/issues/158

However, with xow being a user-space daemon running in the local user context, it could prevent the screensaver intent while the controller is connected.

medusalix, 2021-01-05:
Yeah, I believe we could use D-Bus to inhibit the screensaver, but I agree with @kakra that this should be fixed by the games.

kakra, 2021-01-05:
SDL games should already do it and communicate the intent "playing a game" to the desktop manager. But it's up to the desktop environment to use that as a hint for blocking the screen saver. Users of systemd may be lucky by running the game through `systemd-inhibit --what=idle`. Proton should receive a patch to just inhibit the screen saver while it is running because you actually only run games with it - such a request should be issued in Valve's project. KDE users may want to create a gaming activity and force-assign the Steam client window and the Proton game window to it, then set the gaming activity to never invoke screen blanking (which is what I did).

closed 2022-03-20 by medusalix