medusalix/xow #151: Error in control transfer: LIBUSB_ERROR_NO_DEVICE

Karsteski, 2021-01-06:
Hello, I'm using the latest microsoft wireless adapter,  Model # HK9-00001.

Here's the error:

    2021-01-06 05:10:30 INFO  - xow v0.5-20-g8a0771b ©Severin v. W.
    2021-01-06 05:10:30 INFO  - Waiting for device...
    2021-01-06 05:10:31 ERROR - Error in control transfer: LIBUSB_ERROR_NO_DEVICE
    2021-01-06 05:10:31 ERROR - Error in bulk write: LIBUSB_ERROR_IO
    2021-01-06 05:10:31 ERROR - Failed to write command
    2021-01-06 05:10:31 ERROR - Failed to burst write register
    2021-01-06 05:10:31 ERROR - Failed to write MAC address
    terminate called after throwing an instance of 'Mt76Exception'
      what():  Failed to init registers
    Aborted (core dumped)

Any help welcome, thanks :)

RobertLSnead, 2021-11-02:
Having same issue, not sure what it is.

Urbanorecord, 2022-02-21:
i have same problem.

VanKurt, 2022-05-01:
Same problem here (Manjaro 21.2.6).

MisterrrX, 2023-01-08:
Same issue