medusalix/xow #157: Rumble not working with v0.5

Roger-ZRZ, 2021-01-26:
As title, rumble not working; 

Version: https://github.com/medusalix/xow/commit/d16f61c886b59d930e14676ef3654914ee7f0331; 
Dependencies: all newest; 
Distro: Pop_os (wrapped ubuntu) 20.10;
Controller: series X controller+USB_dongle;
Tested with: steam rumble test, also limited amount of games;

Any more configuration or error info I can provide? TIA;

kakra, 2021-01-26:
This may be fixed with an upcoming v0.6, until then you could try a git version of the package if your distribution provides that.

medusalix, 2021-01-27:
@kaska He's on d16f61c according to the initial post. So it seems like the rumble patch caused some issues...

Roger-ZRZ, 2021-01-27:
> This may be fixed with an upcoming v0.6, until then you could try a git version of the package if your distribution provides that.

I compiled and installed it directly from git and is on https://github.com/medusalix/xow/commit/d16f61c886b59d930e14676ef3654914ee7f0331 already; 
The latest commit (https://github.com/medusalix/xow/commit/56d6f24d151102bde71784b0cef689a8790a5bdc) adds several lines of documentation which shouldn't really lead to any changes; 

Again, I will be happy to provide any diagnostic information;

kakra, 2021-01-27:
Ah okay, it said v0.5 in the title ;-)

Roger-ZRZ, 2021-01-27:
Ah lol sorry for the confusion, was referring to the release number;

kakra, 2021-01-27:
@Roger-ZRZ `git describe --tags --abbrev=0` is a good way to prevent that. ;-)

medusalix, 2021-01-27:
@Roger-ZRZ Would you mind posting the logs (using `journalctl -b -u xow`)?

Roger-ZRZ, 2021-01-27:
cat .temp
-- Logs begin at Thu 2020-11-12 06:20:54 EST, end at Wed 2021-01-27 13:06:35 EST. --
Jan 27 13:01:28 dt-ruizhezh systemd[1]: Started Xbox One Wireless Dongle Driver.
Jan 27 13:01:34 dt-ruizhezh xow[1272]: 2021-01-27 13:01:34 INFO  - xow v0.5-22-g56d6f24 ©Severin v. W.
Jan 27 13:01:34 dt-ruizhezh xow[1272]: 2021-01-27 13:01:34 INFO  - Waiting for device...
Jan 27 13:01:35 dt-ruizhezh xow[1272]: 2021-01-27 13:01:35 INFO  - Wireless address: 62:45:b5:05:b2:43
Jan 27 13:01:35 dt-ruizhezh xow[1272]: 2021-01-27 13:01:35 INFO  - Dongle initialized
Jan 27 13:04:21 dt-ruizhezh xow[1272]: 2021-01-27 13:04:21 INFO  - Pairing enabled
Jan 27 13:04:26 dt-ruizhezh xow[1272]: 2021-01-27 13:04:26 INFO  - Controller '1' connected
Jan 27 13:04:26 dt-ruizhezh xow[1272]: 2021-01-27 13:04:26 INFO  - Device announced, product id: 0b12
Jan 27 13:04:26 dt-ruizhezh xow[1272]: 2021-01-27 13:04:26 INFO  - Serial number: _manually-removed_
Jan 27 13:04:27 dt-ruizhezh xow[1272]: 2021-01-27 13:04:27 INFO  - Battery level: 2
Jan 27 13:06:35 dt-ruizhezh xow[1272]: 2021-01-27 13:06:35 INFO  - Controller '1' disconnected

This is how the log is produced:
1. Uninstalled > updated to https://github.com/medusalix/xow/commit/56d6f24d151102bde71784b0cef689a8790a5bdc > installed > reboot; 
2. Re-pair controller; [success]
3. Open steam controller configuration, click identify, no rumble (there should be);
4. Turn off controller;

medusalix, 2021-01-27:
@Roger-ZRZ Hmm, your logs seem fine. You could try setting breakpoints on some of the lines of this function (or you could insert log statements):

https://github.com/medusalix/xow/blob/9d5a11b4e8610b65581c6794069ef2e70fcefbd9/controller/controller.cpp#L255-L272

Make sure that `performRumble` is actually getting called by the function.

kakra, 2021-01-27:
I could only imagine that `while (rumbleBuffer.get(rumble))` always returns false, or the wait condition never fires.