medusalix/xow #159: High CPU usage on linux 5.10

alpozcan, 2021-03-06:
Tested on Ubuntu 21.04 kernel `Linux n8 5.10.0-14-generic #15-Ubuntu SMP Fri Jan 29 15:10:03 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`.

xow version 9d5a11b4e8610b65581c6794069ef2e70fcefbd9.

```
● xow.service - Xbox One Wireless Dongle Driver
     Loaded: loaded (/etc/systemd/system/xow.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2021-03-06 17:47:13 AEDT; 4min 35s ago
   Main PID: 834 (xow)
      Tasks: 6 (limit: 76906)
     Memory: 1.0M
     CGroup: /system.slice/xow.service
             └─834 /usr/local/bin/xow

Mar 06 17:47:19 n8 xow[834]: 2021-03-06 17:47:19 DEBUG - Channel 161, power: 43
Mar 06 17:47:19 n8 xow[834]: 2021-03-06 17:47:19 DEBUG - Channel 165, power: 43
Mar 06 17:47:19 n8 xow[834]: 2021-03-06 17:47:19 INFO  - Dongle initialized
Mar 06 17:47:21 n8 xow[834]: 2021-03-06 17:47:21 INFO  - Controller '1' connected
Mar 06 17:47:22 n8 xow[834]: 2021-03-06 17:47:22 INFO  - Device announced, product id: 0b12
Mar 06 17:47:22 n8 xow[834]: 2021-03-06 17:47:22 DEBUG - Firmware version: 5.5.2641.0
Mar 06 17:47:22 n8 xow[834]: 2021-03-06 17:47:22 DEBUG - Hardware version: 1032.1.1.1
Mar 06 17:47:22 n8 xow[834]: 2021-03-06 17:47:22 INFO  - Serial number: 09711773224036
Mar 06 17:47:23 n8 xow[834]: 2021-03-06 17:47:23 INFO  - Battery level: 2
Mar 06 17:47:46 n8 xow[834]: 2021-03-06 17:47:46 DEBUG - Rumble count: 0, duration: 5000, delay: 0
```

xow is using almost 2 cores of my Intel(R) Core(TM) i5-8259U.

```
    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                                                              
    834 xow       20   0  376412   4700   4240 S 170.1   0.0   5:21.05 xow           
```

This is with an Xbox Series S controller with share button.

The constantly high CPU utilisation continues even when the gamepad is disconnected.

kapsh, 2021-03-06:
Probably the same as #141.

alpozcan, 2021-03-06:
Yes looks like it's the same issue. I'm on libusb 1.0.24 as well.

medusalix, 2021-03-07:
Yep, there's unfortunately nothing I can do about this. It's a known bug and will hopefully be fixed in the next version of `libusb`.

closed 2021-03-07 by medusalix