medusalix/xow #162: make: *** [Makefile:32: xow.o] Error 127

ziogref, 2021-03-26:
When running
make BUILD=RELEASE

I get the following output.

g++  -Wall -Wpedantic -std=c++11 -MMD -MP -Og -g -DDEBUG -DVERSION=\"v0.5-24-g9d5a11b\" -c -o xow.o xow.cpp
make: g++: No such file or directory
make: *** [Makefile:32: xow.o] Error 127

I git cloned to my user home directory and the root directory, cd into the XOW dir and run the command and this is the only output i get.

medusalix, 2021-03-26:
> make: g++: No such file or directory

You need a C++ compiler to build xow. If you're running Debian/Ubuntu you'd have to install `build-essential`, for Arch/Manjaro you need to install `base-devel`.

ziogref, 2021-03-26:
Thanks.

Might be worth mentioning that in the instructions in the prerequisites.

also now I get this error?

MattKowalskii, 2021-04-30:
I'm having this same exact problem on a fresh pop_os install! Have you found a fix yet?

closed 2021-09-21 by medusalix