medusalix/xow #163: What do the battery levels mean?

Hubro, 2021-03-26:
When I connect a controller and the log output says `Battery level: 1`, what does that mean? What is the scale? Could this information be added to README?

kakra, 2021-03-26:
It's not a scale, it's just some Microsoft-defined undocumented levels. The best guess is here:
https://github.com/atar-axis/xpadneo/blob/master/docs/reports/xb1s_battery_event.md

0. super-critical = about to die any time soon
1. low
2. medium
3. full

This probably maps to something like <5% or <10%, then around 30, around 60, and >90. It probably internally mapped to to certain voltage levels depending on the detected battery type (rechargable, disposable, li-ion) where disposable batteries have the longest endurance (probably 40 instead of 30 hours). Mapping to percentage is not linear to remaining runtime, so it's difficult to map that. For xpadneo we thus report only levels critical, low, medium, full to user space, also report that we don't support percentage levels, and let user space decide how to show it.

Interestingly one user found that the controller is able to report percentage-based battery levels in the Bluetooth protocol but apparently it doesn't do that for mine. It could be a cool project to let the controller rumble all motors from fully charged to empty and watch the level indicator and the Bluetooth reports to see when the level changes and record to percentage value associated with it (and also calculate the runtime backward after the controller died from battery).

kakra, 2021-03-26:
BTW: The controller will turn off rumble motors at level 0, maybe also at level 1 - I never looked at when it actually does that exactly.

Hubro, 2021-03-27:
@kakra Thanks! This is still useful information to have in the README though, I think :) People could easily think "3" meant "3%" and go charge their controller.

kakra, 2021-03-27:
I think a better solution would just be to log the battery level as a word and not as a number, we really don't need some indirection to a readme file.

medusalix, 2021-03-29:
> I think a better solution would just be to log the battery level as a word and not as a number, we really don't need some indirection to a readme file.

That's a great idea. I've added human-readable battery levels in 8dd0888540e64329d5d6aa1715bd201ae0e6869d.

closed 2021-03-29 by medusalix