medusalix/xow #166: Errors Building in Fedora 34 Beta

slahapman, 2021-04-08:
Hello. 
On Fedora 34, I keep getting the following error when attempting to build. I also tried the Fedora RPM instructions, and got the same issue. I have libcurl, cabextract, Development Tools and libraries installed. 

```
$ make BUILD=RELEASE
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -O3 -DVERSION=\"v0.5-25-g8dd0888\" -c -o xow.o xow.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -O3 -DVERSION=\"v0.5-25-g8dd0888\" -c -o controller/controller.o controller/controller.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -O3 -DVERSION=\"v0.5-25-g8dd0888\" -c -o controller/gip.o controller/gip.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -O3 -DVERSION=\"v0.5-25-g8dd0888\" -c -o controller/input.o controller/input.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -O3 -DVERSION=\"v0.5-25-g8dd0888\" -c -o dongle/dongle.o dongle/dongle.cpp
In file included from dongle/usb.h:21,
                 from dongle/mt76.h:21,
                 from dongle/dongle.h:21,
                 from dongle/dongle.cpp:19:
dongle/../utils/bytes.h:35:19: error: ‘size_t’ does not name a type
   35 |     inline static size_t padding(size_t count)
      |                   ^~~~~~
dongle/../utils/bytes.h:24:1: note: ‘size_t’ is defined in header ‘<cstddef>’; did you forget to ‘#include <cstddef>’?
   23 | #include <array>
  +++ |+#include <cstddef>
   24 | 
dongle/../utils/bytes.h:53:9: error: ‘size_t’ has not been declared
   53 |         size_t skipBegin,
      |         ^~~~~~
dongle/../utils/bytes.h:54:9: error: ‘size_t’ has not been declared
   54 |         size_t skipEnd = 0
      |         ^~~~~~
dongle/../utils/bytes.h:67:12: error: ‘size_t’ does not name a type
   67 |     inline size_t size() const
      |            ^~~~~~
dongle/../utils/bytes.h:67:12: note: ‘size_t’ is defined in header ‘<cstddef>’; did you forget to ‘#include <cstddef>’?
dongle/../utils/bytes.h:83:30: error: ‘size_t’ was not declared in this scope; did you mean ‘std::size_t’?
   83 |     inline const T* toStruct(size_t offset = 0) const
      |                              ^~~~~~
      |                              std::size_t
In file included from /usr/include/c++/11/cstdint:38,
                 from dongle/../utils/bytes.h:21,
                 from dongle/usb.h:21,
                 from dongle/mt76.h:21,
                 from dongle/dongle.h:21,
                 from dongle/dongle.cpp:19:
/usr/include/c++/11/x86_64-redhat-linux/bits/c++config.h:2430:33: note: ‘std::size_t’ declared here
 2430 |   typedef __SIZE_TYPE__         size_t;
      |                                 ^~~~~~
In file included from dongle/usb.h:21,
                 from dongle/mt76.h:21,
                 from dongle/dongle.h:21,
                 from dongle/dongle.cpp:19:
dongle/../utils/bytes.h:83:5: error: ‘toStruct’ declared as an ‘inline’ field
   83 |     inline const T* toStruct(size_t offset = 0) const
      |     ^~~~~~
dongle/../utils/bytes.h:83:48: error: expected ‘;’ before ‘const’
   83 |     inline const T* toStruct(size_t offset = 0) const
      |                                                ^~~~~~
      |                                                ;
dongle/../utils/bytes.h:83:21: error: data member ‘toStruct’ cannot be a member template
   83 |     inline const T* toStruct(size_t offset = 0) const
      |                     ^~~~~~~~
dongle/../utils/bytes.h:99:41: error: ‘size_t’ has not been declared
   99 |     inline void append(const T &object, size_t size = sizeof(T))
      |                                         ^~~~~~
dongle/../utils/bytes.h:114:21: error: ‘size_t’ has not been declared
  114 |     inline void pad(size_t count)
      |                     ^~~~~~
dongle/../utils/bytes.h:124:31: error: ‘size_t’ has not been declared
  124 |     inline uint8_t operator[](size_t index) const
      |                               ^~~~~~
dongle/../utils/bytes.h:129:21: error: declaration of ‘operator[]’ as non-function
  129 |     inline uint8_t& operator[](size_t index)
      |                     ^~~~~~~~
[... 78 lines cut ...]
      |                                                                ^
dongle/dongle.cpp:257:42: error: ‘const class Bytes’ has no member named ‘toStruct’
  257 |         const RxInfoCommand *info = data.toStruct<RxInfoCommand>();
      |                                          ^~~~~~~~
dongle/dongle.cpp:257:64: error: expected primary-expression before ‘>’ token
  257 |         const RxInfoCommand *info = data.toStruct<RxInfoCommand>();
      |                                                                ^
dongle/dongle.cpp:257:66: error: expected primary-expression before ‘)’ token
  257 |         const RxInfoCommand *info = data.toStruct<RxInfoCommand>();
      |                                                                  ^
dongle/dongle.cpp:279:41: error: ‘const class Bytes’ has no member named ‘toStruct’
  279 |         const RxInfoPacket *info = data.toStruct<RxInfoPacket>();
      |                                         ^~~~~~~~
dongle/dongle.cpp:279:62: error: expected primary-expression before ‘>’ token
  279 |         const RxInfoPacket *info = data.toStruct<RxInfoPacket>();
      |                                                              ^
dongle/dongle.cpp:279:64: error: expected primary-expression before ‘)’ token
  279 |         const RxInfoPacket *info = data.toStruct<RxInfoPacket>();
      |                                                                ^
dongle/dongle.cpp: In member function ‘void Dongle::readBulkPackets(uint8_t)’:
dongle/dongle.cpp:304:33: error: request for member ‘toBytes’ in ‘buffer’, which is of non-class type ‘int’
  304 |             Bytes data = buffer.toBytes(transferred);
      |                                 ^~~~~~~
make: *** [Makefile:32: dongle/dongle.o] Error 1
```

slahapman, 2021-04-08:
MAKE=DEBUG results.

```
$ make BUILD=DEBUG
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -Og -g -DDEBUG -DVERSION=\"v0.5-25-g8dd0888\" -c -o xow.o xow.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -Og -g -DDEBUG -DVERSION=\"v0.5-25-g8dd0888\" -c -o controller/controller.o controller/controller.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -Og -g -DDEBUG -DVERSION=\"v0.5-25-g8dd0888\" -c -o controller/gip.o controller/gip.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -Og -g -DDEBUG -DVERSION=\"v0.5-25-g8dd0888\" -c -o controller/input.o controller/input.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -Og -g -DDEBUG -DVERSION=\"v0.5-25-g8dd0888\" -c -o dongle/dongle.o dongle/dongle.cpp
In file included from dongle/usb.h:21,
                 from dongle/mt76.h:21,
                 from dongle/dongle.h:21,
                 from dongle/dongle.cpp:19:
dongle/../utils/bytes.h:35:19: error: ‘size_t’ does not name a type
   35 |     inline static size_t padding(size_t count)
      |                   ^~~~~~
dongle/../utils/bytes.h:24:1: note: ‘size_t’ is defined in header ‘<cstddef>’; did you forget to ‘#include <cstddef>’?
   23 | #include <array>
  +++ |+#include <cstddef>
   24 | 
dongle/../utils/bytes.h:53:9: error: ‘size_t’ has not been declared
   53 |         size_t skipBegin,
      |         ^~~~~~
dongle/../utils/bytes.h:54:9: error: ‘size_t’ has not been declared
   54 |         size_t skipEnd = 0
      |         ^~~~~~
dongle/../utils/bytes.h:67:12: error: ‘size_t’ does not name a type
   67 |     inline size_t size() const
      |            ^~~~~~
dongle/../utils/bytes.h:67:12: note: ‘size_t’ is defined in header ‘<cstddef>’; did you forget to ‘#include <cstddef>’?
dongle/../utils/bytes.h:83:30: error: ‘size_t’ was not declared in this scope; did you mean ‘std::size_t’?
   83 |     inline const T* toStruct(size_t offset = 0) const
      |                              ^~~~~~
      |                              std::size_t
In file included from /usr/include/c++/11/cstdint:38,
                 from dongle/../utils/bytes.h:21,
                 from dongle/usb.h:21,
                 from dongle/mt76.h:21,
                 from dongle/dongle.h:21,
                 from dongle/dongle.cpp:19:
/usr/include/c++/11/x86_64-redhat-linux/bits/c++config.h:2430:33: note: ‘std::size_t’ declared here
 2430 |   typedef __SIZE_TYPE__         size_t;
      |                                 ^~~~~~
In file included from dongle/usb.h:21,
                 from dongle/mt76.h:21,
                 from dongle/dongle.h:21,
                 from dongle/dongle.cpp:19:
dongle/../utils/bytes.h:83:5: error: ‘toStruct’ declared as an ‘inline’ field
   83 |     inline const T* toStruct(size_t offset = 0) const
      |     ^~~~~~
dongle/../utils/bytes.h:83:48: error: expected ‘;’ before ‘const’
   83 |     inline const T* toStruct(size_t offset = 0) const
      |                                                ^~~~~~
      |                                                ;
dongle/../utils/bytes.h:83:21: error: data member ‘toStruct’ cannot be a member template
   83 |     inline const T* toStruct(size_t offset = 0) const
      |                     ^~~~~~~~
dongle/../utils/bytes.h:99:41: error: ‘size_t’ has not been declared
   99 |     inline void append(const T &object, size_t size = sizeof(T))
      |                                         ^~~~~~
dongle/../utils/bytes.h:114:21: error: ‘size_t’ has not been declared
  114 |     inline void pad(size_t count)
      |                     ^~~~~~
dongle/../utils/bytes.h:124:31: error: ‘size_t’ has not been declared
  124 |     inline uint8_t operator[](size_t index) const
      |                               ^~~~~~
dongle/../utils/bytes.h:129:21: error: declaration of ‘operator[]’ as non-function
  129 |     inline uint8_t& operator[](size_t index)
      |                     ^~~~~~~~
dongle/../utils/bytes.h:129:30: error: expected ‘;’ at end of member declaration
[... 78 lines cut ...]
dongle/dongle.cpp:257:42: error: ‘const class Bytes’ has no member named ‘toStruct’
  257 |         const RxInfoCommand *info = data.toStruct<RxInfoCommand>();
      |                                          ^~~~~~~~
dongle/dongle.cpp:257:64: error: expected primary-expression before ‘>’ token
  257 |         const RxInfoCommand *info = data.toStruct<RxInfoCommand>();
      |                                                                ^
dongle/dongle.cpp:257:66: error: expected primary-expression before ‘)’ token
  257 |         const RxInfoCommand *info = data.toStruct<RxInfoCommand>();
      |                                                                  ^
dongle/dongle.cpp:279:41: error: ‘const class Bytes’ has no member named ‘toStruct’
  279 |         const RxInfoPacket *info = data.toStruct<RxInfoPacket>();
      |                                         ^~~~~~~~
dongle/dongle.cpp:279:62: error: expected primary-expression before ‘>’ token
  279 |         const RxInfoPacket *info = data.toStruct<RxInfoPacket>();
      |                                                              ^
dongle/dongle.cpp:279:64: error: expected primary-expression before ‘)’ token
  279 |         const RxInfoPacket *info = data.toStruct<RxInfoPacket>();
      |                                                                ^
dongle/dongle.cpp: In member function ‘void Dongle::readBulkPackets(uint8_t)’:
dongle/dongle.cpp:304:33: error: request for member ‘toBytes’ in ‘buffer’, which is of non-class type ‘int’
  304 |             Bytes data = buffer.toBytes(transferred);
      |                                 ^~~~~~~
make: *** [Makefile:32: dongle/dongle.o] Error 1

```

medusalix, 2021-04-08:
Thanks for the logs, I've been able to reproduce this on a virtual machine. It looks like GCC 11 now requires an additional `#include` to make xow compile.
I've fixed this in 99100fb52953e0218bc140bb04febd8ae51b1506, give the latest master a try.

slahapman, 2021-04-08:
Yep, that seemed to fix it. Compiled and installed completely fine with no issues. Did some testing and controller was connected right away.

Thank you!

closed 2021-04-09 by medusalix