medusalix/xow #168: Xbox Wireless Headset support

nilsroehrig, 2021-04-14:
Hi there. I've recently got a Xbox Wireless Headset and since I already own the wireless dongle, I thought I'd give it a try on Linux (elementary OS 5.1 in my case, based off Ubuntu 18.04). I haven't had used xow before and for the controller, it works absolutely fine. The headset won't pair however. 
Is there any interest to get this working? If so, how may I be able to help?

D1Ck3n, 2021-05-27:
+1 I would also be very interested in it! I also have the headset and would love to use it with my Manjaro/Arch!

D1Ck3n, 2021-05-28:
Hey, small update:
i'm able to pair the new Xbox wireless Headset with the Xbox Wireless USB Adapter but i think it will be detected as controller not as headset :( i dont see it in the media control

```
$ xow 
2021-05-28 16:38:38 INFO  - xow v0.5-26-g99100fb ©Severin v. W.
2021-05-28 16:38:38 INFO  - Waiting for device...
2021-05-28 16:39:24 INFO  - Wireless address: 62:45:b4:f4:46:40
2021-05-28 16:39:24 INFO  - Dongle initialized
2021-05-28 16:39:33 INFO  - Pairing enabled
2021-05-28 16:39:33 INFO  - Pairing enabled
2021-05-28 16:39:34 INFO  - Pairing enabled
2021-05-28 16:39:34 INFO  - Pairing enabled
2021-05-28 16:39:35 INFO  - Pairing disabled
2021-05-28 16:39:38 INFO  - Controller '1' connected
2021-05-28 16:39:38 INFO  - Device announced, product id: 0b17
2021-05-28 16:39:38 INFO  - Serial number: 00720010264049
2021-05-28 16:39:59 INFO  - Battery level: medium
2021-05-28 16:40:55 INFO  - Controller '1' disconnected
2021-05-28 16:41:11 INFO  - Controller '1' connected
2021-05-28 16:41:11 INFO  - Battery level: medium
2021-05-28 16:41:11 INFO  - Device announced, product id: 0b17
2021-05-28 16:41:11 INFO  - Serial number: 00720010264049
```

DwarflinDeveloping, 2021-08-13:
> Hey, small update:
> i'm able to pair the new Xbox wireless Headset with the Xbox Wireless USB Adapter but i think it will be detected as controller not as headset :( i dont see it in the media [...]
I have exactly the same problem. It connects to my headset (and loses connection every few seconds for whatever reason) but it doesn't show up in my speaker selection.