medusalix/xow #169: xow uses up 20% of my CPU constantly, even if controller is off

thingsiplay, 2021-04-19:
I am on Manjaro for a few months now and the issue is still there. xow constantly uses up 20% of my 4 core with hyperthreading CPU system without even having the controller on. It is paired with my wireless dongle from Microsoft and the systemctl service is enabled. Right when I boot up and do nothing, it will start the xow service and just consume 20% constantly even if I don't play. I usually stop the service after booting and only start the service when I want to play a game with gamepad support. But 20% just for the gamepad support is a bit ridiculous and can't be right.

I don't know what information I should provide. I was compiling the source code from this Github page directly and just now tried if installing xow from the AUR would solve the issue. But its the same.

kakra, 2021-04-20:
Which libusb version? Could you post a log? `journalctl -eu xow.service`

thingsiplay, 2021-04-20:
The entire log of output from `journalctl -eu xow.service`:

[journalctl -eu xow.service.txt](https://github.com/medusalix/xow/files/6342447/journalctl.-eu.xow.service.txt)

I didn't play much with the gamepad on this system so far. These usb and libusb related stuff are installed on my system:

```
$pacman -Ql libusb
local/hidapi 0.10.1-1
    Simple library for communicating with USB and Bluetooth HID devices
local/lib32-libgusb 0.3.6-1
    GObject wrapper for libusb1
local/lib32-libusb 1.0.24-1
    A cross-platform user library to access USB devices (32-bit)
local/libgusb 0.3.6-1
    GObject wrapper for libusb1
local/libusb 1.0.24-2
    Library that provides generic access to USB devices
local/libusb-compat 0.1.7-1
    Library to enable user space application programs to communicate with USB devices
local/libusbmuxd 2.0.2-1
    USB Multiplex Daemon
```
The gamepad itself works just fine and I played mostly one or two games on Steam with it so far and quick tests with RetroArch worked too. So everything seems to be just fine, besides this constant high CPU usage. xow has one hyperthreading core on 100%. 

I should mention, that I was using Ubuntu LTS before switching to Manjaro in January (a few months back) and don't remember such a problem. Maybe this is a useful hint. Here is output from `lsbusb` too:

```
$ lsusb -t -v
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        ID 2109:8110 VIA Labs, Inc. Hub
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 1038:1702 SteelSeries ApS
    |__ Port 1: Dev 2, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        ID 1038:1702 SteelSeries ApS
    |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 2109:2811 VIA Labs, Inc. Hub
        |__ Port 4: Dev 6, If 0, Class=Vendor Specific Class, Driver=, 480M
            ID 045e:02e6 Microsoft Corp. Wireless XBox Controller Dongle
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 3: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        ID 174c:3074 ASMedia Technology Inc. ASM1074 SuperSpeed hub
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 1997:2433 Shenzhen Riitek Technology Co., Ltd wireless mini keyboard with touchpad
    |__ Port 1: Dev 2, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        ID 1997:2433 Shenzhen Riitek Technology Co., Ltd wireless mini keyboard with touchpad
    |__ Port 9: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 174c:2074 ASMedia Technology Inc. ASM1074 High-Speed hub
        |__ Port 2: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
            ID 05e3:0608 Genesys Logic, Inc. Hub
            |__ Port 1: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
                ID 0079:181c DragonRise Inc.
            |__ Port 2: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
                ID 0583:2060 Padix Co., Ltd (Rockfire) 2-axis 8-button gamepad
            |__ Port 3: Dev 9, If 0, Class=Hub, Driver=hub/4p, 480M
                ID 05e3:0608 Genesys Logic, Inc. Hub
                |__ Port 2: Dev 13, If 2, Class=Audio, Driver=snd-usb-audio, 480M
                    ID 0b05:183c ASUSTek Computer, Inc.
                |__ Port 2: Dev 13, If 0, Class=Audio, Driver=snd-usb-audio, 480M
                    ID 0b05:183c ASUSTek Computer, Inc.
                |__ Port 2: Dev 13, If 3, Class=Audio, Driver=snd-usb-audio, 480M
                    ID 0b05:183c ASUSTek Computer, Inc.
                |__ Port 2: Dev 13, If 1, Class=Audio, Driver=snd-usb-audio, 480M
                    ID 0b05:183c ASUSTek Computer, Inc.
                |__ Port 2: Dev 13, If 4, Class=Human Interface Device, Driver=usbhid, 480M
                    ID 0b05:183c ASUSTek Computer, Inc.
        |__ Port 4: Dev 5, If 0, Class=Hub, Driver=hub/3p, 480M
            ID 04b4:6572 Cypress Semiconductor Corp. Unprogrammed CY7C65642 hub
            |__ Port 3: Dev 8, If 0, Class=Human Interface Device, Driver=usbhid, 12M
                ID 1e7d:3232 ROCCAT Ryos MK Pro Keyboard
            |__ Port 3: Dev 8, If 1, Class=Human Interface Device, Driver=usbhid, 12M
                ID 1e7d:3232 ROCCAT Ryos MK Pro Keyboard
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
        ID 8087:8001 Intel Corp. Integrated Hub
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
        ID 8087:8009 Intel Corp.
```

thingsiplay, 2021-04-20:
After reading in the other thread #141 I did the same and downgraded libusb for now with the command `sudo pacman -U https://archive.archlinux.org/packages/l/libusb/libusb-1.0.23-3-x86_64.pkg.tar.zst` . And added a rule for pacman to ignore this package from being updated in `sudo vim /etc/pacman.conf` with a line `IgnorePkg = libusb` . Now I do not get these CPU spikes anymore, but I wonder if this causes any incompatibility with the rest of the system.

closed 2021-04-20 by thingsiplay