medusalix/xow #170: Xbox controller model 1914 keep disconecting and reconnecting

benjdero, 2021-05-13:
I bought the bundle with Xbox One Series X controller (model 1914) and Windows Wireless Adapter.

I used them countless hours on Windows without any problem. It have latest available firmware.

When using it on my RetroPie setup on Raspberry Pi thanks to xow, it seems that the controller frequently disconnect. Not sure if related to #144 

The share button also can't be used, I wanted to set it as the HotKey button on EmulationStation, but it doesn't seems to be recognized.

```
pi@retropie:~ $ sudo systemctl status xow
● xow.service - Xbox One Wireless Dongle Driver
   Loaded: loaded (/etc/systemd/system/xow.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2021-05-13 14:29:19 CEST; 18min ago
 Main PID: 421 (xow)
    Tasks: 6 (limit: 1633)
   CGroup: /system.slice/xow.service
           └─421 /usr/local/bin/xow

mai 13 14:47:35 retropie xow[421]: 2021-05-13 14:47:35 DEBUG - Hardware version: 1032.1.1.1
mai 13 14:47:35 retropie xow[421]: 2021-05-13 14:47:35 INFO  - Serial number: 09713353907047
mai 13 14:47:37 retropie xow[421]: 2021-05-13 14:47:37 INFO  - Battery level: medium
mai 13 14:47:46 retropie xow[421]: 2021-05-13 14:47:46 INFO  - Controller '1' disconnected
mai 13 14:47:55 retropie xow[421]: 2021-05-13 14:47:55 INFO  - Controller '1' connected
mai 13 14:47:55 retropie xow[421]: 2021-05-13 14:47:55 INFO  - Device announced, product id: 0b12
mai 13 14:47:55 retropie xow[421]: 2021-05-13 14:47:55 DEBUG - Firmware version: 5.5.2641.0
mai 13 14:47:55 retropie xow[421]: 2021-05-13 14:47:55 DEBUG - Hardware version: 1032.1.1.1
mai 13 14:47:55 retropie xow[421]: 2021-05-13 14:47:55 INFO  - Serial number: 09713353907047
mai 13 14:47:57 retropie xow[421]: 2021-05-13 14:47:57 INFO  - Battery level: medium
```

ebbit1q, 2021-05-13:
this issue is most definitely the same as #144

I wish we had a solution for it but right now I'm using the new controller wired while I can use the old model wireless with the same dongle.

benjdero, 2021-05-16:
This is kinda annoying. When using xow and wireless connected controller I can't use the "power" button as the "hotkey" button for retroarch. Is there any chance you add support for the "share" button?

OriginLegend, 2021-09-17:
Not sure if this issue is related to one I am facing (seems like it is).

For me the issue was the empty-packet received by the driver results in an automatic disconnect:
https://github.com/medusalix/xow/blob/master/dongle/dongle.cpp#L272

As a result, I added a build-flag to disable this behavior: EMPTY_PACKET_DISCONNECT
Pull-Request: https://github.com/medusalix/xow/pull/185

If you want to test it make sure you clean your workspace and rebuild with this flag:
make clean
make BUILD=RELEASE EMPTY_PACKET_DISCONNECT=0

Let me know if this resolves the issues for ya'll as well

Yionnara, 2024-09-16:
not working for me ;(
```bash
[00:43]root@apta-lap-004-t:~/xow [master] $ make clean
make BUILD=RELEASE EMPTY_PACKET_DISCONNECT=0
rm -f xow xow.o controller/controller.o controller/gip.o controller/input.o dongle/dongle.o dongle/mt76.o dongle/usb.o utils/log.o utils/reader.o xow.d controller/controller.d controller/gip.d controller/input.d dongle/dongle.d dongle/mt76.d dongle/usb.d utils/log.d utils/reader.d
rm -f firmware.sh
rm -f xow.service
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -O3 -DVERSION=\"v0.5-36-gd335d60\" -DFIRMWARE=\"/lib/firmware/xow_dongle.bin\" -c -o xow.o xow.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -O3 -DVERSION=\"v0.5-36-gd335d60\" -DFIRMWARE=\"/lib/firmware/xow_dongle.bin\" -c -o controller/controller.o controller/controller.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -O3 -DVERSION=\"v0.5-36-gd335d60\" -DFIRMWARE=\"/lib/firmware/xow_dongle.bin\" -c -o controller/gip.o controller/gip.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -O3 -DVERSION=\"v0.5-36-gd335d60\" -DFIRMWARE=\"/lib/firmware/xow_dongle.bin\" -c -o controller/input.o controller/input.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -O3 -DVERSION=\"v0.5-36-gd335d60\" -DFIRMWARE=\"/lib/firmware/xow_dongle.bin\" -c -o dongle/dongle.o dongle/dongle.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -O3 -DVERSION=\"v0.5-36-gd335d60\" -DFIRMWARE=\"/lib/firmware/xow_dongle.bin\" -c -o dongle/mt76.o dongle/mt76.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -O3 -DVERSION=\"v0.5-36-gd335d60\" -DFIRMWARE=\"/lib/firmware/xow_dongle.bin\" -c -o dongle/usb.o dongle/usb.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -O3 -DVERSION=\"v0.5-36-gd335d60\" -DFIRMWARE=\"/lib/firmware/xow_dongle.bin\" -c -o utils/log.o utils/log.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -O3 -DVERSION=\"v0.5-36-gd335d60\" -DFIRMWARE=\"/lib/firmware/xow_dongle.bin\" -c -o utils/reader.o utils/reader.cpp
g++  -o xow xow.o controller/controller.o controller/gip.o controller/input.o dongle/dongle.o dongle/mt76.o dongle/usb.o utils/log.o utils/reader.o -lpthread -lusb-1.0
sed "s|#FIRMWARE#|/lib/firmware/xow_dongle.bin|" install/firmware.sh.in > firmware.sh
sed 's|#BINDIR#|/usr/local/bin|' install/service.in > xow.service
[00:43]root@apta-lap-004-t:~/xow [master] $ service xow stop
[00:43]root@apta-lap-004-t:~/xow [master] $ ./xow 
2024-09-17 00:43:35 INFO  - xow v0.5-36-gd335d60 ©Severin v. W.
2024-09-17 00:43:35 INFO  - Waiting for device...
2024-09-17 00:43:36 INFO  - Wireless address: 62:45:b5:06:47:ab
2024-09-17 00:43:36 INFO  - Dongle initialized
2024-09-17 00:43:45 INFO  - Pairing enabled
2024-09-17 00:43:45 INFO  - Pairing enabled
2024-09-17 00:43:45 INFO  - Pairing enabled
2024-09-17 00:43:46 INFO  - Controller '1' connected
2024-09-17 00:43:47 INFO  - Device announced, product id: 0b12
2024-09-17 00:43:47 INFO  - Serial number: 09710509793338
2024-09-17 00:43:47 INFO  - Battery level: medium
2024-09-17 00:44:19 INFO  - Controller '1' disconnected
2024-09-17 00:44:26 INFO  - Controller '1' connected
2024-09-17 00:44:26 INFO  - Device announced, product id: 0b12
2024-09-17 00:44:26 INFO  - Serial number: 09710509793338
2024-09-17 00:44:27 INFO  - Battery level: medium
```

Yionnara, 2024-09-16:
I think it works fine when it is used, but keep disconnecting while is not, i dont have any problem since i launch a game

TheRealIndru, 2025-02-03:
I am having a similar issue. The controller keeps disconnecting the instant is connecting.

Any ideas on what I can do to fix it? It's a Series X controller.

`2025-02-03 15:02:39 INFO  - xow v0.5-36-gd335d60 ©Severin v. W.
2025-02-03 15:02:39 INFO  - Waiting for device...
2025-02-03 15:02:39 INFO  - Wireless address: 62:45:b5:04:6a:e0
2025-02-03 15:02:39 INFO  - Dongle initialized
2025-02-03 15:02:52 INFO  - Pairing enabled
2025-02-03 15:02:53 INFO  - Pairing enabled
2025-02-03 15:02:58 INFO  - Pairing enabled
2025-02-03 15:03:05 INFO  - Pairing enabled
2025-02-03 15:03:06 INFO  - Pairing disabled
2025-02-03 15:03:06 INFO  - Controller '1' connected
2025-02-03 15:03:07 INFO  - Device announced, product id: 0b12
2025-02-03 15:03:07 INFO  - Serial number: 09710344106139
2025-02-03 15:03:08 INFO  - Controller '1' disconnected
2025-02-03 15:04:51 INFO  - Pairing enabled
2025-02-03 15:04:52 INFO  - Pairing enabled
2025-02-03 15:04:54 INFO  - Controller '1' connected
2025-02-03 15:04:55 INFO  - Device announced, product id: 0b12
2025-02-03 15:04:55 INFO  - Serial number: 09710344106139
2025-02-03 15:04:55 INFO  - Pairing enabled
2025-02-03 15:04:57 INFO  - Controller '1' disconnected
2025-02-03 15:05:02 INFO  - Pairing enabled
2025-02-03 15:05:03 INFO  - Pairing enabled
2025-02-03 15:05:05 INFO  - Controller '1' connected
2025-02-03 15:05:06 INFO  - Device announced, product id: 0b12
2025-02-03 15:05:06 INFO  - Serial number: 09710344106139
2025-02-03 15:05:08 INFO  - Controller '1' disconnected
2025-02-03 15:05:14 INFO  - Controller '1' connected
2025-02-03 15:05:14 INFO  - Device announced, product id: 0b12
2025-02-03 15:05:14 INFO  - Serial number: 09710344106139
2025-02-03 15:05:16 INFO  - Controller '1' disconnected
2025-02-03 15:05:21 INFO  - Controller '1' connected
2025-02-03 15:05:21 INFO  - Controller '1' disconnected
2025-02-03 15:05:26 INFO  - Pairing enabled
2025-02-03 15:05:27 INFO  - Pairing enabled
2025-02-03 15:05:34 INFO  - Pairing disabled
2025-02-03 15:05:34 INFO  - Controller '1' connected
2025-02-03 15:05:35 INFO  - Device announced, product id: 0b12
2025-02-03 15:05:35 INFO  - Serial number: 09710344106139
2025-02-03 15:05:37 INFO  - Controller '1' disconnected`

zerkerX, 2025-02-03:
From what I've seen, most of the recent development has been against the kernel module version, [xone](https://github.com/medusalix/xone). It doesn't have this issue.

zerkerX, 2025-02-04:
Then I would suggest starting a discussion topic on the `xone` repo. Xone works for me.