medusalix/xow #176: xow stops at "Waiting for device"

cademullen-1219, 2021-07-01:
So xow was working for me for a little while, but just recently it has stopped detecting my adapter. I tried a bunch of different USB ports on my PC, countless reboots and resets of the xow service with the adapter unplugged/plugged in; no luck. I've heard that dual booting with Windows can cause issues, and I have to use Windows for my work, so I fear that may be what's causing this. What's even worse is that _Windows_ doesn't even recognize the device anymore, it says "Device Descriptor Request Failed". My best guess is that it's some kind of firmware issue with the adapter, but I have no idea how to go about fixing it. My apologies if this is a known issue, I've done my best to search for help and I haven't found anything. I hope somebody can help me out here, I'm at a loss.

Here's a recent log:

```-- Boot a129129ba4da4183a425466e9c15ff2b --
Jun 30 22:31:44 cade-h110ma systemd[1]: Started Xbox One Wireless Dongle Driver.
Jun 30 22:31:50 cade-h110ma xow[733]: 2021-06-30 22:31:50 INFO  - xow v0.5-27-g700529b ©Severin v. W.
Jun 30 22:31:50 cade-h110ma xow[733]: 2021-06-30 22:31:50 INFO  - Waiting for device...
Jun 30 22:34:37 cade-h110ma systemd[1]: Stopping Xbox One Wireless Dongle Driver...
Jun 30 22:34:37 cade-h110ma systemd[1]: xow.service: Deactivated successfully.
Jun 30 22:34:37 cade-h110ma systemd[1]: Stopped Xbox One Wireless Dongle Driver.
```

Here's an older log from when it was working (expected output):

```-- Boot 09695b8832534e528e7f76f0ff485d7d --
Jun 18 10:53:09 cade-h110ma systemd[1]: Started Xbox One Wireless Dongle Driver.
Jun 18 10:53:14 cade-h110ma xow[721]: 2021-06-18 10:53:14 INFO  - xow v0.5-27-g700529b ©Severin v. W.
Jun 18 10:53:14 cade-h110ma xow[721]: 2021-06-18 10:53:14 INFO  - Waiting for device...
Jun 18 10:53:14 cade-h110ma xow[721]: 2021-06-18 10:53:14 INFO  - Wireless address: 62:45:b4:fd:e1:18
Jun 18 10:53:14 cade-h110ma xow[721]: 2021-06-18 10:53:14 INFO  - Dongle initialized
Jun 18 11:02:44 cade-h110ma xow[721]: 2021-06-18 11:02:44 INFO  - Controller '1' connected
Jun 18 11:02:45 cade-h110ma xow[721]: 2021-06-18 11:02:45 INFO  - Device announced, product id: 02ea
Jun 18 11:02:45 cade-h110ma xow[721]: 2021-06-18 11:02:45 INFO  - Serial number: 02600564056637
Jun 18 11:02:48 cade-h110ma xow[721]: 2021-06-18 11:02:48 INFO  - Battery level: medium
```
Screenshots from Windows:

[image]

[image]

Wireless adapter model: 6HN-00002
Xbox controller model: 1708

kakra, 2021-07-01:
Try to use a different USB port. If that doesn't help, find an active USB hub (with external power source), connect that to your computer, and try connecting the dongle to that hub.

cademullen-1219, 2021-07-01:
I tried lots of different ports on my PC and none of them worked. I also tried on a fairly new Macbook running Windows through bootcamp, and that didn't work either. I got the same "Device Descriptor Request Failed" error.

Something that I should mention is that I got this adapter on eBay, which may explain why it's faulty. The strange thing is that it _was_ working earlier; perhaps the device was extremely degraded from previous use and only had a short lifespan as a result. I think I might want to try getting a new adapter from the Microsoft store.

kakra, 2021-07-01:
Yeah, maybe it was on eBay because it was not working reliably in all devices already for the previous owner...

cademullen-1219, 2021-07-01:
It was from a vendor that resells used adapters and claims that they are tested and working. Hopefully the last time I make that mistake... At any rate I've ordered a new one from Microsoft and it should get here in about a week, with any luck this will fix my issues.

mpaccione, 2022-04-08:
Hi I am having a similar issue however just upgraded Linux. 
Did you figure out a fix?

TheVanMan2345, 2022-04-08:
Not quite a fix, but a workaround. 

I could only replicate the issue if I had booted up win11 and then boot back to Manjaro. Once I boot back into Manjaro I have to force quit xow, unplug the adapter, restart xow, wait until the message “waiting for device” appears and then plug back in the Xbox adapter. 

That will keep it working until the next time I boot into windows.

mpaccione, 2022-04-08:
Gotcha. I ended up going with xboxdrv and it works although the mappings
are all wonk so I have to figure that out.