medusalix/xow #181: Uses 100% one CPU thread

zaryuta, 2021-08-22:
Uses 100% of one CPU thread. Tried on Manjaro kernel 5.13 and PopOs kernel 5.12.

Galcian79, 2021-09-06:
I patched libusb from source, i confirm now it works as intended.

sorgelig, 2021-09-10:
@Galcian79 
which patch you've used and to which libusb version?

Galcian79, 2021-09-10:
> @Galcian79
> which patch you've used and to which libusb version?

The version i patched is the latest 1.0.24
This is the upstreamed patch https://github.com/libusb/libusb/commit/32a22069428cda9d63aa666e92fb8882a83d4515

sorgelig, 2021-09-11:
i see.. 
i think it's better to compile statically with older libusb and leave current libusb version in system for stability purpose. At least till 1.0.25 release.

medusalix, 2022-03-20:
Duplicate of #141.

closed 2022-03-20 by medusalix