medusalix/xow #186: Error when trying to MAKE=BUILD

R0bert0r, 2021-09-19:
Hello, 
          Im reciving the followng error when trying to make the file and im not sure how to fix it: 

```
tom@tom-AX370:~/Downloads/xow$` make BUILD=RELEASE
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -O3 -DVERSION=\"v0.5-27-g700529b\" -c -o xow.o xow.cpp
In file included from xow.cpp:21:
dongle/usb.h:29:10: fatal error: libusb-1.0/libusb.h: No such file or directory
   29 | #include <libusb-1.0/libusb.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:32: xow.o] Error `1
```

Please help me figure out the error. 

thanks

kakra, 2021-09-19:
Install libusb-dev or libusb-headers

R0bert0r, 2021-09-21:
> b

I installed libusb-dev but im still getting the same erroe when running make BUILD=RELEASE

```
make BUILD=RELEASE
g++  -Wall -Wpedantic -std=c++11 -MMD -MP -O3 -DVERSION=\"v0.5-27-g700529b\" -c -o xow.o xow.cpp
In file included from xow.cpp:21:
dongle/usb.h:29:10: fatal error: libusb-1.0/libusb.h: No such file or directory
   29 | #include <libusb-1.0/libusb.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:32: xow.o] Error 1

```

williamtcastro, 2021-12-30:
@nick-prat thanks, runing `sudo apt install libusb-1.0-0-dev` worked for me.

trewest, 2022-01-16:
for folks on fedora, `sudo dnf install -y libusbx-devel`

fjrider, 2022-01-28:
Love the controller driver, thank you!

On Linux Mint (Ubuntu Base) I could not get BUILD=RELEASE to run.

I found out I had to install g++ (not sure what that is or does but then it worked).

closed 2022-03-20 by medusalix